picolibc/picolibc #274: About tinystdio's dismal performance

rdiez, 2022-04-09:
The Newlib stdio (known as "legacy stdio" in Picolibc) defines a FILE structure with the following members (among others):

```
_read  (struct _reent *, void *, char *      , _READ_WRITE_BUFSIZE_TYPE);
_write (struct _reent *, void *, const char *, _READ_WRITE_BUFSIZE_TYPE);
```

Note that these calls have a data length argument.

The "new" Tinystdio family defines FILE with the following members:

```
int (*put)(char, struct __file *);	/* function to write one char to device */
int (*get)(      struct __file *);	/* function to read one char from device */
```

Note that these routines transfer only 1 byte at a time.

I stepped through the following function call with the debugger:

`sprintf(` buffer, "%d", 12345678 );

And it is true, Tinystdio makes one function call per digit. That alone means a sizeable performance loss.

Furthermore, while Newlib divides by 10 on each iteration, like you would expect, Tinystdio calls __ultoa_invert(), which ends up using divmod10(), a routine described as "krufty" in the source code. That will probably make the code at least an order of magnitude slower. I am guessing that this "optimisation" is useful on 8-bit AVR microcontrollers, but counterproductive on regular 32-bit CPUs.

Have I missed some configuration option to optimise all this?

If not, such a drastic performance drop should be documented, because users coming from Newlib probably do not expect it.

Is this the only performance surprise? Or are there other areas where Tinystdio is much slower than Newlib?

keith-packard, 2022-04-09:
> And it is true, Tinystdio makes one function call per digit. That alone means a sizeable performance loss.

Yes, it makes a function call per digit (or, in fact, for any output at all). However, that doesn't mean it makes a call all the way down to whatever implements basic I/O on your device -- you can create a FILE that provides buffering, like the tinystdio POSIX I/O module does. I think that should offer nearly the same performance as moving buffering up inside the stdio library; but if you've got numbers that show a significant performance impact, I'd love to see them.
 
> Furthermore, while Newlib divides by 10 on each iteration, like you would expect, Tinystdio calls __ultoa_invert(), which ends up using divmod10(), a routine described as "krufty" in the source [...]

Yeah, you can imagine this came from the original AVR C library. For systems with native 64-bit long long types, or when building Picolibc without long long support, you'll note that 'divmod10' does just use divide and remainder operators.

> Is this the only performance surprise? Or are there other areas where Tinystdio is much slower than Newlib?

I don't think anyone has done performance measurements on either library, so I really couldn't say. I suspect picolibc probably does floating point conversions faster as it doesn't use arbitrary precision arithmetic, but I haven't even measured that.

rdiez, 2022-04-10:
> Yes, it makes a function call per digit (or, in fact, for any output at all). However, that doesn't mean it makes a call all the way down to whatever implements basic I/O on your device

I am not using FILE I/O at all in my firmware. I am just worried that things like sprintf will be much slower if every byte needs a function call. Think of the following:

`sprintf( buffer, "100 bytes and %s", "100 bytes" )`

That is going to translate to 200 function calls. It may not be 200 times slower, but it probably won't be far off either.

keith-packard, 2022-04-10:
It's really not all that terrible -- there is so much to deal with on every byte that the added overhead of a couple of branches (as that's really the whole overhead of a function call these days) is generally not a significant performance issue. I'd encourage you to measure the performance of sprintf("long string %s", other_string) or some other stdio-based string operation compared with strcat and see just how much difference there actually is. Even better, compare that with the newlib stdio backend.

keith-packard, 2022-04-12:
I'm going to close this as we don't know of any concrete performance issues. If you end up measuring performance and finding a significant gap between tinystdio and newlib stdio, please post what you find and re-open this issue.

closed 2022-04-12 by keith-packard

rdiez, 2022-04-16:
I just ran a very non-scientic sprintf test on Qemu simulating an ARM Cortex-M3.
With Newlib: 3,62 s
With Picolibc's tinystdio: 29,9 s
That's factor 8. I must admit I expected an even bigger slowdown.

The test code is:

```
static const char s_testData[] = "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890"
                                 "1234567890";

static const char s_testFmt[] = "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "1234567890"
                                "%s";
static
void Test5 ( void )
{
  assert( strlen( s_testData ) == 100 );
  assert( strlen( s_testFmt  ) == 102 );

  // Compile this code with -fno-builtin.

  char buffer[300];

  for ( unsigned i = 0; i < 2'000'000; ++i )
  {
    sprintf( buffer, s_testFmt, s_testData );
  }
}

```

keith-packard, 2022-04-18:
Thanks for the helpful example. I was thinking about this yesterday and wondered if open coding the relevant bits of fputc inside vfprintf would help much. It does help some, eliminating *another* function call, but there's still a lot of function calls. What it also does is *shrink* the overall example, so the net is less code *and* slightly improved performance. Take a look and see what you think.

https://github.com/picolibc/picolibc/tree/inline-printf-put

I'm getting about 11.8 seconds for tinystdio compared with 2.3 seconds for newlib stdio, in release mode, also running under qemu for cortex m3.

rdiez, 2022-04-18:
I see tinystdio wasn't checking for errors from putc. I hope there aren't many more surprises in store.

The inline-printf-put optimisation you did is an improvement. However, I think that the only sensible way forward is to modify "struct __file" so that put() and get() can take a block of data, like Newlib does with _read() and _write() in "struct __sFILE".

keith-packard, 2022-04-18:
> I see tinystdio wasn't checking for errors from putc. I hope there aren't many more surprises in store.

I keep finding more test suites to run picolibc through, which catch different bugs. Last week was Zephyr, which found that vfprintf wasn't checking putc return values *and* a couple more missing (unsigned char) casts.

> The inline-printf-put optimisation you did is an improvement. However, I think that the only sensible way forward is to modify "struct __file" so that put() and get() can take a block of data, like Newlib does with _read() and _write() in "struct __sFILE".

Yes, if your goal is performance, then doing chunks of data in a single call would be a good next step. However, that works against the explicit goal of tinystdio to save memory for smaller systems. If you're using stdio for internal computations, then it's going to disappoint, but if stdio is connected to external serial or USB devices, then I think the current design is fast enough.

If you want to experiment with replacing the get/put calls with read/write and inserting buffering into the FILE struct, I'd certainly be willing to help get that merged as an option when building the library, but we're going to keep the existing char-at-a-time code because it's really useful on small targets with limited use of stdio.

rdiez, 2022-04-19:
> If you want to experiment with replacing the get/put calls with read/write and inserting buffering into the FILE struct,

I wouldn't go as far as adding buffering to tinystdio. Improving get/put with multibyte support would be nice though, I'll put it on my (very very long) to-do list.

What's the status of the Newlib stdio in Picolibc? It is described as "legacy stdio". Does that mean it is discouraged? Do some (or all) of the libc I/O tests get executed against that "legacy stdio" part?

keith-packard, 2022-04-20:
Newlib stdio works, and is exhaustively tested in the same way that tinystdio is. The biggest gaps:

 * it's still tied to struct _reent, which means it uses a lot more memory while not using TLS for OS-independent reentrancy. I gave up trying to untangle it as I don't expect many people to use it.
 * it uses malloc for everything, so make sure you've got plenty of heap.
 * floating point conversions are slow and use malloc in ways that isn't entirely failure-proof (out-of-memory can cause a panic)