picolibc/picolibc #277: Better support musl

xhebox, 2022-04-12:
refer enjoy-digital/litex#1272.

In short, when compile with https://github.com/enjoy-digital/litex/blob/7416943f9c083ae2f3845ae82957a80e9e21c49c/litex/soc/software/libc/Makefile#L41-L58 on riscv64-pc-linux-musl machine:

1. `size_t` is not defined `sys/_types.h`
2. `wint_t` is not defined in `sys/_types.h`

keith-packard, 2022-04-14:
This is confusing -- why are you trying to mix picolibc and musl? They're both C libraries and a system generally only needs one. Musl seems to be designed as a glibc replacement, with Linux support. Picolibc is designed for embedded systems running either no OS or a small RTOS.

xhebox, 2022-04-14:
> This is confusing -- why are you trying to mix picolibc and musl? They're both C libraries and a system generally only needs one. Musl seems to be designed as a glibc replacement, with Linux support. Picolibc is designed for embedded systems running either no OS or a small RTOS.

Sorry for being not clear. I am not trying to mix them. My host is musl-based, and for litex, I need to cross-compile picolibc to riscv64. And I highly suspect that it is the difference of libc causing compiling errors. Because same configuration with a gcc/glibc toolchain will compile picolibc just fine but not gcc/musl.

xhebox, 2022-04-14:
I would guess that picolibc is still refering header files(stddef?), and it is not completely dependency-free. And the stddef is from musl, thus it does not import the type correctly as glibc.

rdiez, 2022-04-14:
> Picolibc is designed for embedded systems running either no OS or a small RTOS.

That's right. Why are you trying to cross-compile to riscv64-pc-linux-musl then? That target is no embedded system, but is intended for a full-blown Linux.

xhebox, 2022-04-14:
> That's right. Why are you trying to cross-compile to riscv64-pc-linux-musl then? 

I am trying to use litex, which downloaded `riscv64-unknown-linux-gnu` cross-compiler to compile picolibc as the target libc for their soc emulator. I just did the same thing with a custom `riscv64-pc-linux-musl` toolchain.

Maybe my word is confusing, I actually meant that use cross-compiler `riscv64-pc-linux-musl` to compile to `riscv64-pc-elf-picolibc`(which is used by litex soc softwares, e.g. bios).

keith-packard, 2022-04-18:
Because the compiler is also built for a specific OS, you kinda are still 'cross compiling' when you use a compiler which targets Linux to build embedded software that runs without Linux. It generally works, but you need to be careful to avoid referring to any native Linux libraries. You'll see examples in picolibc which use the linux x86 and aarch64 compilers.

xhebox, 2022-04-19:
> Because the compiler is also built for a specific OS, you kinda are still 'cross compiling' when you use a compiler which targets Linux to build embedded software that runs without Linux. It [...]

I think `litex` does avoid refering native linux libraries(`-nostdlib` or so), but picolibc is refering host headers. The problem is that picolibc expect glibc headers instead of musl headers...

xhebox, 2022-04-19:
@keith-packard OK. i think I figured it out:

I need `COMMONFLAGS += -nostdinc -I/lib/gcc/riscv64-pc-linux-musl/11.2.0/include -I/lib/gcc/riscv64-pc-linux-musl/11.2.0/include-fixed` to compile picolibc correctly for `riscv64-xx-linux-musl`.

enjoy-digital/litex#976 removed `nostdinc`, **but that is not correct**. Without `nostdinc`, the compiler will refer system headers, which, I guess, is not the intention of picolibc. But since glibc has similar macros as in `/lib/gcc`, so referring system headers did not lead to any error during compilation.

I changed libc to musl, and the real problem is exposed to me.

rdiez, 2022-04-19:
> I need `COMMONFLAGS += -nostdinc -I/lib/gcc/riscv64-pc-linux-musl/11.2.0/include -I/lib/gcc/riscv64-pc-linux-musl/11.2.0/include-fixed` to compile picolibc correctly for `riscv64-xx-linux-musl`.

I thought you said the target was riscv64-pc-elf-picolibc, and not riscv64-xx-linux-musl.

You shouldn't need to specify any include files from riscv64-pc-linux-musl or the like. Or is that where your Picolibc headers will land? That would mean you are building as root and installing under /lib, which probably is not what you want.

> https://github.com/enjoy-digital/litex/pull/976 removed nostdinc, but that is not correct. Without nostdinc, the compiler will refer system headers, which, I guess, is not the intention of picolibc as an embedded libc for bios software.

-nostdinc and the like are only needed if you are using a compiler built for another system with another libc. Reusing a compiler is problematic, especially if you want to use C++ later on. I suggest that you build your toolchain from scratch for Picolibc, so that you do not have any problems with include files from other systems and/or libcs. This project of mine does that, and does not have any such issues: https://github.com/rdiez/JtagDue

xhebox, 2022-04-19:
> I thought you said the target was riscv64-pc-elf-picolibc, and not riscv64-xx-linux-musl.

Nah, sorry for my problematic explanation. I compile `riscv64-xx-linux-musl` because it is more convenient than `riscv64-pc-elf-picolibc`. And the most important point is that `litex` did not use `riscv64-pc-elf-picolibc` toolchain, so do I. I have a detailed explanation about what I wanted to do at the end.

> You shouldn't need to specify any include files from riscv64-pc-linux-musl or the like. Or is that where your Picolibc headers will land? That would mean you are building as root and installing under /lib, which probably is not what you want.

`/lib/gcc/riscv64-pc-linux-musl/11.2.0/include` is where `stddef.h/stdbool.h/limits.h` headers for riscv64 architecture are located. These headers are referenced by picolibc. picolibc is not dependency-free in the aspect of headers.

And musl-based gcc is not able to lookup for `limits.h`, also it will include `stddef.h` from musl instead of the one in `/lib/gcc/riscv64-pc-linux-musl/11.2.0/include`. These two problems will not occur on non-musl toolchain, because gcc is hardcoded to be so

> -nostdinc and the like are only needed if you are using a compiler built for another system with another libc. Reusing a compiler is problematic, especially if you want to use C++ later on. I [...]

I, of course, could build my own toolchain `riscv64-pc-elf-picolibc`. But it is better to works out of box without compiling a custom toolchain. `litex` guys will definitely prefer `riscv64-pc-linux-gnu` static binaries, which is distributed officially. Considering `litex` will only use picolibc for bios software, it is OK to use the toolchain as if it is `riscv64-pc-elf-xxx`.

This means, in 10 minutes, you could start `litex` and use it for whatever you want to do. without compiling gcc.

> This project of mine does that, and does not have any such issues: https://github.com/rdiez/JtagDue

Nice project, starred. I could explain more about what I wanted to do. I am designing a micro-processor and want to use `litex` to do a full simulation.

And `litex` is using officially distributed `riscv64-pc-linux-gnu` static binaries to compile the bios software for simulation. But the official binaries are really out-dated. So I compile my own `riscv64-xx-linux-musl` shared binaries with latest gcc/binutils. Then, I found several problems with my toolchain, one of them is this issue.

rdiez, 2022-04-19:
I think that limits.h comes with Picolibc, and stddef.h and stdbool.h come with GCC (gcc-11.2.0/gcc/ginclude/).

I am guessing that litex uses Linux for its main firmware, and a bare metal environment for its BIOS. In order to reuse the GCC for Linux/Musl to build the BIOS, you are going to have to be very careful not to mix headers.

The litex BIOS does not use C++, so I suppose you could use the GCC .specs files that Picolibc generates. But if your compiler is installing headers like stddef.h to some global location where Linux and Musl headers also land, it is going to be hard to get GCC to only include the right ones and not touch anything from Musl or Linux. A header file mixture may work today, because libc flavours tend to be somewhat compatible, but may break with the next version of Musl.

This is an area that I do not know well enough. Maybe Keith can tell us how this header separation or coexistence should work.

xhebox, 2022-04-19:
> I think that limits.h comes with Picolibc, and stddef.h and stdbool.h come with GCC (gcc-11.2.0/gcc/include/).

limits.h comes from `/lib/gcc/*/include-fixed`. If you are using one from picolibc, you need to bootstrap picolibc first. That is what you said, build a `riscv64-pc-linux-picolibc` toolchain.

> I am guessing that litex uses Linux for its main firmware, and a bare metal environment for its BIOS.

`litex` does not come with a linux firmware, you need to build your own. There is one example in litex-hub/linux-on-litex-vexriscv, though. As a processor designer, I am happy with a bare metal environment :)

> This is an area that I do not know well enough. Maybe Keith can tell us how this header separation or coexistence should work.

It is something related to `sysroot`, I have `/riscv64-pc-linux-musl` directory where all musl-based softwares are installed there. And my x86_64 host  files are under `/`. So these file could not be mixed. But includes-path may be wrong since the default will not always work..

rdiez, 2022-04-19:
You are right, Picolibc's limits.h refers to GCC's limits.h if it exists. There may be a standard where all these headers are listed, but I'm afraid I don't know enough in this respect.

> That is what you said, build a riscv64-pc-linux-picolibc toolchain.

No quite. If you are not building a Linux-based firmware (like a bootloader which does not use Linux, or your own bare metal firmware) then you toolchain shouldn't have "pc", "linux" or "musl" in its name, and shouldn't have or use any Musl or Linux headers. For example, the toolchain I am using is called "arm-none-eabi", and if you look at my Makefile, you will see no Musl, glibc or Linux references inside.

You can reuse an existing toolchain or compiler under certain circumstances, and the sysroot you mentioned is probably one of the methods, but I cannot help you further on this path.

If I were you, I wouldn't expect anything in the embedded software world to work "out of the box". 8-)

> picolibc expect glibc headers instead of musl headers

That's not correct as far as I know. Like Keith said, Picolibc is a replacement for those libraries, so it shouldn't need any headers from them. At most headers like stddef.h which come with GCC. If your build system manages to include a header from Linux, from Musl or from glibc, then something is not right, and it is going to break sooner or later.

xhebox, 2022-04-19:
> That's not correct as far as I know. Like Keith said, Picolibc is a replacement for those libraries, so it shouldn't need any headers from them. At most headers like stddef.h which come with GCC.
 
Yeah, I was meant so. But my phrase are not accurate enough. Your explanation on target triples and headers are very concise and clear. That inspired me a lot. I never really figured out what exactly cross-compiler does until this issue.

> If your build system manages to include a header from Linux, from Musl or from glibc, then something is not right, and it is going to break sooner or later.

No, I always build them with correct prefixes and sysroot. I never try to mix them together.

keith-packard, 2022-04-21:
yeah, it sounds like when you build gcc with musl support, that pulls some musl header files into the standard gcc search path which get used in preference to the internal gcc ones. That could result from `#include_next` tricks that some gcc headers use for mysterious reasons.

As you've gotten this to build by adjusting your compiler flags to include `-nostdinc`, I'll go ahead and close this issue; if you want to write up a patch for the docs to explain what happened in case other people hit the same issue, that'd be helpful.

closed 2022-04-21 by keith-packard