picolibc/picolibc #283: Confusion about configuration setting io-long-long

rdiez, 2022-04-16:
The documentation in doc/build.md mentions setting `io-long-long`, but file meson_options.txt has 2 variants:

```
option('io-long-long', type: 'boolean', value: false,
       description: 'enable long long type support in IO functions like printf/scanf')
option('newlib-io-long-long', type: 'combo', choices: ['true', 'false', 'auto'], value: 'auto',
       description: 'enable long long type support in IO functions like printf/scanf')
```
They have exactly the same description. If `newlib-io-long-long` is there for some compatibility reasons, the description and/or documentation should mention it, and whether it is deprecated.

Furthermore, doc/build.md states:

`[...] For tiny-stdio, this only affects the integer-only versions, the full version always includes long long support.`

and

`long long support is always enabled for the tinystdio full printf/scanf modes, the io-long-long option adds them to the limited (float and integer) versions, [...]
`

Those 2 are somewhat redundant, but not quite the same. If those "full" and "limited" variants refer to `format-default` values `double`, `float` and `integer`, then it is better to use those terms.

keith-packard, 2022-04-18:
> The documentation in doc/build.md mentions setting `io-long-long`, but file meson_options.txt has 2 variants:
>
> ```
> option('io-long-long', type: 'boolean', value: false,
>        description: 'enable [...]

There are several such compatibility definitions that could use an update.

> Furthermore, doc/build.md states:
>
> `[...] For tiny-stdio, this only affects the integer-only versions, the full version always includes long long support.`
>
> and
>
> `long long support is always enabled [...]

This is a bit confusing, but the 'format-default' value doesn't control whether the double/integer versions of vfprintf and vfscanf are available, but only what function these names refer to in the absence of a linker symbol assignment operation. So, I'm not sure how to refer to the version of vprintf in the library which handles only integer numbers and which may be called either vfprintf or __i_vprintf...

eli-schwartz, 2022-04-18:
Note that since Meson 0.60.0 you can use https://mesonbuild.com/Build-options.html#deprecated-options to mark those options as deprecated, however currently you support 0.53 so that feature cannot be used.

keith-packard, 2022-04-20:
Yeah, we're sticking with 0.53 because there's a common ubuntu version that shipped with that. Someday we'll switch :-)

closed 2026-11-10 by keith-packard