picolibc/picolibc #286: m68k: memcpy does unaligned accesses on 68010

alexisvl, 2022-04-23:
This is a minor thing on a specific CPU build type I'm probably the only one to ever use, but...

m68k memcpy.S (inherited from newlib) has the following:

```
#if defined (__mcoldfire__) || defined (__mc68010__) || defined (__mc68020__) || defined (__mc68030__) || defined (__mc68040__) || defined (__mc68060__)
# define MISALIGNED_OK 1
#else
# define MISALIGNED_OK 0
#endif
```

It's a stretch to call misaligned writes "OK" on 68010. It supports them in _software_, if and only if the operating system provides an address error handler that knows how to emulate the access. Newlib/picolibc does not provide this, and even if it did, vectoring to a software emulation routine for every single word/longword access is WAY slower than just copying byte for byte. The 68010 should probably be removed from this.

Doesn't seem to be present in any other files, just memcpy.S.

keith-packard, 2022-04-23:
Thanks for the report; does #287 look right to you? It seems to generate the correct instructions when tested here, but I don't have any 68010 hardware to test with.

alexisvl, 2022-04-24:
Yup, that's exactly what I just built and tested - was going to submit a PR myself. Confirmed working.

keith-packard, 2022-04-24:
Thanks for verifying that it works!

PR #287 (2022-04-24): m68k: Don't use unaligned access on m68010 memcpy
It's a stretch to call misaligned writes "OK" on 68010. It supports
them in _software_, if and only if the operating system provides an
address error handler that knows how to emulate the
access. Newlib/picolibc does not provide this, and even if it did,
vectoring to a software emulation routine for every single
word/longword access is WAY slower than just copying byte for
byte. The 68010 should probably be removed from this.

Closes #286 

Reported by: Alexis Lockwood
closed 2022-04-24 by keith-packard