picolibc/picolibc #290: picolib suitable for position independent code?

basprins, 2022-05-01:
Hi, 

I am looking at picolibc as a replacement for libc_nano, to use it for my project on an nxp imx cpu (an arm cortex-m7). We need position independence, and we write our code in c++. In a prototype I am running into issues when compiling our project with -fpic. Well, compiling goes fine, but when I upload the binary to target, I run into hard faults as soon as I execute functions from libc. 

The first problems (`__libc_init_array`) I "worked around" by copying the function from libc source code into my own startup file. This works. So now I have my position independent prototype working. But, when I call `printf` in my app just to test another call of libc_nano, again I go into a hard fault. 

So I get the feeling libc_nano is not suitable for position independency. While googling I came across this project. It seems rather popular, and active. So I started to wonder wouldn't I be better of depending on picolibc instead of libc_nano. And, most important, could you tell me if picolibc is / can be compiled with -fpic? 

Would be more than grateful if somebody could fine some time to react on this! 

Many thanks in advance!

keith-packard, 2022-05-02:
I don't know of anything in picolibc or any other C library that would break when using -fPIC, but I've never needed to actually relocate stuff. I don't think any of the assembly code in the library would need to be changed; for ARM, all we've got are optimized versions of string and memory functions. I think you need to update the GOT (_GLOBAL_OFFSET_TABLE_) when loading the app? Which means the GOT would need to be in RAM, but at a fixed position relative to the code?

Here's a web page I found which seems to explain how this might work on an embedded system; I suspect you already know most of this though. https://mcuoneclipse.com/2021/06/05/position-independent-code-with-gcc-for-arm-cortex-m/

basprins, 2022-05-03:
Dear@keith-packard 
Googling your name is pretty impressive. Thanks for taking the time to reply here!. I am pretty sure you are right, and something else is the problem in my specific case, but I am running into wall after wall. So I don't mind experimenting with picolibc and try if it solves my weird problems. I'll give recompiling picolibc with -fpic a spin. 

Reading through the build docu, am I correct if all I need to adjust is

`
[properties]
c_args = [ '-nostdlib', '-msave-restore', '-fno-common', '-fpic' ]
`

To reply on your questions above:
- Yes, I update the got (indeed, moved to sram) for every address which is referring to flash. I do this in my startup code. When I delete the `libc_init_array` from my startup code, I can move my application to any location in flash and it runs a silly freertos task which blinks an led. (obviously any static c++ constructor would contain garbage now, since I omit `libc_init_array` as a test).
- When I "copy" the `libc_init_array` function from picolib into my startup.cpp file, it also works, now with static c++ constructors properly working too. Any other call to std lib will hard fault my app. 
- So it really looks like libc_nano refuses to be addressed through the .got. Which kind of made sense in my head, because if it wasn't compiled with -fpic to begin with, until you said you never had to bother with this, which gives me the feeling I am still missing something.
- Yes, I read the link you supplied. And many many many others. Still can't seem to find the "linking pin" :)

keith-packard, 2022-05-03:
> Reading through the build docu, am I correct if all I need to adjust is
> 
> `[properties] c_args = [ '-nostdlib', '-msave-restore', '-fno-common', '-fpic' ]`

Well, that's a whole meson adventure... It's probably best to add `'-fpic'` this in the the 'c' property in the binaries section so that it applies to both compiling and linking. Alternatively, you can include it in both c_args and c_link_args

> * When I "copy" the `libc_init_array` function from picolib into my startup.cpp file, it also works, now with static c++ constructors properly working too. Any other call to std lib will hard fault [...]

No, I've never bothered with this because I've never needed to relocate binaries on embedded systems -- I always compile them for a fixed target address. You'll certainly have to recompile the library with -fpic (or -fpie? I don't understand the differences).

>     * Yes, I read the link you supplied. And many many many others. Still can't seem to find the "linking pin" :)

I think you're going to need to debug this at the assembly level so you can figure out what it's doing wrong.

basprins, 2022-05-03:
Hi @keith-packard 
When I compile picolib with 

```
c = ['arm-none-eabi-gcc', '-nostdlib', '-fPIC', '-mpic-register=r9', '-msingle-pic-base', '-mno-pic-data-is-text-relative']
```

and disassemble the resulting libc.a with `arm-none-eabi-objdump --disassemble-all libc.a`, I don't see a .got table at all.

Are my added compiler arguments deleted somewhere in the meson chain? Or did I misunderstood something you suggested?

keith-packard, 2022-05-03:
It doesn't look like the GOT appears in the disassembly output; I checked by using objdump --syms where I see many references to _GLOBAL_OFFSET_TABLE_.

basprins, 2022-05-04:
Hi @keith-packard 
Not sure what the problem is for `printf`, it still hard faults. But it seems to be the only function which causes a problem. Since I don't need semihosting anyway (I wrote my own uart logger) I think I am good to go. I just retested stuff with a bunch of other libc function (`memset`, `memcpy`, `strtol`) and they work just fine when compiling with -fpic.

Still a lot I don't really understand on how the cpu works with the .got table, but frankly, I will settle for this now :). 

Thanks for your time and help! I'll close this issue and move on with picolib !

keith-packard, 2022-05-04:
Good luck! Let us know if you figure out the underlying issue.

basprins, 2022-05-16:
Hi @keith-packard ,
Finally made it through :). Just wanted to let you know, in the end, when all bugs related to linker symbols with/without LMA offset are resolved: everything works just fine. With picolib, with nxp's shipped arm toolchain, and with latest and greatest arm toolchain. 

It can just really be a pain resolving mistakes with addresses, when IDE is showing a linker symbols address with the "not .got resolved address" before you actually grab a copy of that address.

basprins, 2022-06-14:
I again ran into issues with libc-nano. I made a separate project so that I could easier test things. long story short: it's not possible to have a reliable relocatable app with arm's default shipped libc-nano library.

Recompiling picolibc with -fpic, solved my problems.

I shared some minor test results on my original question on stack:
https://stackoverflow.com/questions/72572067/position-independent-codfe-on-mcu-everything-works-except-vsnprintf-c-alter/72616907#72616907

So, we are moving forward with picolibc ! 

Thx again for the help!

closed 2022-06-14 by basprins