picolibc/picolibc #295: Default newlib elix mismatch with picolibc

DFOVIT, 2022-05-23:
I think that the meaning of the enable-newlib-elix-level option in picolibc differs from the one in newlib.
In newlib when enable-newlib-elix-level is not defined or set to 0 it will build the libc with level 4 elix functions (as seen here [newlib sourceware](https://sourceware.org/git?p=newlib-cygwin.git;a=blob;f=newlib/libc/stdio/Makefile.inc;h=2126ceaf591164a19c5cb1eb000e7b7b658462e7;hb=HEAD#l141)).
In picolib  when enable-newlib-elix-level is not defined, it will build the libc without elix level 4 functions ([picolibc](https://github.com/picolibc/picolibc/blob/main/newlib/libc/stdio/meson.build#L219))

keith-packard, 2022-05-23:
I hadn't ever really looked into what ELIX meant, or how it was supposed to work. It looks like it should just be set to 4 by default, and perhaps allow users to set it to a different level? Given how little of the library bothers to even look at the value, I'm tempted to just remove the configuration option and set it to 4?

keith-packard, 2022-05-23:
fwiw, the default value in picolibc appears to be '2'. Would just changing the default to '4' suffice?

keith-packard, 2022-05-23:
oh, and the _ELIX_LEVEL value is never set, so any code which depends on that will not see the specified value.

DFOVIT, 2022-05-23:
Yes it seems reasonable to set the default to 4.

keith-packard, 2022-05-24:
Now to fix the newlib code so that it actually builds with warnings enabled.

keith-packard, 2022-06-08:
This was fixed in #296

closed 2022-06-08 by keith-packard