picolibc/picolibc #298: CMake install staging error

jwinarske, 2022-06-06:
I'm hitting an install error when staging an install:

```
mkdir build && cd build
cmake .. -DTESTS=OFF -DCMAKE_TOOLCHAIN_FILE=`pwd`/../cmake/TC-arm-none-eabi.cmake -DCMAKE_STAGING_PREFIX=`pwd`/out/usr
make install -j
```

```
CMake Error at cmake_install.cmake:50 (file):
  file INSTALL cannot find "/b/picolibc/picolibc.specs": No such file or
  directory.

make: *** [Makefile:95: install] Error 1
[joel@fedora build]$ ls -la picolibc.specs 
-rw-r--r--. 1 joel joel 1448 Jun  6 07:54 picolibc.specs
[joel@fedora build]$ 
```

keith-packard, 2022-06-06:
Did you happen to test the most recent cmake fix, #297 ? With that, this works for me now.

jwinarske, 2022-06-06:
Moving to tip of tree it's fine.  I was on last tag.  Thanks!

closed 2022-06-06 by jwinarske