picolibc/picolibc #306: Support for RISC-V PLIC

ThyMYthOS, 2022-06-28:
Is it possible to use the picolibc crt0 to handle interrupts? I.e. will it signal SIG_INT? Or do we have to implement our own trap function?

ThyMYthOS, 2022-06-29:
OK, so I basically have to copy the crt0.c code, add support for interrupts and then link with `-nostdlib` to use my modified code, right?

keith-packard, 2022-06-29:
> OK, so I basically have to copy the crt0.c code, add support for interrupts and then link with `-nostdlib` to use my modified code, right? Or should I just overwrite the `mtvec` in `_init` with my own trap handler?

(use -nostartfiles instead of -nostdlib if you can; otherwise you'd have to include -lc -lgcc too).

The risc-v crt0.c code only installs a trap handler if you're using the semihosting version of crt0 -- that version installs a trap handler to catch application faults during testing and dump out the registers. If you use the 'regular' crt0, you would just install a trap handler in your own code, either in main() or in a constructor. That way, you could still take advantage of the rest of the crt0 code, including setting up $sp and $gp and initializing the data and bss areas.

> I already have some startup code from Microchip (formerly Microsemi) for their MI-V. Goal is to run FreeRTOS with picolibc on the MI-V. Seems that FreeRTOS installes it's own trap handler when `portasmHAS_CLINT = 1`.

Sounds like you'll be set then -- just use the default crt0 and let FreeRTOS set the trap handler.

ThyMYthOS, 2022-06-29:
Thank's alot!

closed 2022-06-29 by ThyMYthOS