picolibc/picolibc #307: picolibc.spec is not install location independent.

sgstreet, 2022-07-04:
When building picolibc using crosstool-ng 1.25.0 the attached specs file is produced.  The entries containing 

_-picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include/%*; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include_

are problematic as they force the an absolute location for the tool.  If the path /home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eab is remove executable using picolibc can not be linked.  Maybe a better idea would be to follow the newlib-nano.specs approach.

**picolibc.specs:**

%rename link	picolibc_link
%rename cpp	picolibc_cpp
%rename cc1	picolibc_cc1
%rename cc1plus	picolibc_cc1plus

*cpp:
-isystem %{-picolibc-prefix=*:%*/picolibc/include; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include/%*; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include} %(picolibc_cpp)

*cc1:
%{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1) 

*cc1plus:
-isystem %{-picolibc-prefix=*:%*/picolibc/include; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include/%*; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include} -idirafter %{-picolibc-prefix=*:%*/picolibc/include; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include/%*; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/include} %{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1plus)  

*link:
%{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfprintf=__f_vfprintf} %{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfscanf=__f_vfscanf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfprintf=__d_vfprintf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfscanf=__d_vfscanf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfprintf=__i_vfprintf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfscanf=__i_vfscanf} -L%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib/%M; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib/%*/%M; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib/%M} -L%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib/%*; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib} %{!T:-Tpicolibc.ld} %(picolibc_link) --gc-sections 

*lib:
--start-group %(libgcc)  -lc %{-oslib=*:-l%*} --end-group

*endfile:

*startfile:
%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib/%M/crt0%{-crt0=*:-%*%O;:%O}%s; -picolibc-buildtype=*:/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib/%*/%M/crt0%{-crt0=*:-%*%O;:%O}%s; :/home/stephen/Workspace/compilers/crosstool-ng/build/arm-none-eabi/../../local/arm-rtos-toolkit-eabi/arm-none-eabi/picolibc/arm-none-eabi/lib/%M/crt0%{-crt0=*:-%*%O;:%O}%s}

sgstreet, 2022-07-04:
This example appears to work:

**picolibc.specs**

%rename link	picolibc_link
%rename cpp	picolibc_cpp
%rename cc1	picolibc_cc1
%rename cc1plus	picolibc_cc1plus

*cpp:
-isystem %:getenv(GCC_EXEC_PREFIX ../../picolibc/include) %(picolibc_cpp)

*cc1:
%{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1) 

*cc1plus:
-isystem %:getenv(GCC_EXEC_PREFIX ../../picolibc/include) -idirafter %:getenv(GCC_EXEC_PREFIX ../../picolibc/include) %{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1plus)  

*link:
%{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfprintf=__f_vfprintf} %{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfscanf=__f_vfscanf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfprintf=__d_vfprintf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfscanf=__d_vfscanf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfprintf=__i_vfprintf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfscanf=__i_vfscanf} -L%:getenv(GCC_EXEC_PREFIX ../../picolibc/arm-none-eabi/lib/%M) -L%:getenv(GCC_EXEC_PREFIX ../../picolibc/arm-none-eabi/lib) %{!T:-Tpicolibc.ld} %(picolibc_link) --gc-sections 

*lib:
--start-group %(libgcc)  -lc %{-oslib=*:-l%*} --end-group

*endfile:

*startfile:
%:getenv(GCC_EXEC_PREFIX ../../picolibc/arm-none-eabi/lib/%M/crt0%{-crt0=*:-%*%O;:%O}%s)

keith-packard, 2022-07-04:
Yeah, ideally crosstool-ng would build with a 'systemroot' set -- that makes relocating the toolchain possible. As a fallback, you can specify a `--picolibc-prefix` parameter when building with GCC to change the top-level directory. That should work even with the crosstool-ng toolchain (if it has a recent enough picolibc version).

sgstreet, 2022-07-07:
Hacking the picolibc.specs to use the environment variable approach works.  Having the picolibc.specs use the picolib-prefix as follows does not (see below) does not.  I suspect that using --picolib-prefix support will still generate a issue due the lines like this from the _picolibc.specs.in_ file:

`
*cpp:
-isystem %{-picolibc-prefix=*:%*/@INCLUDEDIR@; -picolibc-buildtype=*:@PREFIX@/@INCLUDEDIR@/%*; :@PREFIX@/@INCLUDEDIR@} %(picolibc_cpp)
`

The _-picolibc-buildtype_ is causing the generated picolibc.specs file to contain references to the toolchain build path.

**picolibc.specs**

%rename link picolibc_link
%rename cpp	picolibc_cpp
%rename cc1	picolibc_cc1
%rename cc1plus	picolibc_cc1plus

*cpp:
-isystem %{-picolibc-prefix=*:%*/picolibc/include} %(picolibc_cpp)

*cc1:
%{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1) 

*cc1plus:
-isystem %{-picolibc-prefix=*:%*/picolibc/include} -idirafter %{-picolibc-prefix=*:%*/picolibc/include} %{!ftls-model:-ftls-model=local-exec} %(picolibc_cc1plus)  

*link:
%{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfprintf=__f_vfprintf} %{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfscanf=__f_vfscanf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfprintf=__d_vfprintf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfscanf=__d_vfscanf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfprintf=__i_vfprintf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfscanf=__i_vfscanf} -L%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib/%M} -L%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib} %{!T:-Tpicolibc.ld} %(picolibc_link) --gc-sections 

*lib:
--start-group %(libgcc)  -lc %{-oslib=*:-l%*} --end-group

*endfile:

*startfile:
%{-picolibc-prefix=*:%*/picolibc/arm-none-eabi/lib/%M/crt0%{-crt0=*:-%*%O;:%O}%s}

keith-packard, 2022-07-07:
I'm a bit confused -- if you set `--picolibc-prefix`, then `--picolibc-buildtype` shouldn't have any effect, it should only look for header files in the include subdirectory. Or is --picolibc-prefix doing the wrong thing for you? Maybe that should not use  `@INCLUDEDIR@` and just use `include`?

sgstreet, 2022-07-08:
Rebuilding the my tool chain with --picolibc-prefix.  For crosstool-ng, what would a good value be to generate a relocatable toolchain?  This?

`
--picolibc-prefix=../../picolibc/
`

keith-packard, 2022-07-09:
The --picolibc-prefix argument adds a pile of path elements to the include path (the whole @INCLUDEDIR@ value, which is picolibc/include when using crosstool-ng). I *think* you'd want to use `--picolibc-prefix=/some/path`, or whatever would make `/some/path/picolibc/include/stdio.h` work.

I think it's probably wrong to use `@INCLUDEDIR@` when using `--picolibc-prefix` as that seems likely to only lead to confusion as you need to know what `@INCLUDEDIR@` was set to during picolibc configuration.

You might track manually hacking the picolibc.specs file to reduce the `-picolibc-prefix=*:%*/picolibc/include;` clause to look like `-picolibc-prefix=*:%*/include` instead and see if that does what you expect.

Yes, it might be useful to have a mode where it used the GCC_EXEC_PREFIX value for crosstool-ng, but that's a pretty special case. Right now, it *does* have a mode for using the gcc sysroot mechanism, and I'd really like to figure out how crosstool-ng could use that as it makes things "just work" by having all paths in the system relative to the directory gcc was executed from.

sgstreet, 2022-07-09:
I just tried `%{-picolibc-prefix=*:%*/include}` and friends. Sigh, unfortunately it does not work.

sgstreet, 2022-07-09:
I did some research and experimentation on [GCC_EXEC_PREFIX](https://gcc.gnu.org/onlinedocs/gcc/Environment-Variables.html). As is typical of GCC documentation the behavior is not completely clear.  But the question is _what does GCC do when GCC_EXEC_PREFIX is not set?_ I played around with the GCC_EXEC_PREFIX like this:

`
GCC_EXEC_PREFIX=/tmp /path/to/compiler/arm-none-eabi-gcc -o hello-world hello-world.c
`

and as expected this fails to cc1 and friends.  While 

`
GCC_EXEC_PREFIX=/path/to/compiler/ /path/to/compiler/arm-none-eabi-gcc -o hello-world hello-world.c
`

functions as expected. Given this, I suspect/believe that the GCC compiler controller sets the variable GCC_EXEC_PREFIX to /path/to/compiler if it is not found the current environment.

So for crosstool-ng toolchains both nanolib and  picolibc are treated as "companion libraries" and reside at a well known location relative to the compiler executable. Namely `GCC_EXEC_PREFIX/../../picolibc`

Now of course nothing about GCC is ever clear, because the real world crosstool-ng path to arm-none-eabi-gcc is `/path/to/compiler/bin`  Which of course does not match the "well known location" above.  I suspect the actual GCC_EXEC_PREFIX is set to `/path/to/compiler/arm-none-eabi/bin` or  `/path/to/compiler/arm-none-eabi/lib` but I do not know.

Assuming the newlib-nano companion library is correct I looked at:

**/path/to/compiler/arm-none-eabi/lib/nano.specs**

%rename link	newlib_nano_link
%rename cpp	newlib_nano_cpp
%rename cc1plus	newlib_nano_cc1plus

*cpp:
-isystem %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-none-eabi/include) %(newlib_nano_cpp)

*cc1plus:
-idirafter %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-none-eabi/include) %(newlib_nano_cc1plus)

*link:
-L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-none-eabi/lib/%M) -L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-none-eabi/lib)

After some consideration, I think `/path/to/compiler/nano.specs` is a crosstool-ng generated companion library artifact and may be picolibc should do the same?

sgstreet, 2022-07-09:
Yep, I have been barking up the wrong tree.  The following is an excerpt from crosstool-ng: https://github.com/crosstool-ng/crosstool-ng/blob/master/scripts/build/companion_libs/350-newlib_nano.sh

```
    if [ "${CT_NEWLIB_NANO_INSTALL_IN_TARGET}" = "y" ]; then
        cat > "${CT_SYSROOT_DIR}/lib/nano.specs" <<EOF
%rename link                nano_link
%rename link_gcc_c_sequence nano_link_gcc_c_sequence
%rename cpp_unique_options  nano_cpp_unique_options
*cpp_unique_options:
-isystem %:getenv(GCC_EXEC_PREFIX ../../${CT_TARGET}/include/newlib-nano) %(nano_cpp_unique_options)
*nano_libc:
-lc_nano
*nano_libgloss:
%{specs=rdimon.specs:-lrdimon_nano} %{specs=nosys.specs:-lnosys}
*link_gcc_c_sequence:
%(nano_link_gcc_c_sequence) --start-group %G %(nano_libc) %(nano_libgloss) --end-group
*link:
%(nano_link) %:replace-outfile(-lc -lc_nano) %:replace-outfile(-lg -lg_nano) %:replace-outfile(-lm -lm_nano) %:replace-outfile(-lstdc++ -lstdc++_nano) %:replace-outfile(-lsupc++ -lsupc++_nano) %:replace-outfile(-lrdimon -lrdimon_nano)
*lib:
%{!shared:%{g*:-lg_nano} %{!p:%{!pg:-lc_nano}}%{p:-lc_p}%{pg:-lc_p}}
EOF
    else
        cat > "${CT_SYSROOT_DIR}/lib/nano.specs" <<EOF
%rename link	newlib_nano_link
%rename cpp	newlib_nano_cpp
%rename cc1plus	newlib_nano_cc1plus
*cpp:
-isystem %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/${CT_TARGET}/include) %(newlib_nano_cpp)
*cc1plus:
-idirafter %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/${CT_TARGET}/include) %(newlib_nano_cc1plus)
*link:
-L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/${CT_TARGET}/lib/%M) -L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/${CT_TARGET}/lib)
EOF
    fi
```

sgstreet, 2022-07-09:
Looks like the https://github.com/crosstool-ng/crosstool-ng/blob/master/scripts/build/companion_libs/340-picolibc.sh should be modified to use the newlib-nano approach to companion libraries.

sgstreet, 2022-07-10:
Ok, I got `--picolibc-prefix=/path/to/toolchain/install` to work.  The secret was to not include the picolibc directory in the path.

I should have done this earlier but attached are some verbose output from GCC.

The first one is from the as built by crosstoll-ng:
[gcc-verbose-no-picolibc-prefix.txt](https://github.com/picolibc/picolibc/files/9078137/gcc-verbose-no-picolibc-prefix.txt)

The next one add the `--picolibc-prefix` to point at the installed location of the toolchain.
[gcc-verbose-with-picolibc-prefix.txt](https://github.com/picolibc/picolibc/files/9078138/gcc-verbose-with-picolibc-prefix.txt)

And the last one has my modified `picolibc.specs` using the GCC_EXEC_PREFIX
[gcc-verbose-using-gcc-exec-prefix-specs.txt](https://github.com/picolibc/picolibc/files/9078139/gcc-verbose-using-gcc-exec-prefix-specs.txt)

IMHO my `picolibc.specs` file delivers on the promise of a relocatable toolchain because is does not require the additional `--picolibc-prefix` option and everything just works.

Ok, sorry I have beat this to death with multiple posts.  From your perspective what should be the next steps?

As an aside.  I was very excited to find the picolibc project and it's integration into crosstool-ng.  This is a great project and the `local-exec` TLS approach moves newlib is the correct direction.  I was already rebuilding the [arm gnu toolchain](https://developer.arm.com/Tools%20and%20Software/GNU%20Toolchain) to enable the correct TLS models to allow my custom cortex-m scheduler to leverage the `__aebi_read_tp` api in all of it's simplicity.  I could not figure out how to work around the numerous limitation of newlib/newlib-nano.  Even the small `struct __reent` was big.  Not to mention the stdio stuff.  Thanks for all the hard work on this project.

keith-packard, 2022-07-10:
> Ok, I got `--picolibc-prefix=/path/to/toolchain/install` to work. The secret was to not include the picolibc directory in the path.

Awesome. And, yes, this is 'expected', as crosstool-ng sets the meson 'includedir' value to picolibc/include. It does seem surprising to me as well, so I'm thinking I should replace @INCLUDEDIR@ in picolibc.specs.in with just 'include'.

> And the last one has my modified `picolibc.specs` using the GCC_EXEC_PREFIX [gcc-verbose-using-gcc-exec-prefix-specs.txt](https://github.com/picolibc/picolibc/files/9078139/gcc-verbose-using-gcc-exec-prefix-specs.txt)

Did you have to set the GCC_EXEC_PREFIX environment variable? Or does GCC somehow figure all of this out?

> IMHO my `picolibc.specs` file delivers on the promise of a relocatable toolchain because is does not require the additional `--picolibc-prefix` option and everything just works.

Agreed. I'd love to figure out how the sysroot stuff is supposed to work in GCC and have crosstool-ng use that -- picolibc has support for that, using the '%R' value in the specs file, and that works great, not requiring any setup by the user at all.

> Ok, sorry I have beat this to death with multiple posts. From your perspective what should be the next steps?

If the GCC_EXEC_PREFIX stuff works without having to set an environment variable, then that seems like an easy change to make in picolibc -- provide a way to specify the paths in picolibc.specs relative to the GCC_EXEC_PREFIX value. If you have to set the GCC_EXEC_PREFIX environment variable, then that still seems less than ideal, and figuring out how to get crosstool-ng to use the GCC sysroot stuff would be nice as that makes things "just work", and picolibc already knows how to use that.

>  Thanks for all the hard work on this project.

Glad you're finding it useful!

sgstreet, 2022-07-10:
> Did you have to set the GCC_EXEC_PREFIX environment variable? Or does GCC somehow figure all of this out?

I did not have to set GCC_EXEC_PREFIX, GCC just figured it out.

keith-packard, 2022-07-10:
That's awesome. And even better than then %R hack as that requires gcc to be built in a magic mode which also does 'other things'. I'll have to experiment and see what makes sense as a patch.

keith-packard, 2022-07-28:
I went and looked at the GCC code --turns out it sets GCC_EXEC_PREFIX in the environment if it wasn't already set. It all sort makes sense now.

keith-packard, 2022-08-22:
ok, so pr #330 should let you use --Dsysroot-install even if GCC isn't built with a sysroot. When that is the case, picolibc uses GCC_EXEC_PREFIX in picolibc.specs to locate the prefix for the GCC tree. Picolibc must be built with a matching prefix value. If not, the meson setup command will exit with an error indicating the correct value for `-Dprefix=`. Yes, it would be nice if it could just fix this, but meson doesn't appear to allow that.

closed 2022-08-22 by keith-packard

jnohlgard, 2022-11-18:
Just an FYI in case anyone tries to use a non-sysroot GCC with sysroot-install picolibc (kind of a weird combo):

`GCC_EXEC_PREFIX` is only set automatically by GCC if the toolchain has been moved from its original prefix location (`./configure --prefix=` setting when building GCC). If the `gcc` binary is still in its original location it will not export  the `GCC_EXEC_PREFIX` variable when calling `cc1` and others.

commit aa6a535f (2022-08-22): Generate specs files paths using GCC_EXEC_PREFIX for sysroot-install

When the -Dsysroot-install=true option is specified, and the GCC
toolchain hasn't been built with the sysroot option, we can still use the
GCC_EXEC_PREFIX environment variable as the GCC driver makes sure that is
set when scanning the specs file.

This allows picolibc to be installed relative to gcc and find the
necessary files using relative paths instead of hard-coding absolute
paths in the specs file.

Closes: #307
PR #330 (2022-08-22): Generate specs files paths using GCC_EXEC_PREFIX for sysroot-install
When the -Dsysroot-install=true option is specified, and the GCC
toolchain hasn't been built with the sysroot option, we can still use the
GCC_EXEC_PREFIX environment variable as the GCC driver makes sure that is
set when scanning the specs file.

This allows picolibc to be installed relative to gcc and find the
necessary files using relative paths instead of hard-coding absolute
paths in the specs file.

Closes #307