picolibc/picolibc #311: State of C++ support

LnnrtS, 2022-07-28:
I am trying to replace the newlib binaries shipped with `arm-none-eabi-gcc` with an embedded source build of picolibc.

I managed to resolve some issues with the help of comments in in this issue tracker but right now I am stuck with the same problem described [here](https://github.com/picolibc/picolibc/issues/299). In [another comment](https://github.com/picolibc/picolibc/issues/34#issuecomment-769406079) you (@keith-packard) mentioned that this will be resolved with gcc 11 but at least with the official gcc 11.2 from the ARM website, `#include_next` is still used in c++ headers that wrap c headers.

Since `libstdc++` is built against newlib for that particular toolchain, just replacing newlib with picolibc (as implied in #299) is only going to work if they are fully ABI compatible (which is not the case, right?)

Could you please summarize what scenarios/use cases are possible or will be possible in the near future regarding c++ support? Will there be simpler options than compiling a custom toolchain or also adding `libstdc++` as embedded source?

Thanks! :)

keith-packard, 2022-07-28:
The underlying issue is that the include file search path has the libstdc++ header directory followed by the newlib header directory without any way to insert the picolibc header directory between them. When `#include_next` is used by the libstdc++ headers, the compiler will always find the newlib headers.

Replacing `#include_next` with `#include` means that if you place the picolibc header directory ahead of the libstdc++ directory, then the picolibc files will be found.

Now you've got me wondering if we couldn't just add the libstdc++ header directory ahead of the picolibc directory -- that way you'd end up with something like this:

    libstdc++ ... picolibc ... libstdc++ ... newlib

That might work and not require changes to gcc. Of course, you still need to rebuild libstdc++ against picolibc so that any C library dependencies there work correctly (mostly streamio).

LnnrtS, 2022-07-28:
I see. From `libstdc++` view it makes sense to force include exactly what it was built against. That's why I am a bit sceptical if working against that is a good idea. I can imagine to run ABI problems if I change the compiler flags of the picolibc build.

Can you explain what changes you upstreamed to gcc?

Any suggestion on how to add libstdc++ as embedded source? I see they are using autotools as build system generator..

keith-packard, 2022-07-28:
That's a good point; libstdc++ has to compile against the right header files. However, while building libstdc++, it adds its own header directory using `-I`, so the picolibc headers (using `-isystem`) follow them and so `#include_next` works. It's only while using libstdc++, where the headers are found in the system include path built-in to gcc, that the picolibc files end up earlier in the include path than the newlib headers, and so `#include_next` fails.

The changes upstreamed to gcc provide an option while building libstdc++ to use only stdio interfaces instead of a liberal mix of stdio and posix interfaces. This makes the I/O modules in libstdc++ work on top of picolibc. With that, you can build libstdc++ using picolibc and everything works as expected. Of course, it's not easy to build just libstdc++ as it's included in GCC. Debian has a separate libstdc++ package for arm-none-eabi which makes it work, as does crosstool-ng. I've heard whispering about splitting libstdc++ out of the GCC tree. That would be awesome, but I can't really afford to wait, so we've got steaming piles of kludges around for now.

LnnrtS, 2022-07-29:
>That's a good point; libstdc++ has to compile against the right header files.

..and has to be built with the same compiler flags (or at least those that change the ABI which is often not obvious), right?

Have you tried building `libcxx` of llvm against picolibc? With a naive try I got `libcxx` 'almost' built from their repo against gcc. The cmake code and source looks pretty clean; certainly more attractive than `libstdc++`'s autotools

jnohlgard, 2022-07-29:
Following this thread to see what comes out.
I embed picolibc as a git submodule in my project (that uses C++23 and cmake) to avoid relying on random newlib versions from distro repos, but I have not yet found a way to build libcxx or libstdc++ standalone together with picolibc.

LnnrtS, 2022-07-29:
> I have not yet found a way to build libcxx or libstdc++ standalone together with picolibc.

What's the target platform and compiler you were using for that?

jnohlgard, 2022-07-29:
RISC-V (rv32imac) using some recent GCC version (https://github.com/riscv-collab/riscv-gnu-toolchain). I would like to avoid libstdc++ because of the tight coupling to a specific GCC version, so it looks like libcxx would be the best way forward for me.

keith-packard, 2022-07-29:
If you're building your own toolchain, then you could build libstdc++ using picolibc at the same time. Or, submit patches to that toolchain to do that.

keith-packard, 2022-08-23:
> RISC-V (rv32imac) using some recent GCC version (https://github.com/riscv-collab/riscv-gnu-toolchain). I would like to avoid libstdc++ because of the tight coupling to a specific GCC version, so it looks like libcxx would be the best way forward for me.

Have you looked at crosstool-ng? That should build a toolchain with recent GCC bits along with integrated picolibc and associated libstdc++ bits.

keith-packard, 2022-08-24:
Ok, I've submitted a PR to https://github.com/riscv-collab/riscv-gnu-toolchain which adds picolibc as an option. When selected, picolibc becomes the default C library so that the usual search paths 'just work' and C++ compiles out of the box. That required some minor patches to picolibc so that you can still use --specs=picolibc.specs if you want (they're still useful for the printf options and, until GCC takes some picolibc patches, for --oslib= support).

LnnrtS, 2022-08-24:
I managed to build `libcxx` against `picolibc` in a single project with the application (same compiler flags for everything) so for me the problem is resolved, too.

rdiez, 2023-06-11:
I have been building my own GCC toolchain with Newlib for many years, in order to use C++ on a rather constrained bare-metal target. This is the project which uses it:

https://github.com/rdiez/DebugDue

I have written a relatively straightforward makefile to build the toolchain. It is mentioned in the top-level README, and the makefile itself is located in the `Toolchain` subdirectory. There is also a small demo firmware which runs on an emulator (QEMU). All scripts are fully commented. It took me years to collect all necessary information and knowledge. By publishing that project, I am hoping that other people do not need to start from scratch (there are lots of hoops to jump through).

After a number of attempts over the last years, I just managed to build the toolchain and the demo firmware with Picolibc instead of Newlib without any manual workarounds. Specifying TARGET_LIBC=picolibc when running the makefile will do the trick.

pete-restall, 2023-11-14:
@rdiez - thank you, you've saved me a lot of headaches.  I always dread building cross-compilers, it's such a time sink.  And thank you to the picolibc team as well, it was just what I was looking for.

I've forked and adapted your DebugDue toolchain's Makefile to build a cross toolchain for MIPS rather than ARM (initially for PIC32MM microcontrollers) and I've just programmed a devboard with a blinky using both gcc and g++ (-std=c++23 !) that were built using picolibc.  I have no idea what bear traps might be laying in wait when I start building out but these simple examples certainly work a treat; less than 3KiB of flash.

Since Imagination sold out and stopped providing the pre-built MIPS toolchain and Microchip XC32 is woefully out of date, I'm hoping this combination provides me with all the niceness of the newer C++ features going forwards for my bare metal projects.

Again, thank you all for the effort and a couple of excellent projects.

rdiez, 2023-11-15:
Glad to help!

BlamKiwi, 2024-04-04:
I just want to add to this, we haven't had any issues with using picolibc with libc++. You may want to turn off some flags, such as threading support (no <pthread.h> shim available), but everything else seems to be fine. I ran into some small issues where <alloca.h> isn't referenced properly by some parts of LLVM (notably libunwind). 

Using libc++ means you can also build the standard library on demand in your build system, making it much easier to address ABI compatibility concerns.

closed 2026-11-10 by keith-packard