picolibc/picolibc #384: Are two linker scripts really needed?

jnohlgard, 2022-11-22:
The commit message on 91545564e7645b21ff93049fce4c74f1bfef6884 implies that discarding `.eh_frame` sections in pure C builds would reduce the memory footprint of the resulting binary. 
But since only C++ code with exceptions turned on should place anything in the exception handler sections, why would it affect the program size of C code where the `.gcc_except_table` section is empty?

rdiez, 2022-11-23:
I am also interested in this area. I suspect that you are right. I would have thought that you only need one linker script file for both C and C++ projects, and whether the C++ code uses exceptions at all should depend on compiler flags like _-fno-exceptions_.

I am guessing that the linker script files that Picolibc provides are mainly for running its test suite in a controlled, virtual environment, and should otherwise only be used as an example.

If you are targetting bare metal, you will need to write your own linker scripts. After all, the memory layout usually depends on your target microcontroller. On the other hand, if you are using a realtime OS, then that OS should provide a toolchain with the right linker scripts.

jnohlgard, 2022-11-23:
The meson.build script is incredibly convoluted around the specs file generation and removing the generation of another file would at least reduce some of that complexity.

arichardson, 2022-11-23:
I agree that having a single linker script would be nice and I think this should be achievable.

It appears that .eh_frame is emitted by default even for C code with recent compilers, but you can avoid emitting full unwind tables using compiler flags:

```
❯ gcc -shared /tmp/test.c -o /tmp/test.so && readelf -W --section-headers /tmp/test.so | grep .eh 
  [15] .eh_frame_hdr     PROGBITS        0000000000002008 002008 000024 00   A  0   0  4
  [16] .eh_frame         PROGBITS        0000000000002030 002030 00007c 00   A  0   0  8
❯ gcc -shared /tmp/test.c -o /tmp/test.so -fno-unwind-tables && readelf -W --section-headers /tmp/test.so | grep .eh
  [15] .eh_frame_hdr     PROGBITS        0000000000002008 002008 000024 00   A  0   0  4
  [16] .eh_frame         PROGBITS        0000000000002030 002030 00007c 00   A  0   0  8
❯ gcc -shared /tmp/test.c -o /tmp/test.so -fno-unwind-tables -fno-asynchronous-unwind-tables && readelf -W --section-headers /tmp/test.so | grep .eh
  [15] .eh_frame         PROGBITS        0000000000002008 002008 000004 00   A  0   0  8
```

I am not sure why the last command still emits a 4 byte .eh_frame, might need another flag.

keith-packard, 2022-11-23:
yeah, I saw eh_frame sections even in embedded systems for C code. Having example linker scripts which are suitable for embedded work seems useful, even if it makes the build scripts a bit twisty. I haven't checked this in a while though, perhaps something else has changed which makes the separation necessary?

rdiez, 2022-11-23:
There is some information on this subject here:

https://stackoverflow.com/questions/26300819/why-gcc-compiled-c-program-needs-eh-frame-section

The upshot is that, if you discard sections in your linker script, things are going to break. It is a matter of documenting what breaks and providing flags to not discard sections.

I wouldn't discard sections in this way for optimisation purposes. Instead, I would document what compiler flags reduce memory footprint. This way, the compiler can warn if a feature is not available with the given compiler flags.

I wonder if the linker would not discard most of the information in .eh_frame anyway if it is not really used anywhere.

keith-packard, 2022-11-24:
> The upshot is that, if you discard sections in your linker script, things are going to break. It is a matter of documenting what breaks and providing flags to not discard sections.

Picolibc needs to be built with .eh_frame sections enabled so that it can be used in C++ applications (and, I guess, some C applications?) where that information is required. The sample linker scripts show how to eliminate that (generally) unnecessary information for C applications.

> I wouldn't discard sections in this way for optimisation purposes. Instead, I would document what compiler flags reduce memory footprint. This way, the compiler can warn if a feature is not available with the given compiler flags.

The linker script is the only place these sections can be removed if picolibc is built for use with both C and C++ applications. Of course, if you're building picolibc for a specific application, then you could specify the correct compiler options to avoid including these sections, but at that point there's no functional difference between that and discarding them in the linker script.

> I wonder if the linker would not discard most of the information in .eh_frame anyway if it is not really used anywhere.

As far as I've been able to discover, you have to explicitly discard them in the script. Given the general nature of embedded applications, having the sample linker script demonstrate how to strip out information which is rarely required seems useful. Perhaps the documentation could be improved to note that in some cases that information might be necessary, even for C applications?

jnohlgard, 2022-11-24:
Does picolibc actually need to have unwind tables though? AFAIK the only reason you would need them in C code when building a c++ program is if you throw exceptions across picolibc functions. For example a C++ function calls a picolibc function which in turn calls a C++ function which may throw an exception. This could happen if you have a C function that takes a function pointer to use as a callback. Not sure if throwing exceptions like this would also be undefined behavior according to the C++ standards, but I would definitely see it as a code smell in a C++ project if exceptions are being thrown from a callback function or signal handler.

jnohlgard, 2022-11-24:
I found out that gcc on x86_64 has `-fasynchronous-unwind-tables` by default, while most other platforms don't. I have not yet encountered any picolibc objects that have a .eh_frame/.ARM.extab section when built for riscv or arm.

keith-packard, 2022-11-24:
>  This could happen if you have a C function that takes a function pointer to use as a callback. 

There are a few instances of this in the standard libc API, including qsort. Picolibc adds more with stdio. Interesting that you've found differences between various toolchains though; I would have thought that behavior would be reasonably consistent. I guess not.

Does anyone still think we should drop one or the other of the sample linker scripts? It seems like they both have uses, especially as examples to users writing their own.

rdiez, 2022-11-25:
If I understand the comments above correctly, Picolibc provides example linker script files which suggest unsafe optimisations. By unsafe I mean some things are going to break, and the user sees no warnings, either at compilation time, or in the example linker script as comments. Is that right? If so, that is no healthy state.

On the subject of unwind tables for plain C, or whatever information is in those sections, I thought that throwing C++ exceptions across C code is generally undefined behaviour. That is, it should be considered a programming error.

Or is there a list of C routines somewhere, like qsort(), which are nevertheless supposed to support C++ exceptions?

If Picolibc does not use any features which need such sections, then Picolibc should be built with the right compilation flags, so that those sections are not created or have no data inside. That is probably the only safe way to do it.

The trick is finding out which flags (if any) affect the ABI, which means that building Picolibc with those flags may prevent the user application from using some features, like backtrace(). The user should then decide whether Picolibc should be built with or without those flags.

jnohlgard, 2022-11-25:
Like I wrote above: when you build picolibc with the default flags there is no .eh_frame section or equivalent on any of the embedded platforms that I tested briefly. The only time I get them is on x86_64, which is the default behavior and recommended by people who know more about ABI and system design than I do. It is possible that aarch64-unknown-linux also have them but I didn't look. Regardless, both aarch64 and x86_64 can be expected to have enough memory that a few percent extra binary size is insignificant.

My recommendation is to only have one ld script, and to not discard anything by default in that script.

keith-packard, 2022-11-28:
A bit more information. The 'linux' toolchains used in picolibc CI all seem to enable unwind tables by default, presumably because that allows backtrace information to be presented (e.g. from the kernel itself). Also,the msp430 toolchain, https://www.ti.com/tool/MSP430-GCC-OPENSOURCE, actually provides a multilib configuration which includes `-fexceptions`. I'd love to know why they thought that would be useful, but it does mean that you get C libraries built both ways.

If, as @rdiez suggests, there's no need to include unwind tables within libc itself, would it be reasonable to include `-fno-asynchronous-unwind-tables` when building the library? With that, applications could also use the same option to avoid having `.eh_frame` information generated without needing to `/DISCARD/` it in the linker script, which seems like a neater solution than the current one. If that seems reasonable, then I'd be happy to include both changes in the sample linker script. Otherwise, it seems that we'd be depending on reasonable configuration of the C compiler to avoid unnecessary ROM usage in the target, and that gives me pause.

keith-packard, 2023-02-12:
I've been experimenting with changes to gcc itself to allow picolibc to be used as the 'system' C library. https://github.com/keith-packard/gcc/tree/riscv-picolibc-oslib has the bits. With that, it's clear to me that we don't in fact need two linker scripts, but we *do* need separate specs file fragments for c and c++ -- the c++ one needs to include crtbegin.o and crtend.o so that the `.eh_frame` tables get set up correctly. The cost of not doing that is modest; a couple of functions in crtbegin.o and a end marker for the `.eh_frame` section in crtend.o.

keith-packard, 2023-02-12:
> I've been experimenting with changes to gcc itself to allow picolibc to be used as the 'system' C library. https://github.com/keith-packard/gcc/tree/riscv-picolibc-oslib has the bits. With that, [...]

argh. it looks like libgcc.a is often built with exception information. I'd love to know why, but that's going to dump exception handling frame information into the output if we use the same linker script for C++ and C.

jnohlgard, 2023-02-12:
I also have been doing some work around using picolibc as the default system libc. Not made any PR yet, but I have a WIP branch for crosstool-ng with the required Kconfig changes here: https://github.com/jnohlgard/crosstool-ng/tree/picolibc-system-libc

A combined WIP branch with some other changes that are useful for a complete picolibc C++ toolchain is here: https://github.com/jnohlgard/crosstool-ng/tree/picolibc-toolchain 

My plan is to provide a toolchain with picolibc as system libc for arm and riscv targets as pre-built packages to ease adoption in embedded contexts, but I haven't had the time to finish that project yet.

jnohlgard, 2023-02-12:
(misclick close)

keith-packard, 2023-02-12:
> My plan is to provide a toolchain with picolibc as system libc for arm and riscv targets as pre-built packages to ease adoption in embedded contexts, but I haven't had the time to finish that project yet.

That sounds awesome. 

From my experiments with the riscv-gnu-toolchain, I *think* we need changes in gcc to make this work well. Gcc doesn't have picolibc support and I think it needs that. I'm working on a series of patches here: https://github.com/keith-packard/gcc/tree/riscv-picolibc-oslib and would like to get that finished up and merged upstream. I've posted a couple of earlier versions to the gcc-patches list and got a few comments, but my most recent version has not generated any progress.

The goal of that series is:

 1. Allow the default C library to be specified at configure time, instead of inferring it from the triplet.
 2. Add picolibc (and newlib) as possible C library choices -- it turns out gcc only knows about glibc, uclibc, bionic and musl.
 3. Add a custom config.h fragment for picolibc that sets up the --crt0 and --oslib options.
 4. Make it possible for g++ to have different spec contents than gcc. g++ can include crtbegin.o and crtend.o as well as use a different linker script (those are still provided by picolibc).

I think that once this basic set of features is available, we'll be able to keep the config.h fragment up to date with picolibc.

As crosstool-ng is happy to carry out-of-tree patches for gcc, it might make sense to focus efforts there and figure out how this should work based on what you're doing. If you want to review the gcc patches above and perhaps experiment with them in crosstool-ng, that would be really helpful.

Should this toolchain use a different triplet from arm-none-eabi or riscv64-unknown-elf? Neither of those appear to follow the "official" gnu triplet format, but then there doesn't seem to be much consistency there anyways. arm-picolibc-eabi? riscv64-picolibc-elf?

keith-packard, 2023-02-12:
ok, I gave your version of ct-ng a try and it worked great! I pointed it at a version of gcc with the ct-ng and my patches applied: https://github.com/keith-packard/gcc/tree/picolibc-crosstool-ng so that I could use the --oslib and --crt0 options and build the picolibc hello-world examples:

```
$ arm-picolibc-eabi-gcc hello-world.c -mcpu=cortex-m3 --oslib=semihost --crt0=semihost -Tarm.ld
$ ../scripts/run-thumbv7m a.out
hello, world
```

Because the target triplet matches *-picolibc-*, picolibc gets selected automatically as the default C library, but the patched version of gcc also accepts `--with-default-libc=picolibc`, which might be good to include in case the crosstool-ng user changes the `CT_TARGET_VENDOR` value.

closed 2026-11-10 by keith-packard