picolibc/picolibc #405: Interest in pthread shim and long double support?

BlamKiwi, 2022-12-12:
If this was implemented, would there be interest in actually having this included upstream? pthread and long double support is useful for accessing a greater variety of libraries, notably they are required for supporting libc++/libc++abi/libunwind on SMP targets without modifying upstream LLVM.

As far as long double support is concerned, libgcc and compiler-rt both support the primitive long double operations. You would just need to implement the <math.h> functions. Musl libc would be a candidate implementation to port as their license is compatible with picolibc (MIT). 

A <pthread.h> shim is a lot more difficult. A good chunk of it can be implemented without needing any kind of OS support. Things get a lot more complicated when you actually need to expose a replacement for SYS_futex and friends. A possible compromise is to provide a pthread implementation that can't actually "do threads".

BlamKiwi, 2022-12-12:
Oh! I've been so busy hacking away at 1.7.9 I didn't notice long double had been added. When you do think this code would be considered stable/ready?

As far as pthread goes, probably the best approach would be to leverage the fact that most RTOSes provide a Semaphore primitive. That's all that is needed to implement all the pthread synchronization primitives. Picolibc could expose a type erased interface for signal/wait. The tricky part is then thread creation/cancellation support. 

I will see if I can get a spinlock implementation for libunwind's RWMutex upstreamed. That will help improve picolibc's C++ support. I might not be able to put in a merge request until next year though.

keith-packard, 2022-12-12:
> Oh! I've been so busy hacking away at 1.7.9 I didn't notice long double had been added. When you do think this code would be considered stable/ready?

Yeah, I started with code from [openlibm](https://openlibm.org/), then fixed a bunch of bugs. I've got it passing the existing errno/exception tests and a pretty basic precision test that I added. It supports Intel 80-bit and IEEE 128 bit formats for everything other than bessel functions, and has limited support for IBM double-double code.

I'm planning on working on a new picolibc release in the next few weeks, part of that process will be getting the glibc test suite running for long double code.

> As far as pthread goes, probably the best approach would be to leverage the fact that most RTOSes provide a Semaphore primitive. That's all that is needed to implement all the pthread [...]

I guess I wonder if any RTOS would use a picolibc-based implementation, or if they'd prefer to use an internal one (as Zephyr is doing).

> I will see if I can get a spinlock implementation for libunwind's RWMutex upstreamed. That will help improve picolibc's C++ support. I might not be able to put in a merge request until next year though.

That would be cool -- picolibc already uses atomics in stdio to avoid needing locks for basic operations. Of course, it inherits newlib's mutex abstraction for a few places that require mutexes, but those are limited to buffered I/O in stdio and a bunch of locale code (which rarely ends up in embedded systems).

BlamKiwi, 2022-12-12:
`I guess I wonder if any RTOS would use a picolibc-based implementation, or if they'd prefer to use an internal one (as Zephyr is doing).`

At a minimum it might be helpful for when vendors only provide a FreeRTOS based BSP. FreeRTOS-pthread is both incomplete and not considered stable. Musl libc could provide a useful reference here again, they have a clean implementation of the pthread library with well defined syscall points.

Rust and C++20 also directly enable co-operative scheduling. When you pair them with picolibc, SMP support is the main thing that is missing out-of-the-box. This is the use-case I'm more interested in, although I understand it's pretty niche.

keith-packard, 2022-12-12:
> At a minimum it might be helpful for when vendors only provide a FreeRTOS based BSP. FreeRTOS-pthread is both incomplete and not considered stable. Musl libc could provide a useful reference here [...]

Ok, sounds like you've got a good use case (or two). And, yes, a well defined (and limited) syscall interface would be a great basis for a library implementation; using what Musl already uses would be helpful. FreeRTOS is also reasonably important to my current employer too.

> Rust and C++20 also directly enable co-operative scheduling. When you pair them with picolibc, SMP support is the main thing that is missing out-of-the-box. This is the use-case I'm more interested in, although I understand it's pretty niche.

If you can survive with mutexes, Picolibc already has an abstraction for those based on what newlib uses. And, of course, atomics.

Happy to help review code and figure out how to get this into the library.

BlamKiwi, 2024-06-30:
Closing this issue as we have internally moved on from needing libunwind support and a pthread shim is out of scope.

closed 2024-06-30 by BlamKiwi