picolibc/picolibc #409: Single-file-headerified version?

cnlohr, 2022-12-21:
I am really interested in this because it could add a simple libc for things like wasm, and on all sorts of processors, but, having to compile a libc for a target is particularly inconvenient.

I haven't seen this yet, so I'm asking if anyone here is aware of this.  Would be great! if it was in picolibc but if another libc that's ok too.

The rise of the single-file-headers (SFH) has made this much, much easier.  And in many situations I can easily compile to a binary blob.  It would be very helpful if there was a distilled version of this, maybe even a "minimal" version.  Which would just be selected functionality, maybe printf, sprintf, vsnprintf, rand, srand, qsort, some string functions, and maybe even a few math ones.

Additionally, if the function decorator on the SFH is made available, defining functions as `static`  enable the compiler to make significant optimizations that are otherwise very difficult to take. 

Primarily the important part to me is being able to use my own compiler tooling so I don't have to worry about using the tooling with my libc to build a libc.  Instead I can just `#include "picolibc.h"` and it's off to the races.

keith-packard, 2022-12-21:
I haven't heard of this in the context of a whole C library, but (as you say), perhaps some modest subset, or perhaps even a configurable subset, of the library could be concatenated together with some kind of scripting mechanism? I'm not sure it would be easy though; there are probably many assumptions about file scope separating various bits of the system. Note that the library also contains lots of architecture-specific code, so it's not like you could create a portable single-file library anyways.

jnohlgard, 2022-12-21:
@cnlohr If you want to experiment with this you can start by trying to make a unity build of the library. That will give you a starting point for fixing all of these assumptions that @keith-packard mentions. 
I expect there are a ton of places where file scoped variables have the same name as another variable in a different file and macros that are not meant to exist outside of their defined c file.

See https://mesonbuild.com/Unity-builds.html for some more info around unity builds.

cnlohr, 2023-01-18:
Hmm ... I am really interested in something that would be generic even if at a small cost to performance.  I.e. things that aren't architecture dependent, because I frequently find myself targeting very bizarre things/situations.  Is there any mechanism to gear as much as possible toward architecture-agostic code?

keith-packard, 2023-01-18:
There are architecture-independent implementations of almost everything in the library that you could use with this, with the notable exception of setjmp/longjmp. There are a few other things detected at meson runtime, including presence of 'long double' types, thread local storage support, and availability of a pile of compiler builtins. I'd suggest starting by creating a single .c file that #include's a few of picolibc files and see how much progress you make.

cnlohr, 2023-01-18:
Understood.

closed 2026-11-10 by keith-packard