picolibc/picolibc #417: error: use of unknown builtin '__builtin_isinfl' when compiling with clang 15 and cmake

theoparis, 2023-01-16:
```
clang-15: warning: optimization flag '-fsignaling-nans' is not supported [-Wignored-optimization-argument]
/usr/src/picolibc/newlib/libm/common/isinfl.c:37:10: error: use of unknown builtin '__builtin_isinfl' [-Wimplicit-function-declaration]
  return __builtin_isinfl(x);
```

Not sure what the problem is but I used the following cmake command:
```
cmake -B build -G Ninja -DCMAKE_BUILD_TYPE=MinSizeRel \
	-DCMAKE_EXE_LINKER_FLAGS="-fuse-ld=lld -nostdlib" \
	-DCMAKE_SYSROOT=/usr/src/sysroot \
	-DCMAKE_C_COMPILER=clang -DCMAKE_INSTALL_PREFIX=/usr/src/sysroot && \
	cmake --build build --parallel $(nproc) && \
	cmake --install build
```

I ended up patching picolibc by removing the references to the __builtin functions but I don't think its a good solution :joy:

keith-packard, 2023-01-16:
I bet the cmake build isn't working with clang. cmake is really for zephyr compatibility, so if you aren't using zephyr, you should be using the meson build system instead. Yeah, I know, another build system.

keith-packard, 2023-01-24:
Please see if these changes help out at all.

PR #420 (2023-01-24): cmake: Automatically detect compiler characteristics
The cmake stuff in picolibc was built to work with Zephyr using GCC. As such, all of the compiler-specific definitions were hard-coded, which makes them fail when building with clang (or anything but gcc).

This series makes all of those auto-detected, just like with the meson build. It's a bit kludgy, because cmake really doesn't want to run the compiler at configure time. It's a bit slow, but better than failing entirely.

This series adds tests using cmake + clang to make sure it all works.

Closes: #417