picolibc/picolibc #430: AGPL-licensed file can make using picolibc difficult

arichardson, 2023-02-15:
I would like to use picolibc, however the presence of a AGPL-licensed file makes it near impossible to do so due to licensing policy (even though I have no intention of using the  `scripts/GeneratePicolibcCrossFile.sh` script). I wonder if it would be possible to relicense it to GPL or move it outside of the main picolibc repository?

rdiez, 2023-02-15:
I am the author of that file and relicensing it is out of the question. But you can probably arrange a replacement with Keith.

arichardson, 2023-02-15:
@rdiez thanks for commenting here! In that case, I will see if using a forked repository with the file removed is acceptable instead.

fran-cap, 2023-02-18:
Would love to use this repo at work (its small and fast!) but worry is that the AGPL script file renders everything as AGPL.
If Keith can confirm the library isnt built with that script I think its ok but I'd like to get that in writing first.

keith-packard, 2023-02-18:
The script has nothing to do with building, testing or installing the library. It is strictly used to help people using picolibc write the meson cross compilation control files without having to learn how meson works or keep them up to date when things change. You can easily verify this by removing the file, and by checking for references within the source tree with 'git grep GeneratePicolibc'. 

Picolibc will never incorporate any non-BSD/MIT licensed material into code which is part of the library or used in the build process. Picolibc may include useful material with other licenses that provide optional features like this script, or which are used solely to test the library, like the printf test cases.

If you have competent legal advice which argues that the presence of these licenses in non-relevant files might affect the licensing of software using picolibc, I'd encourage you to have that written up and I'll have my lawyer review it. If they agree, then I'd remove the files from the repository. While I am a proponent of share-alike licenses, this project is not intended to try and used those to increase the availability of free software, rather the contrary. I spent considerable effort ridding the repository of all non BSD/MIT licensed source code (including getting permission to adjust some 4-clause BSD licenses to 3-clause) to make the license status of picolibc clear.

fran-cap, 2023-02-18:
Makes sense @keith-packard thanks for responding.

keith-packard, 2023-03-01:
I'm going to close this issue; please let me know if you have any remaining concerns.

closed 2023-03-01 by keith-packard