picolibc/picolibc #447: semihosting: open() does not take O_BINARY into account

rgrr, 2023-03-19:
subject says it all: although the corresponding modes are defined in semihost.h, open() ignores O_BINARY.

Simple fix could be:
```C
int
open(const char *pathname, int flags, ...)
{
...
if (flags & O_BINARY)
   semiflags |= 0x01;
```

If I should provide a PR, I will do on request.

keith-packard, 2023-03-19:
We'll need a PR in any case to run the patch through CI, so please submit one. Thanks for finding this gap in the code!

rgrr, 2023-03-20:
Hello Keith,
would it be ok to use something like the above (little bit nicer elaborated) and check correctness with static_assert()s?

keith-packard, 2023-03-20:
Yeah, seems fine to me -- there's no predefined constant for 'oh, and also do binary please'. Whatever seems good to you will probably be just fine.

rgrr, 2023-03-21:
Gosh, it's as always more complicated than on the first thought... O_BINARY/O_TEXT are only defined for Cygwin (thank you Microsoft for text mode!)^.

Now I'm wondering how to set the SH_OPEN_* file mode correctly on the target?  Is it **always binary** and just set differently in special Cygwin cases?  BUT: is picolib used under Cygwin anyway?

One of the problems with the mode is, that the other side (namely pyOCD) treats files differently depending on the SH_OPEN_* mode.

keith-packard, 2023-03-21:
Picolibc retains a pile of cygwin code because much of the library came from newlib, which is the C library for cygwin. However, it is never tested on cygwin and unlikely to work there at this point.

I'm not entirely sure how you would make this work -- picolibc expects the POSIX API to work according to POSIX, not the cygwin-ified not-quite-POSIX stuff. You could define a new open bit, hack up semihosting to look for it and hack up tinystdio to check for 'b' in the mode and mix that in. But, if that same code was then run on a system providing POSIX semantics, that code would see an undefined bit in the open mode.

Given that the semihosting code is mostly intended for debugging, I suspect it might be reasonable to just always open files in binary mode and expect the user to translate between text mode and binary mode outside of the debugging environment.

rgrr, 2023-03-21:
...
> Given that the semihosting code is mostly intended for debugging, I suspect it might be reasonable to just always open files in binary mode and expect the user to translate between text mode and binary mode outside of the debugging environment.

That makes life much easier, see PR #450

keith-packard, 2023-04-03:
All done now

closed 2023-04-03 by keith-packard