picolibc/picolibc #448: OpenBSD support (arm-none-eabi-gcc 7.4)

begriffs, 2023-03-20:
The latest stable OpenBSD version (7.2) uses an arm-none-eabi-gcc port with a fairly old old version of the software.

When I run `build-arm-none-eabi$ ../scripts/do-arm-configure` it raises an error about missing C18 support. Can the build settings be made less strict, or are C18 features actually needed to build picolibc?

```
Build started at 2023-03-20T15:05:26.076681
Main binary: /usr/local/bin/python3.9
Build Options: -Dincludedir=picolibc/arm-none-eabi/include -Dlibdir=picolibc/arm-none-eabi/lib -Dtests=true '--cross-file ../scripts/cross-arm-none-eabi.txt'
Python system: OpenBSD
The Meson build system
Version: 0.63.2
Source dir: /tmp/picolibc-1.8
Build dir: /tmp/picolibc-1.8/build-arm-none-eabi
Build type: cross build
Project name: picolibc
Project version: 1.8
-----
Detecting compiler via: arm-none-eabi-gcc -nostdlib --version
compiler returned <Popen: returncode: 0 args: ['arm-none-eabi-gcc', '-nostdlib', '--version']>
compiler stdout:
arm-none-eabi-gcc (Linaro GCC 7.4-2019.02) 7.4.1 20181213
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

compiler stderr:

Running command: arm-none-eabi-gcc -nostdlib -E -dM -
-----
Detecting linker via: arm-none-eabi-gcc -nostdlib -Wl,--version
linker returned <Popen: returncode: 0 args: ['arm-none-eabi-gcc', '-nostdlib', '-Wl,--version']>
linker stdout:
GNU ld (GNU Binutils) 2.31.1
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

linker stderr:
collect2 version 7.4.1 20181213
/usr/local/lib/gcc/arm-none-eabi/7.4.1/../../../../arm-none-eabi/bin/ld -X -L/usr/local/lib/gcc/arm-none-eabi/7.4.1 -L/usr/local/lib/gcc/arm-none-eabi/7.4.1/../../../../arm-none-eabi/lib --version

../meson.build:35:0: ERROR: Value "c18" (of type "string") for combo option "C language standard to use" is not one of the choices. Possible choices are (as string): "none", "c89", "c99", "c11", "gnu89", "gnu99", "gnu11".
```

keith-packard, 2023-03-21:
> When I run `build-arm-none-eabi$ ../scripts/do-arm-configure` it raises an error about missing C18 support. Can the build settings be made less strict, or are C18 features actually needed to build picolibc?

Nope, I've tested with c99, c11 and c18, picolibc is happy with any of those. I set the default to c18 in the meson.build file because that provides the best error detection environment during compilation.

begriffs, 2023-03-23:
Thanks! Adding this parameter fixed my build.

closed 2023-03-23 by begriffs