picolibc/picolibc #453: Is R_ARM_ABS32 linker warning cause for concern?

begriffs, 2023-04-01:
After I added `assert()` to my code, the linker produces this message:

```
/usr/local/lib/gcc/arm-none-eabi/7.4.1/../../../../arm-none-eabi/bin/ld:
  /usr/local/picolibc/arm-none-eabi/lib/thumb/v7e-m/fpv4-sp/hard/libc.a(libc_signal_signal.c.o)(.stab+0xde8): R_ARM_ABS32 used with TLS symbol _sig_func
```

Is this a potential problem?

keith-packard, 2023-04-01:
I don't know, but I suspect it's a bad sign. This precise usage is tested in picolibc CI with both gcc and clang though and it doesn't get that error. That makes me suspect that the compiler is doing something wrong. What toolchain are you using?

jnohlgard, 2023-04-01:
Is gcc 7.4.1 tested by CI though? It's pretty old.

begriffs, 2023-04-01:
Environment:
* arm-none-eabi-gcc (Linaro GCC 7.4-2019.02) 7.4.1 20181213
   * Built from [OpenBSD port](https://www.ports.to/path/devel/arm-none-eabi/gcc-linaro.html) with [patch](https://marc.info/?l=openbsd-ports&m=159897831109275&w=2) to enable `--with-multilib-list=rmprofile`
* Using CFLAGS: `-mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16`

The assert() does appear to work at runtime on my Nucleo STM32F411RET6, and prints its output via semihosting.

<details>
<summary>Backtrace from calling assert in main</summary>
<pre><code>#0  0x08003c80 in sys_semihost ()
#1  0x08003baa in sys_semihost_exit (exception=24, subcode=131107) at ../semihost/sys_exit.c:53
#2  0x08003b6e in _exit (code=134) at ../semihost/exit.c:51
#3  0x08003b8a in kill (pid=24, sig=131107) at ../semihost/kill.c:44
#4  0x08003abe in abort () at ../newlib/libc/stdlib/abort.c:62
#5  0x080023e8 in __assert_func (file=0x18 "\001$", line=131107, func=0x0, failedexpr=0x8004004 <__text_end+22> "main == NULL") at ../newlib/libc/stdlib/assert.c:66
#6  0x0800012a in main () at demo_blink_rtos.c:40
</code></pre>
</details>

<details>
<summary>Backtrace from calling assert in a FreeRTOS task</summary>
<pre><code>#0  0x08003c40 in sys_semihost ()
#1  0x08003b6a in sys_semihost_exit (exception=24, subcode=131107) at ../semihost/sys_exit.c:53
#2  0x08003b2e in _exit (code=134) at ../semihost/exit.c:51
#3  0x08003b4a in kill (pid=24, sig=131107) at ../semihost/kill.c:44
#4  0x08003a7e in abort () at ../newlib/libc/stdlib/abort.c:62
#5  0x080023e8 in __assert_func (file=0x18 "\001$", line=131107, func=0x0, failedexpr=0x8003fb0 <__text_end+2> "blink == NULL") at ../newlib/libc/stdlib/assert.c:66
#6  0x080000cc in blink (args=0x0) at demo_blink_rtos.c:21
#7  0x08001c38 in pxPortInitialiseStack (pxTopOfStack=0x0, pxCode=0xa5a5a5a5, pvParameters=0x80000cd <blink+44>) at freertos/portable/GCC/ARM_CM4F/port.c:216
</code></pre>
</details>

keith-packard, 2023-04-02:
> The assert() does appear to work at runtime on my Nucleo STM32F411RET6, and prints its output via semihosting.
> Backtrace from calling assert in main
> Backtrace from calling assert in a FreeRTOS task

Glad it works. I suspect the 'signal' function to select a custom abort handler might fail though. Let me add another test to check that under qemu -- that way you could run the test suite with your compiler if you want.

keith-packard, 2023-04-03:
I'm gonna mark this as "closed" now that the array of signal handlers is no longer a TLS variable.

closed 2023-04-03 by keith-packard