picolibc/picolibc #460: microblaze: CAS operation in `raise` is overwriting the signal handler with SIGDFL

alpsayin, 2023-04-16:
I've only observed this behaviour with `microblaze` but others might be affected as well.

https://github.com/picolibc/picolibc/blob/a7ac21f172a4de22b3ae850b8f9414ecb6dddeb6/newlib/libc/signal/signal.c#L142
`_sig_func_ptr func` variable is optimised into a register instead of allocated into a stack variable and then  `atomic_compare_exchange_strong` call fails and ends up writing`SIGDFL` into `func`

My proposed and tested solution is to declare `_sig_func_ptr func` volatile to keep it away from being optimised into registers.

But really, compiler should know better and seeing that func's address (&func) is referenced, it should not optimise it into a register, right ?

keith-packard, 2023-04-17:
> I've only observed this behaviour with `microblaze` but others might be affected as well.

Do you think you could cook up a test case to detect this and add it to `test/test-atomic.c`? That way we'd be able to see if it affects any of the architectures with CI testing enabled?

alpsayin, 2023-04-17:
Yes, yes I can. I'll get on it.

keith-packard, 2023-04-17:
oh, are you just saying that when a signal handler is invoked via `raise` that the signal is set back to `SIG_DFL`? I'm afraid that's how signals worked on Unix systems. Signal handlers often restore the desired signal handler as part of their implementation, sometimes using SIG_IGN while executing to avoid recursive signal handlers. It was (and still is) a bit of a mess. At least it's a well-known mess?

The C spec allows either the picolibc behavior *or* a more complicated behavior involving that pends signals until the handler returns and then delivers them at that point. I've taken the simplest approach allowed by the standard here.

alpsayin, 2023-04-18:
> oh, are you just saying that when a signal handler is invoked via raise that the signal is set back to SIG_DFL?
That tripped me for a bit but no, `func` is overwritten before it's called not the function entry in signal table.
But after a disassembly inspection, seems something more sinister is going on. 

This brings me to [this microblaze gcc patch](https://github.com/Xilinx/meta-xilinx/blob/master/meta-microblaze/recipes-devtools/gcc/gcc-12/0008-Patch-microblaze-Fix-atomic-side-effects.patch)
```
Fix atomic side effects. In atomic_compare_and_swapsi, add side effects to prevent incorrect assumptions
 during optimization. Previously, the outputs were considered unused; this  generated assembly code with undefined side effects after invocation of the atomic.
 ```
I couldn't locate the source for `atomic_compare_exchange`, I'd rather check the source first. But anyways, I'm thinking I should redirect my effort into those patches and come back to this later on if it's still around.

Below is the disassembly for raise. If it's too much See last 2 lines only, see `0x0000058c` where the jump address `r18` is set to zero, and that's where I end up jumping. And it's from top of the `main` again.
```
Dump of assembler code for function raise:
   0x0000051c <+0>:	addik	r3, r0, 31	// r3 = 31
   0x00000520 <+4>:	cmpu	r18, r5, r3 // r5 is arg1, this is probably the NSIG check
   0x00000524 <+8>:	bgeid	r18, 124	// 0x5a0 <raise+132>
   0x00000528 <+12>:	addik	r3, r0, 22	// r3 = 22 (EINVAL)
   0x0000052c <+16>:	imm	-28672
   0x00000530 <+20>:	swi	r3, r0, 684 // store r3 into errno
   0x00000534 <+24>:	rtsd	r15, 8
   0x00000538 <+28>:	addik	r3, r0, -1 // exec with above

   0x0000053c <+32>:	lwi	r4, r3, 0  
   0x00000540 <+36>:	xori	r6, r4, 1
   0x00000544 <+40>:	bneid	r6, 136	// 0x5cc <raise+176>
   0x00000548 <+44>:	lwi	r15, r1, 0
   0x0000054c <+48>:	addk	r3, r0, r0 // r3 =0 (return value = 0)
   0x00000550 <+52>:	rtsd	r15, 8
   0x00000554 <+56>:	addik	r1, r1, 28 // exec with above shrink stack

   0x00000558 <+60>:	addk	r7, r6, r0 // jump back here from below; zero r7
   0x0000055c <+64>:	addk	r8, r6, r0 // zero r8
   0x00000560 <+68>:	add	r6, r0, r0 // re-zero r6, odd but ok
   0x00000564 <+72>:	lwx	r7, r3, r0 // load from &_sig_func[sig]
   0x00000568 <+76>:	addic	r9, r0, 0 // r9 = lwx_success ? 1 : 0
   0x0000056c <+80>:	bnei	r9, -8	// retry exclusive load until success
   0x00000570 <+84>:	cmp	r9, r7, r4 // r9 = r7 - r4, comparison
   0x00000574 <+88>:	bnei	r9, 20 // branch if r4!=r7
   0x00000578 <+92>:	swx	r8, r3, r0 // if equal put r8 (zero) to &_sig_func[sig]
   0x0000057c <+96>:	addic	r9, r0, 0  // r9 = swx_success ? 1 : 0
   0x00000580 <+100>:	bnei	r9, -28	// branch back if r9 != 0; retry exclusive load & store (cas) until success
   0x00000584 <+104>:	addi	r6, r0, 1	// r6 = 1
   0x00000588 <+108>:	beqi	r6, -76 // branch further back if r6 == 0, but we wont because r6=1
   0x0000058c <+112>:	addik	r18, r0, 0 // zero r18
   0x00000590 <+116>:	brald	r15, r18  // call r18; WRONG! should've been r4. Where r18 came from no idea...
   0x00000594 <+120>:	or	r0, r0, r0
   0x00000598 <+124>:	brid	-76	// 0x54c <raise+48>

   0x0000059c <+128>:	lwi	r15, r1, 0
   0x000005a0 <+132>:	addk	r3, r5, r5
   0x000005a4 <+136>:	addk	r3, r3, r3 // r3*=4
   0x000005a8 <+140>:	imm	-28672
   0x000005ac <+144>:	addik	r3, r3, 172 // r3  = &_sig_func[r3]
   0x000005b0 <+148>:	lwi	r4, r3, 0 // r4 = sig_func[r3]
   0x000005b4 <+152>:	xori	r6, r4, 1 // check if r4 == SIG_IGN
   0x000005b8 <+156>:	bnei	r6, 12	// 0x5c4 <raise+168>
   0x000005bc <+160>:	rtsd	r15, 8
   0x000005c0 <+164>:	addk	r3, r0, r0 // exec with above return value = 0

   0x000005c4 <+168>:	addik	r1, r1, -28 // expand stack 7 words
   0x000005c8 <+172>:	swi	r15, r1, 0 // push return-address to stack
   0x000005cc <+176>:	bneid	r4, -116	// 0x558 <raise+60> // check if r4 == SIG_DFL
   0x000005d0 <+180>:	addk	r6, r0, r0 // exec with above zero r6
   0x000005d4 <+184>:	brlid	r15, 15380	// 0x41e8 <_exit>
   0x000005d8 <+188>:	addik	r5, r5, 128
   0x000005dc <+192>:	addk	r3, r7, r0
```

keith-packard, 2023-04-18:
> I couldn't locate the source for `atomic_compare_exchange`, I'd rather check the source first. But anyways, I'm thinking I should redirect my effort into those patches and come back to this later on if it's still around.

There isn't any "source" code for these operations; they're buried inside the compiler itself.

alpsayin, 2023-05-25:
Taking notes:
Applying the https://github.com/Xilinx/meta-xilinx/blob/master/meta-microblaze/recipes-devtools/gcc/gcc-12/0008-Patch-microblaze-Fix-atomic-side-effects.patch patch fixes this issue.
But as mentioned before, this will be applied with all the others.

keith-packard, 2023-05-25:
Thanks for the update. It sounds like there's no change proposed for picolibc then and that when the GCC patch is merged, the bug will be fixed. Feel free to leave this open until that happens to serve as a reminder if you like.

alpsayin, 2023-10-01:
Will be fixed with https://github.com/zephyrproject-rtos/gcc/pull/24

keith-packard, 2024-01-06:
That PR has been merged, so I'm going to close this issue.

closed 2024-01-06 by keith-packard