picolibc/picolibc #462: Create dummy libm.a?

xbjfk, 2023-04-25:
clang still tries to link with -lm by default, so maybe creating an empty libm.a would solve this? it is what musl libc does.

keith-packard, 2023-04-25:
picolibc should be creating a dummy libm already, at least with the meson build system. Are you trying to build a general toolchain with cmake? That's not well supported; cmake is only intended for use by Zephyr when building the library for a specific application.

xbjfk, 2023-04-25:
My bad, yeah I was using CMake.

keith-packard, 2023-04-25:
I'll close this issue then; maybe we need better documentation about what the CMake build system is useful for...

closed 2023-04-25 by keith-packard

ekliptik, 2024-01-02:
Hi Keith, where was this completed? I just integrated picolibc into an experimental toolchain with CMake, and we already use it) and didn't expect CMake to be given up on. I chose CMake because after much effort I found out you don't need to generate a cross file for each library configuration.

keith-packard, 2024-01-02:
> Hi Keith, where was this completed? I just integrated picolibc into an experimental toolchain with CMake, and we already use it) and didn't expect CMake to be given up on. I chose CMake because after much effort I found out you don't need to generate a cross file for each library configuration.

The cmake stuff really is just for building the library as a Zephyr module; it's not designed for general use. In particular, there are numerous configuration options simply unavailable in the cmake bits as Zephyr never needs them.

There's a script (GeneratePicolibcCrossFile.sh) provided in the repository that creates a suitable cross file if that helps you use meson.

ekliptik, 2024-01-02:
Ah, I saw this script before, but then I realized I can use CMake. My question was, where is it documented that CMake is only for legacy use cases? I missed it when I started playing with picolib and still can't find it. For now, I'll just `ar -rcs libm.a`

keith-packard, 2024-01-02:
> Ah, I saw this script before, but then I realized I can use CMake. My question was, where is it documented that CMake is only for legacy use cases? I missed it when I started playing with picolib and still can't find it. For now, I'll just `ar -rcs libm.a`

The 'building picolibc' doc only talks about meson? I guess it should mention that there are cmake bits but they are only intended for use with Zephyr?