picolibc/picolibc #467: Broken fma/fmaf functions when compiling with Soft Float + Clang.

BlamKiwi, 2023-05-17:
I've encountered some broken code generation in the fma/fmaf functions when compiling with ARMv8 "soft float". Clang will generate a call to the builtin function as an "optimization",  breaking the library. The issue persists even if I supply `-fno-builtin` or `-fno-builtin-fma` and friends. There may be other functions, but these are the only ones I've encountered.

```
float fma(float a, float b, float c) {
    return (a * b) + c;
}
```
... becomes ...
```
float fma(float a, float b, float c) {
    return fma(a, b, c);
}
``

I've hacked together a quick solution by manually calling `__addsf3` etc. This is obviously not a good fix to consider upstreaming. 

- Would you consider this to be a bug in Clang or the picolibc build system? My understanding is that `-fno-builtin` should disable all functions that don't start with `__builtin_`, implying it's a Clang bug.
- Are you able to suggest a more upstream worthy solution than manually calling the soft float routines?

BlamKiwi, 2023-05-17:
Digging through compiler documentation, Clang and GCC enable floating point contraction for FMA statements. So I was wrong in assuming `-fno-builtin` would do anything here. The appropriate fix is specifying is `-ffp-contract=off`, which disables the compiler folding FMA statements.  

Would you be interested in a merge request that just fixes `fma`/`fmaf`, as this may be broken for other target/compiler combinations. We could supply `-ffp-contract=off` to the build system, but this would pessimise the entire code base.

keith-packard, 2023-05-17:
Yes, happy to merge fixes that apply those options to just the affected files. Take a look at newlib/libc/stdlib/meson.build to see how to apply C flags for a subset of the files in a directory. It involves creating a separate static library and then merging that into the main library. Cmake does this better...

keith-packard, 2023-05-23:
I'm trying to reproduce this issue locally without any success. I'm using clang version 14.0.6. The fix I dug up is to add
this to all of the fma implementations:
```
#ifdef __clang__
#pragma STDC FP_CONTRACT OFF
#endif
```
But, if I can't verify that this does anything, I'm hesitant to give it a try.

BlamKiwi, 2023-05-24:
> Can you provide your complete build setup?

https://godbolt.org/z/5zhsWbabv

That pragma does indeed fix it.

keith-packard, 2023-05-24:
With the options you provided, picolibc doesn't build -- it doesn't support aarch64 without using the FPU as it has a bunch of inline assembly functions for various math functions and no conditional compilation control to disable them. All of which means I cannot test this fix using picolibc source and will just have to trust that the standalone testing I've done will work.

BlamKiwi, 2023-05-24:
> With the options you provided, picolibc doesn't build -- it doesn't support aarch64 without using the FPU as it has a bunch of inline assembly functions for various math functions and no conditional [...]

Yeah we disable ASM requiring NEON/VFP. We have to do the same with the LLVM runtime libraries as well. 

ARMv8/ARMv9 without hard float support is niche hardware, I don't expect picolibc or any other mainstream library to explicitly support it.

keith-packard, 2023-05-24:
> ARMv8/ARMv9 without hard float support is niche hardware, I don't expect picolibc or any other mainstream library to explicitly support it.

I'm happy to add support for it in picolibc; I'd just want to include suitable tests to CI. Looks like this can be detected by looking at `__ARM_NEON` and `__ARM_NEON_FP`?

closed 2023-05-24 by keith-packard

BlamKiwi, 2023-05-26:
> I'm happy to add support for it in picolibc; I'd just want to include suitable tests to CI. Looks like this can be detected by looking at `__ARM_NEON` and `__ARM_NEON_FP`?

Detecting it is easy enough, the main problem is how different compilers treat soft-float for ARM64. 

GCC/ArmClang treat any usage of float/double/long double as a hard error since there's no standardABI.  You would need to add those defines around quite a few functions and headers.  Clang here is the easy one, since they just emit the soft float routines.

keith-packard, 2023-05-26:
Yeah, I managed to get clang to emit all of the relevant code and build picolibc just fine. However, I don't have a soft float library to test with; that's usually libgcc, but, uh, I can't get that to build because of gcc. Is there an alternate soft float library we can use for testing?

keith-packard, 2023-05-26:
btw, the relevant picolibc changes are present on  https://github.com/picolibc/picolibc/tree/aarch64-soft-float

BlamKiwi, 2023-05-26:
We use LLVM's compiler-rt.  LLVM has a libgcc drop-in using compiler-rt as well, although I haven't tested it.  https://compiler-rt.llvm.org/

The below should build just the builtins with the correct settings. The important flags are `LLVM_*` and `COMPILER_RT_*`. The rest are just usual CMake toolchain flags.  

You may need to play with the install paths, by default Clang looks in `${INSTALL_PREFIX}/lib/clang-runtimes/${TRIPLE}` when looking for runtime libraries.

```
ARCH_FLAGS="-march=armv8-a+nofp+nosimd -mgeneral-regs-only -ffreestanding -nostdlib -U__ARM_FP"
TRIPLE=aarch64-none-elf
INSTALL_PREFIX=/usr/local

cmake <PATH_TO_SRC>/llvm-project/runtimes -G Ninja \
    -DCMAKE_INSTALL_PREFIX=${INSTALL_PREFIX} \
    -DCMAKE_SYSROOT=${INSTALL_PREFIX} \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DCMAKE_ASM_COMPILER=clang \
    -DCMAKE_ASM_COMPILER_TARGET=${TRIPLE} \
    -DCMAKE_ASM_FLAGS="${ARCH_FLAGS}" \
    -DCMAKE_C_COMPILER_EXTERNAL_TOOLCHAIN=${INSTALL_PREFIX} \
    -DCMAKE_C_COMPILER=clang \
    -DCMAKE_C_COMPILER_TARGET=${TRIPLE} \
    -DCMAKE_C_FLAGS="${ARCH_FLAGS}" \
    -DCMAKE_CXX_COMPILER_EXTERNAL_TOOLCHAIN=${INSTALL_PREFIX} \
    -DCMAKE_CXX_COMPILER=clang++ \
    -DCMAKE_CXX_COMPILER_TARGET=${TRIPLE} \
    -DCMAKE_CXX_FLAGS="${ARCH_FLAGS}" \
    -DCMAKE_AR=llvm-ar \
    -DCMAKE_NM=llvm-nm \
    -DCMAKE_RANLIB=llvm-ranlib \
    -DLLVM_ENABLE_RUNTIMES='compiler-rt' \
    -DLLVM_DEFAULT_TARGET_TRIPLE=${TRIPLE} \
    -DCOMPILER_RT_BUILD_BUILTINS=ON \
    -DCOMPILER_RT_BUILD_CRT=OFF \
    -DCOMPILER_RT_CRT_USE_EH_FRAME_REGISTRY=OFF \
    -DCOMPILER_RT_BUILD_SANITIZERS=OFF \
    -DCOMPILER_RT_BUILD_XRAY=OFF \
    -DCOMPILER_RT_BUILD_LIBFUZZER=OFF \
    -DCOMPILER_RT_BUILD_PROFILE=OFF \
    -DCOMPILER_RT_BUILD_MEMPROF=OFF \
    -DCOMPILER_RT_BUILD_ORC=OFF \
    -DCOMPILER_RT_BUILD_GWP_ASAN=OFF \
    -DCOMPILER_RT_BAREMETAL_BUILD=ON \
    -DCOMPILER_RT_DEFAULT_TARGET_ONLY=ON \
    -DCOMPILER_RT_USE_BUILTINS_LIBRARY=ON 
```

keith-packard, 2023-05-26:
Thanks, I got it working and picolibc is running. I've got a few more fixes to clean up fenv and setjmp/longjmp. However, varargs doesn't seem to be working.

keith-packard, 2023-05-26:
Yeah, take a look at AArch64ABIInfo::EmitAAPCSVAArg in https://github.com/llvm/llvm-project/blob/main/clang/lib/CodeGen/TargetInfo.cpp -- that function is only looking at the type of the argument to pick between the fp and integer chunks of saved register storage, not whether the fp registers are even in use.

BlamKiwi, 2023-05-27:
Are you able to produce a minimum example of the issue you're observing?  [I can't get a toy example to produce obviously wrong code](https://godbolt.org/z/Ej8ohxn63).

keith-packard, 2023-05-27:
It's not the caller, it's the callee -- the caller doesn't do anything special with the args, it's the callee that saves all of the registers to the stack and then walks three lists -- the saved integer registers, the saved float registers and the original stack. In this case, the double value is passed in x1, which is then saved to the integer register block. However, when fetching the double value, it is pulled from the float register block instead, which hasn't been initialized at all.

https://godbolt.org/z/37hqd1Evb

Take a look at this doc: https://student.cs.uwaterloo.ca/~cs452/docs/rpi4b/aapcs64.pdf -- the Variable argument lists appendix shows how this is supposed to work. The va_list is at `sp + #64`, and you'll see the code store the original sp (via x10) at that address and store the base of the integer register block at `sp + #72` (via x9). However, when it fetches the double value, it loads the address of the float register block (`sp + #80`), carefully increments it by 8 and saves it back there.

BlamKiwi, 2023-05-27:
Oh yeah, I see it. We never hit this edge case as we use C++ var args for our logging system.

I will submit an issue to the LLVM devs, it looks like they propagate some of the `-mgeneral-regs-only` information back to codegen.

keith-packard, 2023-05-27:
Sounds good. When that fix is available, we can finish the rest of the aarch64 soft float bits here; I'm hesitant to merge code that can't run through the picolibc test suite.

BlamKiwi, 2024-04-04:
> Sounds good. When that fix is available, we can finish the rest of the aarch64 soft float bits here; I'm hesitant to merge code that can't run through the picolibc test suite.

See the linked LLVM issue. There is a proper soft-float ABI now for ARM64.

keith-packard, 2024-04-04:
Sounds like we should plan on enabling soft float bits for aarch64 using an updated version of clang. If you want to work on that, feel free to start with the patches from https://github.com/picolibc/picolibc/commits/aarch64-soft-float/. Any idea when those will land in a clang release?

BlamKiwi, 2024-04-05:
It doesn't appear to be in the release-18 branch, so I would assume LLVM19 for now. 

We periodically update our toolchain to follow LLVM releases, it's likely we will pull this into one of our sprints at some point. Otherwise it will be when I have a weekend free.

commit 26c9cde2 (2023-05-24): libm/common: Disable FP contractions on clang for fma/fmaf/fmal

Clang enables conversion of 'a*b+c' into an FMA operation by default,
which compiles the fallback implementation of fma and fmaf into
infinite loops. Add '#pragma STDC FP_CONTRACT OFF' when building with clang
to disable this behavior.

Closes: #467
PR #475 (2023-05-24): libm/common: Disable FP contractions on clang for fma/fmaf/fmal
Clang enables conversion of 'a*b+c' into an FMA operation by default, which compiles the fallback implementation of fma and fmaf into infinite loops. Add '#pragma STDC FP_CONTRACT OFF' when building with clang to disable this behavior.

Closes: #467