picolibc/picolibc #486: Build error: implicit declaration of function '__posix_sflags'

rdiez, 2023-06-10:
I am building Picolibc with posix-io=false, and I am getting the following compilation error:

```
../picolibc-1.8.2/newlib/libc/tinystdio/freopen.c:53:23: error: implicit declaration of function '__posix_sflags' [-Werror=implicit-function-declaration]
   53 |         stdio_flags = __posix_sflags(mode, &open_flags);
      |                       ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

I think the reason is that, in file newlib/libc/tinystdio/meson.build , array srcs_tinystdio_posix does not contain freopen.c , so that it gets built even if POSIX I/O is disabled. This is a direct link to that array:

https://github.com/picolibc/picolibc/blob/0cbe978cf49f4dab363ffcdb4196f2a05e41abf5/newlib/libc/tinystdio/meson.build#LL185C25-L185C25

Shouldn't other files be excluded in the same way too? For example, fread.c .

keith-packard, 2023-06-10:
In this case, yes. However, it looks like there's some unintended dependencies between the POSIX and non-POSIX APIs here. `-Dposix-io=false` is intended to avoid adding any dependency on an underlying POSIX syscall API (open/close/read/write). That should include fopen, fdopen and freopen, but not things like fread and fmemopen which depend solely on internal interfaces. `__posix_sflags` used to only be needed by fopen, fdopen and freopen. With the addition of fmemopen, a portion of the functionality is also needed there. Thanks for pointing out the problem; time to go hack up the code to deal with this.

rdiez, 2023-06-10:
OK, thanks for the explanation.

I would make one of the CI builds exercise this flag. I am guessting that most deeply-embedded targets (bare metal) would not use such POSIX file-oriented operations.

keith-packard, 2023-06-10:
Good plan for CI -- I've added this flag to the smaller build configuration.

keith-packard, 2023-06-10:
Note that enabling posix-io doesn't change any of the rest of the API, it only adds fdopen, fopen, freopen and fclose. If you don't use those, you'll never end up depending on a POSIX API. But, it's useful to ensure that you don't accidentally need something that uses those APIs ...

commit 8331001a (2023-06-11): libc/tinystdio: Fix building with -Dposix-io=false

Two new functions that take a mode string were added, freopen and
fmemopen. The former relies on POSIX I/O and so needed to only be
built when posix-io=true. The latter still needs a way to convert the
open mode string to the internal FILE operation flags. Do this by
creating a new API, __stdio_sflags, which sits atop __posix_sflags
when posix-io=true, but which directly uses a modified version of the
__posix_sflags implementation when posix-io=false.

Closes: #486.
PR #487 (2023-06-11): libc/tinystdio: Fix building with -Dposix-io=false
Two new functions that take a mode string were added, freopen and fmemopen. The former relies on POSIX I/O and so needed to only be built when posix-io=true. The latter still needs a way to convert the open mode string to the internal FILE operation flags. Do this by creating a new API, __stdio_sflags, which sits atop __posix_sflags when posix-io=true, but which directly uses a modified version of the __posix_sflags implementation when posix-io=false.

Closes: #486.
closed 2023-06-11 by keith-packard