picolibc/picolibc #488: FALLTHROUGH collision

rdiez, 2023-06-18:
I have a codebase which defines its own FALLTHROUGH macro, and when trying to switch to Picolibc, there is a collision.

Could Picolibc make its FALLTHROUGH private? It's defined here:

https://github.com/picolibc/picolibc/blob/bf24ef1840270038064a679c0d6425aece1aba17/newlib/libc/include/sys/cdefs.h#L774

Or at least use a different name like PICOLIBC_FALLTHROUGH.

keith-packard, 2023-06-18:
It's a bit unfortunate, but picolibc doesn't have a header file which is included by every C file but not installed with the library. A couple of options:

 1. Rename this to `__PICOLIBC_FALLTHROUGH`.
 2. Add `#ifdef _LIBC` to gate whether it is visible (picolibc adds -D_LIBC when building the library).
 3. Create a suitable globally-included header file and include that everywhere.

Any preference?

rdiez, 2023-06-18:
I do not really mind.

I personally would implement both (1) and (2). I like the idea of some definitions being "private" to Picolibc, so that would be (2) or (3). But including global header file all over the place (3) may be a pain to maintain.

Someone may decide to compile the Picolibc source files directly in his/her project, bypassing the Picolibc build system. In such a scenario, having a separate "namespace" (with a __PICOLIBC_ prefix) may help. That would be option (1).

Furthermore, if Picolibc decides to provide an inline implementation in a header file of some routine which uses FALLTHROUGH, it may not be possible to gate it with #ifdef _LIBC.

So I guess at least (1), and maybe (2) for now.

But I do not really mind. This does not look like a critical decision.

keith-packard, 2023-06-18:
Thanks for thinking this through -- (1) seems like a sufficient fix for now as that leaves the ability to create inline functions using this macro.

rdiez, 2023-06-18:
That's great, thanks.

commit c8f93e81 (2023-06-18): cdefs.h: Use __PICOLIBC_FALLTHROUGH instead of FALLTHROUGH

Avoid placing an internal symbol in a globally visible header file.

Closes: #488.
PR #489 (2023-06-18): cdefs.h: Use __PICOLIBC_FALLTHROUGH instead of FALLTHROUGH
Avoid placing an internal symbol in a globally visible header file.

Closes: #488.
closed 2023-06-18 by keith-packard