picolibc/picolibc #490: Taming atexit

rdiez, 2023-06-18:
In tight, bare-metal embedded environments, dynamic memory allocations can be problematic, especially in interrupt context, and sometimes RAM consumption in general must be kept to a minimum. Furthermore, initialisation and destruction can also be a source of trouble.

That is why atexit has been a pain for me, especially because global, static C++ objects tend to automatically register destructors with atexit. I usually implement some sort of 'shutdown' or 'reset' function in my embedded firmware, where I check that no memory has been leaked, and the atexit mechanism makes this difficult.

I think it is possible to disable atexit altogether, but I think that doing that discards (hides) the atexit registration, which masks potential problems.

The solution I have been using with Newlib is to check on initialisation and on termination that no atexit entries have been registered at all. See for example here:

https://github.com/rdiez/DebugDue/blob/6b19483195cad25404c2f53a467a480f2e91cea5/Project/src/BareMetalSupport/BoardInitUtils.cpp#LL163C6-L163C6

If no atexit entries are ever registered, the linker normally discards even _atexit0, which further reduces RAM usage.

Alas, Picolibc allows no such check, as far as I can see. Header newlib/libc/stdlib/atexit.h is not accessible. I tried declaring the pointer in my source code like this:

  extern struct _atexit *_atexit;

But that does not work, the linker still cannot find it:

  undefined reference to `_atexit'

Is there a way I can check that no atexit routines have been registered at all?

jnohlgard, 2023-06-18:
Your linker error might be solved if you declare it as a weak symbol. Then it will be nullptr if the symbol was not linked in your final executable. I didn't test your specific code but it looks like that would be similar to how your newlib check works.

keith-packard, 2023-06-18:
picolibc is generally built with a smaller implementation of these functions: https://github.com/picolibc/picolibc/blob/main/newlib/libc/stdlib/pico-onexit.c -- this code doesn't allocate the list of functions dynamically as POSIX permits it to provide a fixed number of slots (32). https://pubs.opengroup.org/onlinepubs/9699919799/functions/atexit.html The array holding those pointers is static to the implementation to avoid exposing yet another C library symbol to applications.

What do you want to detect here?

 1. whether atexit/on_exit are linked into the application? A script run after linking would be easiest.
 2. whether atexit/on_exit are ever called? Writing your own version of these that panic'd seems like it would work.

There's also `__cxa_atexit` that you might be seeing in some code? I think that's related to shared libraries?

keith-packard, 2023-06-19:
> Your linker error might be solved if you declare it as a weak symbol. Then it will be nullptr if the symbol was not linked in your final executable. I didn't test your specific code but it looks like that would be similar to how your newlib check works.

yeah, there are a bunch of weak symbol tricks around these code paths to avoid pulling in code in `exit()`. But, I don't think there's anything related to this particular symbol.

rdiez, 2023-06-19:
First of all, thanks for the extra information about pico-onexit.c.

I want to prevent myself from inadvertently adding code which that allocates dynamic memory and/or registers a global destructor I am not expecting.

This can happen by using a new C or C++ library, by defining a new global static C++ object, or by adding a destructor to a C++ class which has global static instances (like singletons).

__cxa_atexit is related to C++, see here for more information:

https://bugs.launchpad.net/gcc-arm-embedded/+bug/1586114

https://stackoverflow.com/questions/42912038/what-is-the-difference-between-cxa-atexit-and-atexit

Calls to __cxa_atexit will land directly or indirectly in atexit slots:

```
-fuse-cxa-atexit
Register destructors for objects with static storage duration with the __cxa_
atexit function rather than the atexit function. This option is required for
fully standards-compliant handling of static destructors, but only works if your
C library supports __cxa_atexit.
```

This comment explains it further:

```
newlib/libc/stdlib/__call_atexit.c

start -> atexit -> __register_exitproc
start -> __libc_init_array -> __cxa_atexit -> __register_exitproc
```

Interestingly enough, atexit guarantees 32 slots, but calls to __cxa_atexit() consume the same slots, so we are not actually guaranteeing 32 atexit slots to the end user. There can be many static C++ objects, and calls to __cxa_atexit() are hidden from the programmer, so you may inadvertently consume many such slots. In my opinion, it is all a big mess, and not just for these reasons.

The problem of unexpected atexit entries is not as far-fetched as it may seem. In fact, I recently discovered a regression in this area in GCC 12:

Bug 107500 - Useless atexit entry for ~constant_init in eh_globals.cc
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=107500

By the way, they fixed it for GCC 13, but are leaving it unfixed for GCC up to at least 12.3, so I have to manually patch GCC for the time being:

https://github.com/rdiez/DebugDue/blob/master/Toolchain/Patches/Gcc12EhGlobalsAtexit.patch

GCC has more long-standing problems in this area anyway. These initialisation and termination aspects are often neglected all over the place, so that is why I believe it is important to check automatically.

Memory allocation with atexit is problematic. Normally, only 32 slots are guaranteed. What happens when attempting to register slot number 33 is not entirely clear, my guess with most real-life implementations is that it may be dropped without warning, which is no good practice. After all, it is difficult to cater for such failures during early initialisation.

Besides, allocating the first atexit slot will consume (I think) 384 bytes at once, which is the size of this array:

`static struct on_exit on_exits[ATEXIT_MAX];`

I have one small target with 16 KiB RAM, and the main reason I am trying to switch to Picolibc is to free up as much SRAM as I can.

If you enable dynamic memory allocation for atexit(), it is hard to strictly control dynamic memory allocation, because only 1 out of 32 slots will trigger a malloc(), and the atexit slot allocation order is difficult to guarantee.

For all of the above, and for some other reasons listed below, I strive to avoid automatic initialisation and destruction in my firmwares:

https://rdiez.miraheze.org/wiki/How_to_write_maintainable_initialisation_and_termination_code

Up until now I have been able to eradicate all such automatic/hidden initialisation and destruction from all my firmwares, but I fear the day where that will no longer be feasible. For example, you may want to use some complex C++ library like Boost which insists on global static objects, and fixing it may be hard.

Therefore, I think the best thing would be the ability to check the number of atexit slots currently registered. This way, your code can accommodate for the odd library or situation where you know that say 2 such entries are used, but no more should suddenly show up.

I can patch Picolibc for my purposes. The reason I am posting here is that I think this could benefit other users too. In fact, I am surprised that such a control mechanism is not already implemented and widespread, at least in embedded environments, where you cannot run Valgrind's Memcheck.

A script which runs after linking would add to the build time, and would not allow the case where you now that a certain number of slots are used.

Writing your own overriding version of atexit/on_exit is problematic. It is not clearly documented how your overriding routines should be. I think on_exit is not really standard. If something changes in Picolibc, your overrides may stop working without notice.

keith-packard, 2023-06-19:
> Interestingly enough, atexit guarantees 32 slots, but calls to __cxa_atexit() consume the same slots, so we are not actually guaranteeing 32 atexit slots to the end user. There can be many static [...]

Yeah, as is often the case, interactions between the C and C++ specs is not entirely satisfactory.

> Therefore, I think the best thing would be the ability to check the number of atexit slots currently registered. This way, your code can accommodate for the odd library or situation where you know that say 2 such entries are used, but no more should suddenly show up.

I can also imagine knowing if some atexit call failed would be useful. I'd be happy to let you add an API that let you get information out of the existing picolibc atexit implementation.

> A script which runs after linking would add to the build time, and would not allow the case where you now that a certain number of slots are used.

It would, however, suit your current needs where you don't want *any* atexit/__cxa_atexit  calls made. Having that detected at build time seems ideal until you discover a need for a non-zero number.

> Writing your own overriding version of atexit/on_exit is problematic. It is not clearly documented how your overriding routines should be. I think on_exit is not really standard. If something changes in Picolibc, your overrides may stop working without notice.

It might be nice to document suites of C library functions that you can provide your own implementation of. There are lots of instances of 'malloc' replacement libraries, for instance, and those magically "know" what the set of required replacement functions are to work with the target C library. Documenting the set of functions necessary to provide your own atexit implementation and then promising to not change that would be pretty reasonable.

rdiez, 2023-06-19:
OK, it looks like I need to give this some more thought. I'll close this issue for now.

closed 2023-06-19 by rdiez