picolibc/picolibc #492: Reducing assert() code size.

rdiez, 2023-06-19:
I find assertions an invaluable development tool. The trouble is, assert() stores the failed condition (as it appears in the source code) as a string in program space, and that sometimes makes the debug firmware overflow the available space.

I have been using for years a method which lets me choose how much information assert() keeps. It is here:

https://github.com/rdiez/DebugDue/blob/master/Toolchain/Patches/NewlibUserImplementationOfAssert.patch

The included file is here:

https://github.com/rdiez/DebugDue/blob/master/Project/src/UserImplementationOfAssert/UserImplementationOfAssert.h

The firmware project can then define ASSERT_TYPE to one of the allowed values, and choose how big an assert() call should be.

I wonder whether Picolibc should implement such a mechanism. I have seen in the Internet other strategies to reduce the size of assert() calls, but nothing standard, widespread or so flexible.

keith-packard, 2023-06-19:
You can implement your own version of assert.h; there's no particular reason to use the one provided by the C library. If you want to propose a fancier version for picolibc, the only requirement would be that it be controlled by symbols prefixed with `__PICOLIBC_` and that its default behavior match the current assert.h.

rdiez, 2023-06-19:
OK, thanks for the hints, I'll close this suggestion for now.

closed 2023-06-19 by rdiez