picolibc/picolibc #506: strncpy() reads beyond source string end

aescolar, 2023-07-06:
When it thinks it can, strncpy() reads 4 bytes at a time from the source:
https://github.com/picolibc/picolibc/blob/main/newlib/libc/string/strncpy.c#L115
but while doing so, it can easily read beyond the source string end.
ASAN is not pleased with this, and reports it as an issue.

The issue can be easily reproduced using the Zephyr integration of picolibc, and many of zephyrs tests, for example:
With todays Zephyr main (235caf2234e7a778d9625989c95a38d3337ce03b), in the Zephyr base folder:
```
$ mkdir build && cd build
build$ cmake ../tests/kernel/sched/preempt/ -DBOARD=native_sim -GNinja -DCONFIG_ASAN=y -DCONFIG_PICOLIBC=y -DCONFIG_NO_OPTIMIZATIONS=y
build$ ninja
build$ zephyr/zephyr.exe
```
Here the error report from ASAN:
```
==329612==ERROR: AddressSanitizer: global-buffer-overflow on address 0x08076444 at pc 0x08064a1e bp 0xf53ff148 sp 0xf53ff138
READ of size 4 at 0x08076444 thread T1
    #0 0x8064a1d in strncpy modules/lib/picolibc/newlib/libc/string/strncpy.c:115
    #1 0x805a043 in z_setup_new_thread zephyr/kernel/thread.c:600
    #2 0x80593df in prepare_multithreading zephyr/kernel/init.c:441
    #3 0x8059510 in z_cstart zephyr/kernel/init.c:567
    #4 0x8068489 in sw_wrapper zephyr/scripts/native_simulator//common/src/nce.c:184
    #5 0xf78432dd in asan_thread_start ../../../../../src/libsanitizer/asan/asan_interceptors.cpp:198
    #6 0xf752364b  (/lib/i386-linux-gnu/libc.so.6+0x12364b)
```

closed 2023-07-07 by keith-packard

aescolar, 2023-07-07:
Thanks @keith-packard 
I forgot to tell that this is building with gcc (12.1) (on ubuntu 22.04)
So, the detection on that patch will not work, though with a change like this so it also detects it in gcc it does.

```diff
diff --git a/newlib/libc/include/sys/config.h b/newlib/libc/include/sys/config.h
index 616573bf11..267469e6c9 100644
--- a/newlib/libc/include/sys/config.h
+++ b/newlib/libc/include/sys/config.h
@@ -353,6 +353,10 @@ SUCH DAMAGE.
 #if __has_feature(address_sanitizer)
 #define PICOLIBC_NO_OUT_OF_BOUNDS_READS
 #endif
+#else
+#if __SANITIZE_ADDRESS__ /* For gcc */
+#define PICOLIBC_NO_OUT_OF_BOUNDS_READS
+#endif
 #endif
 
 #endif /* __SYS_CONFIG_H__ */
```

keith-packard, 2023-07-07:
Thanks, I've pushed another PR for this. I'm thinking that this could move to some internal header file, but there really isn't a good place. Using the __has_feature from cdefs.h would be nice, but it isn't always defined while reading this file. Probably an include ordering issue.

aescolar, 2023-07-10:
> ..but there really isn't a good place. Using the __has_feature from cdefs.h would be nice..

Maybe a new header common to both this one and cdefs?

PR #508 (2023-07-07): libc/string: Keep string code from reading out-of-bounds when asan enabled
When the llvm address sanitizer is enabled while building the library, we need to bypass the optimized string functions which read past the end of the strings looking for null bytes.

Closes: #506