picolibc/picolibc #536: MSP430: excessive ALIGN in picolibc.ld

pjonsson, 2023-08-18:
Trying to use picolibc on a https://www.ti.com/product/MSP430F1611, I ran into a space issue. Trying to figure out where to find some bytes, I noticed the alignment in the generated picolibc.ld is 8, in comparison, the GCC from TI uses 2 for this model.

It's probably not a bug, but I also got a link error on some multiplication routines, and the `-lmul_none` in the default MSP430 configuration was surprising to me.

keith-packard, 2023-08-18:
yeah, it's a bug in that the sample linker script over-aligns on many targets. I think this could be fixed by allowing each target to override the alignment value in the cross compilation script. It uses 8 because that's the strictest alignment required on any supported target...

pjonsson, 2023-08-18:
Is it a deliberate choice to not run MSP430 tests, or are the tests "just" not made yet?

pjonsson, 2023-08-18:
I guess that explains why GCC was complaining about missing entry symbols.

The newlib from TI suggests overriding write() to redirect printf() output. I have looked at the content of the doc directory but I'm still not sure, is there something similar to overriding write() in picolibc for getting a hold of the printf output?

keith-packard, 2023-08-18:
For picolibc, take a look in the 'semihost' directory -- that provides implementations of several different mechanisms for supporting console output while running picolibc tests without any other code. There's several different semihost implementations for arm, riscv, m68k, nios2 and x86,  there's a version which uses a standard 8250 serial port for MIPS and one using the PowerPC OPAL BIOS for that target.

I'm not sure whether qemu supports any kind of semihost support for mps430, or if we'd need to write code for a virtual serial port.

The other piece needed is some picocrt code. That initializes the processor and memory before jumping into `main()`. Take a look at some of the smaller examples, like mips, which just initializes the stack pointer (and maybe the FPU) and then jumps into C code from there.

commit 633fb79d (2023-09-05): msp430: set default align to 2

Set the default align to 2, which
is the same alignment as the TI SDK.

Fixes #536