picolibc/picolibc #544: PIcolibc 1.8.3 m68k-unknown-elf regressions

andrewmroberts69, 2023-08-22:
I've been building picolibc for m68k-unknown-elf, and there are three main regressions for the latest release. I've a fix for one, one relates to recent softfloat breakage, and not sure what to do about the other, but do have a work around:

1) ./semihost/machine/m68k/m68k_semihost.S
the use of: 
bkpt #0 
does not assemblt for mc68000, better to use 
.short 0x4848
which assembles for eveything, and does the same thing.
68000 doesn't do brpt, but does the illegal instruction trap, which ammounts to the same thing (sans breakpt cycle).
--- ./semihost/machine/m68k/m68k_semihost.S.orig        2023-08-13 09:12:10.1481
66853 +0100
+++ ./semihost/machine/m68k/m68k_semihost.S     2023-08-13 09:16:50.412507915 +0
100
@@ -50,7 +50,8 @@
 #ifdef __mcoldfire__
        .short 0x4ac8
 #else
-       bkpt #0
+       /* bkpt #0 */
+       .short 0x4848
 #endif
        .long 0x4e7bf000
        rts

I've been using a m68k tool chain built as:
 /usr/local/m68k-unknown-elf/bin/m68k-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/usr/local/m68k-unknown-elf/bin/m68k-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/local/m68k-unknown-elf/libexec/gcc/m68k-unknown-elf/10.5.0/lto-wrapper
Target: m68k-unknown-elf
Configured with: ../gcc-10.5.0/configure --target=m68k-unknown-elf --prefix=/usr/local/m68k-unknown-elf --with-arch=m68k --enable-languages=c,c++ --with-gnu-as --with-as=/usr/local/m68k-unknown-elf/bin/m68k-unknown-elf-as --with-gnu-ld --with-ld=/usr/local/m68k-unknown-elf/bin/m68k-unknown-elf-ld --with-newlib --disable-bootstrap --disable-shared --enable-multilib --enable-threads=no --enable-target-optspace --disable-libquadmath --disable-libquadmath-support --enable-__cxa_atexit --without-long-double-128 --disable-libmudflap --disable-libgomp --disable-lto --disable-libssp --disable-libada --disable-libstdcxx-verbose --disable-nls --disable-decimal-float --disable-libatomic --disable-libitm --disable-libsanitizer --disable-libvtv --disable-libcilkrts
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.5.0 (GCC)

2) multilibs with m68k-unknown-elf
 /usr/local/m68k-unknown-elf/bin/m68k-unknown-elf-gcc --print-multi-lib
.;
m68000;@mcpu=68000
m68040;@mcpu=68040
m68060;@mcpu=68060
mcpu32;@mcpu=cpu32
mfidoa;@mcpu=fidoa
softfp;@msoft-float
m68040/softfp;@mcpu=68040@msoft-float
m68060/softfp;@mcpu=68060@msoft-float
mfidoa/softfp;@mcpu=fidoa@msoft-float

mfidoa hasn't been working with picolibc forever, so I've been filtering it out. But the current version doesn't work for any of the softfp targets including 68000. There is a test for duplicate names in scripts/duplicate-names
this fails for softfp on all targets, so I had to patch out the hard exit and ignore the error.
--- duplicate-names.safe        2023-08-13 19:16:30.536463040 +0100
+++ duplicate-names     2023-08-13 19:16:49.456618211 +0100
@@ -7,6 +7,6 @@
 if [ -e "$OUTPUT" ] && [ -s "$OUTPUT" ]; then
     echo "Duplicate names in ${FILE}"
     cat "$OUTPUT"
-    exit 1
+    #exit 1
 fi
 exit 0

Keep up the good work,

thanks

andrewmroberts69, 2023-08-22:
`--- ./semihost/machine/m68k/m68k_semihost.S.orig	2023-08-13 09:12:10.148166853 +0100
+++ ./semihost/machine/m68k/m68k_semihost.S	2023-08-13 09:16:50.412507915 +0100
@@ -50,7 +50,8 @@
 #ifdef __mcoldfire__
 	.short 0x4ac8
 #else
-	bkpt #0
+	/* bkpt #0 */
+	.short 0x4848
 #endif
 	.long 0x4e7bf000
 	rts
`
`--- duplicate-names.safe	2023-08-13 19:16:30.536463040 +0100
+++ duplicate-names	2023-08-13 19:16:49.456618211 +0100
@@ -7,6 +7,6 @@
 if [ -e "$OUTPUT" ] && [ -s "$OUTPUT" ]; then
     echo "Duplicate names in ${FILE}"
     cat "$OUTPUT"
-    exit 1
+    #exit 1
 fi
 exit 0
`

Here's the patches again, this time protected from the modern world.

keith-packard, 2023-08-22:
Thanks for the patches. I'm building an m68-unknown-elf toolchain to go explore the soft float and duplicate names issues. I'll apply the semihost patch once I've seen if it actually works with qemu.

Note that there are a handful of qemu patches floating around that are required for hard float support, those are working their way upstream but can be found here: https://github.com/keith-packard/qemu/tree/m68k-mul2

keith-packard, 2023-08-23:
Ok, it turns out there are a pile of bugs with the libgcc soft float support for m68k too. With these and some judicious skipping of tests *and* the qemu patches, I've got all tests passing with a multilib toolchain built with crosstool-ng. I think that has an even larger set of multilib configurations:
```
$ m68k-unknown-elf-gcc --print-multi-lib
.;
m68000;@mcpu=68000
m68020;@mcpu=68020
m68040;@mcpu=68040
m68060;@mcpu=68060
m51qe;@mcpu=51qe
m5206;@mcpu=5206
m5206e;@mcpu=5206e
m5208;@mcpu=5208
m5307;@mcpu=5307
m5329;@mcpu=5329
m5407;@mcpu=5407
m54455;@mcpu=54455
m5475;@mcpu=5475
mfidoa;@mcpu=fidoa
m68020/softfp;@mcpu=68020@msoft-float
m68040/softfp;@mcpu=68040@msoft-float
m68060/softfp;@mcpu=68060@msoft-float
m5475/softfp;@mcpu=5475@msoft-float
mfidoa/softfp;@mcpu=fidoa@msoft-float
```

Here's the libgcc patches. I've send them to the gcc mailing list.

[0001-libgcc-m68k-Fixes-for-soft-float.txt](https://github.com/picolibc/picolibc/files/12414486/0001-libgcc-m68k-Fixes-for-soft-float.txt)

PR #551 (2023-08-24): M68k fixes
Here's a pile of m68k floating point fixes. This switches to a bare metal multilib m68k toolchain so that we can test a long list of targets. This requires patches to gcc and qemu as both of those have numerous floating point issues.

The gcc fixes are here: https://github.com/keith-packard/gcc/tree/m68k-soft-float
The qemu fixes are here: https://github.com/keith-packard/qemu/tree/m68k-mul2

Closes: #544
closed 2023-08-24 by keith-packard