picolibc/picolibc #546: tests: link error because libgcc is not included when linking with test.specs

pjonsson, 2023-08-22:
I get a link error on the tests with MSP430 because of various missing helpers from libgcc. I can't find anything relevant for libgcc under `scripts/`, which makes me believe the problem is more general since the GCC documentation seems quite convinced one needs libgcc.a even with nostdlib:

> One of the standard libraries bypassed by -nostdlib and -nodefaultlibs is libgcc.a, a library of internal subroutines which GCC uses to overcome shortcomings of particular machines, or special needs for some languages. (See [Interfacing to GCC Output](https://gcc.gnu.org/onlinedocs/gccint/Interface.html#Interface) in GNU Compiler Collection (GCC) Internals, for more discussion of libgcc.a.) In most cases, you need libgcc.a even when you want to avoid other standard libraries. In other words, when you specify -nostdlib or -nodefaultlibs you should usually specify -lgcc as well. This ensures that you have no unresolved references to internal GCC library subroutines. (An example of such an internal subroutine is __main, used to ensure C++ constructors are called; see [collect2](https://gcc.gnu.org/onlinedocs/gccint/Collect2.html#Collect2) in GNU Compiler Collection (GCC) Internals.)

Here's my generated test.specs:
```
%rename link test_link

*link:
%{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfprintf=__f_vfprintf} %{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfscanf=__f_vfscanf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfprintf=__d_vfprintf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfscanf=__d_vfscanf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfprintf=__i_vfprintf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfscanf=__i_vfscanf} --gc-sections %(test_link)
```
Curiously, libgcc is included in the same section in picolibc.specs:
```
link:
%{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfprintf=__f_vfprintf} %{DPICOLIBC_FLOAT_PRINTF_SCANF:--defsym=vfscanf=__f_vfscanf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfprintf=__d_vfprintf} %{DPICOLIBC_DOUBLE_PRINTF_SCANF:--defsym=vfscanf=__d_vfscanf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfprintf=__i_vfprintf} %{DPICOLIBC_INTEGER_PRINTF_SCANF:--defsym=vfscanf=__i_vfscanf} -L%{-picolibc-prefix=*:%*/picolibc/msp430/lib/%M; -picolibc-buildtype=*:/usr/local/picolibc/msp430/lib/%*/%M; :/usr/local/picolibc/msp430/lib/%M} -L%{-picolibc-prefix=*:%*/picolibc/msp430/lib; -picolibc-buildtype=*:/usr/local/picolibc/msp430/lib/%*; :/usr/local/picolibc/msp430/lib} %{!T:-Tpicolibc.ld} %(picolibc_link) --gc-sections 

*lib:
--start-group %(libgcc) -lmul_none -lc %{-oslib=*:-l%*} --end-group
```

The path to libgcc.a cannot be hardcoded, there's a separate libgcc.a/libmul_*.a for each multi-lib configuration.

keith-packard, 2023-09-11:
Can we close this issue now? Or do you want to find some even better linker script fixes for this target?

pjonsson, 2023-09-11:
This was about getting linking to work at all, and since the tests pass that has obviously been accomplished, so closing this.

closed 2023-09-11 by pjonsson