picolibc/picolibc #554: CI speedup overview

pjonsson, 2023-08-24:
# Overview

Speeding up the CI consists of some (mostly) orthogonal lines of work:

- #553 (reduce load times in each CI run, free up space for other things in the Github cache area)
- Use ccache in the jobs to reduce build times (Peter measured do-arm-configure and got 3.8x speedup with warm cache)
- [ ] Add ccache support to Meson (Keith has fix)
- [ ] Add ccache support to Cmake (Peter has fix)
- [ ] Add ccache support to CI (Peter will fix)
- Consolidate/re-organize job composition to avoid load times/improve ccache hit rates

# Technical notes

## CI Runners

The documented limit for concurrent runners for the free level is 20, where at most 5 can be macOS.

## Cache

The Github cache is 10 GB and evicts with LRU policy. With the reduced image sizes in #553, there is 7.1 GB of space available for ccache.

The do-arm-configure measurement filled ccache to 7.95% of 5 GB (just under 400 MB). Assuming all other configurations have the same size, that gives room for 17 distinct ccaches before Github starts evicting the working set.

The cache can be compressed a bit (~150 MB -> 95 MB for ARM object files).

# Miscellaneous Ideas/Questions

Can the Github [container registry](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry) eliminate the need for the images to be in the 10 Gb cache?

pjonsson, 2023-08-25:
When I wrote my question, my perspective was mostly about simplicity since I believe eliminating decision points is by far the best way to encourage hesitant observers to make the leap into becoming occasional contributors. A single image makes it easy to explain how to run tests, and where to add additional software required to run new tests.

I'm starting to wonder if the single image is more or less a requirement for slicing the CI jobs to be on a per architecture split instead.

Speaking of images, are `.github/debian-packages.txt` and `.github/packages.txt` used?

pjonsson, 2023-08-27:
What do you think about cancelling the previous jobs of a PR when pushing an update?

I haven't tested, but combining https://docs.github.com/en/actions/using-jobs/using-concurrency#example-only-cancel-in-progress-jobs-or-runs-for-the-current-workflow and the `github.head_ref || github.run_id` fallback value described at https://docs.github.com/en/actions/using-jobs/using-concurrency#example-using-a-fallback-value looks promising.

keith-packard, 2023-08-27:
> What do you think about cancelling the previous jobs of a PR when pushing an update?

That would be awesome. I just don't know how to make it work.

pjonsson, 2023-08-28:
I updated the bullet list at the start, and it seems like we've made quite a lot of progress.

The remaining major thing is fitting the cache for the Linux&Zephyr jobs in the 10GB limit. You described one possibility of re-organizing the jobs on a per-architecture basis instead but that it might result in some quite short and some really long jobs. I think that strategy would enable the most cache sharing, and if there is sufficient space in the Github cache, the longest jobs could be split into 2 or 3 jobs.

We currently have >8GB "free space" in the cache, so an individual cache for each of the zephyr jobs can easily fit in in that. It would be better if the Linux jobs could be cached, but that doesn't fit. Since the number of concurrent runners is also an issue, I will make a PR which speeds up the Zephyr jobs as a start.

I'll take a look at cancelling the jobs when a PR is updated, too.