picolibc/picolibc #557: build system: allow limiting the concurrent number of link jobs

pjonsson, 2023-08-25:
With ccache, it's possible that building the object files is almost immediate, so the only long running commands are the linking jobs. Running too many concurrent linking jobs is usually bad for performance, so it would be nice to be able to limit that.

Did a quick search and there is apparently support in Meson for that: https://mesonbuild.com/Build-options.html#max-links

keith-packard, 2023-08-25:
Given that meson already limits parallel operations to the number of cores available, and given that CI hosts have so few, is this necessary?

pjonsson, 2023-08-25:
I think you're right about the Github CI, my main concern for this ticket was actually my home computer (cheap to get 20+ "cores" today). But I think this ticket is also relevant for downstream users of picolibc who have their own CI.

If the effort/code required for this is high, I think you should close the ticket.

closed 2024-07-02 by keith-packard