picolibc/picolibc #591: ld stating "error:" but build continues

pjonsson, 2023-09-17:
Excerpt from running `.github/do-linux`:
```
[2326/2665] Linking target newlib/libm/test/math_test_6
/opt/msp430-unknown-elf/bin/../lib/gcc/msp430-unknown-elf/13.2.0/../../../../msp430-unknown-elf/bin/ld: error: final size of uleb128 value at offset 0x45 in .debug_loclists from newlib/libc.a(libc_tinystdio_vffprintf.c.o) exceeds available space
/opt/msp430-unknown-elf/bin/../lib/gcc/msp430-unknown-elf/13.2.0/../../../../msp430-unknown-elf/bin/ld: error: final size of uleb128 value at offset 0x49 in .debug_loclists from newlib/libc.a(libc_tinystdio_vffprintf.c.o) exceeds available space
/opt/msp430-unknown-elf/bin/../lib/gcc/msp430-unknown-elf/13.2.0/../../../../msp430-unknown-elf/bin/ld: error: final size of uleb128 value at offset 0x75 in .debug_rnglists from newlib/libc.a(libc_tinystdio_vffprintf.c.o) exceeds available space
```
It seems strange that ld states "error:" and the build then continues.

The message disappears when building without debug info.

keith-packard, 2023-09-17:
Yeah, I poked at this a bit, attempting to elide whatever debug info was causing the overflow. As it doesn't appear to be a fatal error, I gave up after failing to fix it easily. Feel free to poke at it more if you like.

pjonsson, 2023-09-17:
It seems there are others who have issues with the debug information as well, https://e2e.ti.com/support/tools/code-composer-studio-group/ccs/f/code-composer-studio-forum/1217927/msp430-gcc-corrupt-dwarf-debug-info-when-enabling-lto/4601012?tisearch=e2e-sitesearch&keymatch=dwarf%2520source#4601012, has a fix at https://github.com/heytoaster/msp430-gcc-9.3.1.11/commit/7d61031326fe9dabbff200d530c20ace6febf6f5

There's also https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/1082431/msp430-gcc-opensource-linker-error-dwarf-error-line-info-data-is-bigger-0xfffffffc-than-the-space-remaining-in-the-section-0x46, but the "solution" is to not use CMake.

closed 2026-11-10 by keith-packard