picolibc/picolibc #603: build system should check if multilib configuration compiles before using it.

xbjfk, 2023-09-23:


keith-packard, 2023-09-23:
Can you provide a complete cross compilation control file which demonstrates the issue? It should be easy to fix if we can see how it's broken ...

xbjfk, 2023-09-23:
Hi, thank you for your fast response. I did a little more digging (which I should have done initially - sorry about that!) and I think I found the problem - arm-none-eabi should not have an m64 variant - but the meson.build file assumes it does because `cc.cmd_array()` ([used here](https://github.com/picolibc/picolibc/blob/2faa0bdb4aafa16df5b3d1e9174a523d04043970/meson.build#L864)) doesnt include `c_args` - meaning clang is used without the --target argument. For reference, my cross file looks like this:

```
[host_machine]
system = 'none'
cpu_family = 'arm'
cpu = 'arm'
endian = 'little'

[binaries]
c = 'clang-17'
c_ld = 'lld'
ar = 'llvm-ar'

[built-in options]
c_args = ['-mcpu=cortex-m4', '--no-default-config', '--target=arm-unknown-none-eabi']
c_link_args = ['-fuse-ld=lld', '--no-default-config', '--target=arm-unknown-none-eabi']

```
But if I change it to something like this...
```
[binaries]
c = ['clang-17', '--target=arm-unknown-none-eabi']
...

[built-in options]
c_args = ['-mcpu=cortex-m4', '--no-default-config']
...
```

Then it works as expected.

If I change 
```
  target_list = run_command(cc.cmd_array() + ['--print-multi-lib'], check : true).stdout().strip().split('\n')
```
to
```
  target_list = run_command(cc.cmd_array() + get_option('c_args') + ['--print-multi-lib'], check : true).stdout().strip().split('\n')
```
then either cross file works fine. And it seems like `get_option('c_args')` will get the cross compilation arguments back to [0.51](https://github.com/mesonbuild/meson/pull/5263)

keith-packard, 2023-09-23:
Yeah, we could emit an error when options selecting a specific target are detected while multilib is enabled, but that turns out to be tricky. clang usually uses --target, but gcc doesn't, instead preferring a wide range of -m options. I haven't ever seen a clang configuration that supports multilib at all, so one easy fix here is to simply bail if you use clang and ask for multilib?

xbjfk, 2023-09-23:
Multilib was supported for a while in clang, but not to the extent of GCC, until a [recently](https://discourse.llvm.org/t/rfc-multilib/67494) in clang/llvm 17, where they added the `-print-multi-lib` option, among others.
It seems to work fine (assuming you use one of the fixes to call -print-multi-lib correctly) and in my opinion is nicer than having to specify the correct --target you want each time.

 For that reason I think that the call to `-print-multi-lib` should have the c compiler flags. There *should* be no issues with that, as in my original post I did not realize it was calling `clang` without my --target= argument (thereby trying an impossible multilib variant)

keith-packard, 2023-09-23:
> For that reason I think that the call to `-print-multi-lib` should have the c compiler flags. There _should_ be no issues with that, as in my original post I did not realize it was calling `clang` without my --target= argument (thereby trying an impossible multilib variant)

oh, you stuck compiler options in the `[built-in options]` section. Support for that was added in meson 0.56 it seems, which is why none of the examples use it (picolibc still supports meson 0.53). All of the samples place the needed compiler flags right in the `c` definition in the `[binaries]` section as that has always done the right thing. Let's see if #604 doesn't break CI; if you want to test that and post the results here, that would be great, but it's a reasonable change even if it doesn't completely fix this issue.

xbjfk, 2023-09-23:
This sounds good to me - #604 makes everything work perfectly. It also seems to work even when `c_args` is in `properties`.

keith-packard, 2023-09-23:
> This sounds good to me - #604 makes everything work perfectly. It also seems to work even when `c_args` is in `properties`.

awesome. thanks for checking!

commit 2bdd26e8 (2023-09-24): Include the c-args option value when querying multi-lib support

Clang will vary the multi-lib output depending on other compiler
flags. When those flags are placed in the c-args option instead of the
compiler specification itself, we'll need to include that value too.

Closes: #603
PR #604 (2023-09-24): Include the c-args option value when querying multi-lib support
Clang will vary the multi-lib output depending on other compiler flags. When those flags are placed in the c-args option instead of the compiler specification itself, we'll need to include that value too.

Closes: #603
closed 2023-09-24 by keith-packard