picolibc/picolibc #617: Backtrace on fault shows wrong handler

begriffs, 2023-10-02:
Environment:
* _PICOLIBC_VERSION "1.8"
* arm-none-eabi-gdb 7.12.1
* Nucleo F411RE

When I encounter a fault and request a backtrace, GDB always shows `arm_usagefault_isr` as the innermost frame, even when the Configurable Fault Status Registers (CFSR) reveal that it's actually e.g. a hardfault or busfault:

```
#0  0x08002388 in arm_usagefault_isr ()
#1  <signal handler called>
#2  0x08001c66 in prvPortStartFirstTask () at freertos/portable/GCC/ARM_CM4F/port.c:270
#3  0x08001df0 in xPortStartScheduler () at freertos/portable/GCC/ARM_CM4F/port.c:384
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

When I define `arm_hardfault_isr` or `arm_busfault_isr` functions in my own code and override the weak symbols, then GDB shows the correct names in the backtrace.

My hypothesis is that, because these all create weak aliases of `arm_halt_isr`
```
isr_halt(hardfault);
isr_halt(memmanage);
isr_halt(busfault);
isr_halt(usagefault);
```
and `usagefault` is the last one in the list, somehow `arm_usagefault_isr` becomes the name of the debug symbol for this function. Is there any way to give GDB a hint about which one is actually being called?

keith-packard, 2023-10-02:
Yeah, you've diagnosed the issue correctly - the default interrupt table uses the same ISR for all four of the standard fault interrupts. The only way to get GDB to tell you which fault is happening is to use different functions by replacing the weak symbols with ones of your own.

In reality, you're likely to want to replace the entire interrupt vector anyways -- any real hardware will have more than the short list of interrupts provided by the picolib code. If you use the same name, you can continue to use the stock picolibc crt0 code. That's what I do in various picolibc-using projects I've done. Here's an example:

https://github.com/keith-packard/snek/blob/main/chips/samd21/ao-interrupt.c

This one doesn't have different functions for the various faults, but that should be pretty straightforward to adjust.

begriffs, 2023-10-02:
Thanks for the explanation.

closed 2023-10-02 by begriffs