picolibc/picolibc #624: Should stdio streams be opened in startup code?

joe-img, 2023-10-11:
Symptom: Using posix IO for std[out/in/err] does not work with qemu

As far as I understand, riscv uses the arm style semihosting. I can't find anywhere that explicitly says this, but it's hinted to in the [riscv semihosting spec](https://github.com/riscv-software-src/riscv-semihosting/blob/main/riscv-semihosting-spec.adoc#2-risc-v-semihosting). And with qemu, the define CONFIG_ARM_COMPATIBLE_SEMIHOSTING is true for riscv.

> RISC-V semihosting borrows from the design of other publicly available and open source semihosting mechanisms to minimize the development effort required.

Without using posix IO, the writes are done one character at a time with the semihosting call [SYS_WRITEC ](https://developer.arm.com/documentation/dui0471/g/Semihosting/SYS-WRITEC--0x03-?lang=en)to 'the debug channel'. This is reflected in qemu and we see the output.

With posix IO, the writes are done with the semihosting call [SYS_WRITE](https://developer.arm.com/documentation/dui0471/g/Semihosting/SYS-WRITE--0x05-?lang=en). The subtle difference here is that this writes to a "file previously opened with SYS_OPEN". Qemu adheres to this, and seeing as stdout was never opened, the write fails.

What I read implies those streams should already be open:
> Three text streams are predefined. These streams are implicitly opened and unoriented at program startup.

https://en.cppreference.com/w/cpp/io/c/std_streams

I also found an old comment from avr-libc: (https://github.com/sifive/picolibc/blob/master/newlib/libc/tinystdio/stdio.h#L91)
> The standard streams \c stdin, \c stdout, and \c stderr are
    provided, but contrary to the C standard, since avr-libc has no
    knowledge about applicable devices, these streams are not already
    pre-initialized at application startup

Should picolibc be opening FD's 0, 1 and 2 at startup with semihosting enabled?

keith-packard, 2023-10-11:
> As far as I understand, riscv uses the arm style semihosting. I can't find anywhere that explicitly says this, but it's hinted to in the [riscv semihosting [...]

Yup, the initial work on the RISC-V semihosting stuff was simply a copy of the arm semihosting stuff so that it would be easier to get support in various tools. The qemu patches that I managed to land share the bulk of the code between the arm and risc-v implementations even.

> > RISC-V semihosting borrows from the design of other publicly available and open source semihosting mechanisms to minimize the development effort required.

This should probably have been more explicit about the source of the design.

> Without using posix IO, the writes are done one character at a time with the semihosting call [SYS_WRITEC ](https://developer.arm.com/documentation/dui0471/g/Semihosting/SYS-WRITEC--0x03-?lang=en)to [...]

Ah, something I'd never tried doesn't work. There's code in the semihosting path to handle the legacy stdio which must go through the read/write path, but that isn't being used with tinystdio because it assumes you won't have defined `posix-console` -- you don't need that as the arm/risc-v semihosting code has native support for tinystdio stdin/stdout/stderr.

> Should picolibc be opening FD's 0, 1 and 2 at startup with semihosting enabled?

It certainly can; I've submitted a PR, #625 which makes `-Dposix-console=true` work for arm/risc-v semihosting.

keith-packard, 2023-10-30:
PR #625 was merged, fixing this issue.

closed 2023-10-30 by keith-packard

joe-img, 2023-10-31:
Many thanks for fixing this issue Keith!