picolibc/picolibc #633: How to enable semihosting in meson project?

pmiettinen, 2023-10-28:
Related https://github.com/picolibc/pico-hello/issues/1

Examples in this repo seem to be for Makefile and the pico-hello is using revision 1.7. After updating to 1.8.5 the build fails.

I have tried some link arg variations in my own project importing picolibc as meson subproject but didn't manage to get the build passing with semihosting (and printf call). Any tips? If I omit semihosting, then I can get the build to pass.

keith-packard, 2023-10-29:
It's the subproject support that broke -- the crt0 libraries aren't available through the subproject mechanism which relies on variables being set for every exported object. So, yup, this is a picolibc bug (mostly).

keith-packard, 2023-10-30:
I decided to just switch the subproject example to use the picocrt object file variables instead as that didn't require
changes to picolibc. All fixed over there.

closed 2023-10-30 by keith-packard