picolibc/picolibc #644: Libc with only general-purpose registers

zyedidia, 2023-12-05:
Picolibc is great, but I would like to use it in a kernel that only uses general-purpose registers (no floating point or SIMD). I would like to remove the components that rely on FP/SIMD such as the math routines and SIMD-optimized memcpy/memset implementations, leaving just the parts that are useful to me -- mostly just memory allocation, stdio, string, and file operations. Some parts like all the math routines can be fully dropped, but memcpy/memset and printf look like they need to be recompiled with different sources/flags (`-mgeneral-regs-only`).

Please let me know if there's a straightforward way to do this, or if not what you suggest for this use-case. Thanks!

keith-packard, 2023-12-05:
Which target are you building for? They should all deal with building in soft-float mode, and that should avoid using the FP registers. If not, please let us know which target is doing the wrong thing here (x86 was until recently, for example).

As for SIMD stuff, there's not any current settings which control which registers the assembly code uses, but that seems like a fine idea. Adding a meson configuration setting and associated preprocessor symbol that the code could use to control how it was compiled would be the right direction here, unless the compiler already provides some preprocessor symbol set when `-mgeneral-regs-only` is specified on the command line.

Happy to make picolibc usable on more targets, especially ones doing fun stuff like this.

zyedidia, 2023-12-06:
I'm targeting AArch64 and x86-64. Looking at the AArch64 version, it seems like FP/SIMD registers are only being used in printf (probably as part of va_args stuff) even though I have specified `-Dformat-default=long-long` which should disable float support. I think `-mgeneral-regs-only` should solve this. The memcpy/memset implementations are not using SIMD-optimized versions, maybe because the build is optimizing for size, so that seems fine, but without `-mgeneral-regs-only` the compiler may use the SIMD registers anyway for small copies.

The main issue when I enable `-mgeneral-regs-only` is that it causes errors in parts of the build that use floats, such as

```
../picolibc/newlib/libc/tinystdio/ecvt_r.c: In function ‘ecvt_r’:
../picolibc/newlib/libc/tinystdio/ecvt_r.c:49:1: error: ‘-mgeneral-regs-only’ is incompatible with the use of floating-point types
   49 | ecvt_r (double invalue,
```

Maybe such code can be disabled with a build option I'm not aware of.

keith-packard, 2023-12-07:
> I made a brief attempt to try simply removing all the code referenced by errors caused by `-mgeneral-regs-only`, but it quickly became untenable in tinystdio's format functions (printf,scanf,etc.) since it seems like the code assumes that float types are available. I did manage to build with `-mgeneral-regs-only` if I completely removed libm and tinystdio's formatting support, which is actually good enough I think. I can use [tinyprintf](https://github.com/cjlano/tinyprintf) instead to provide an even simpler printf/sprintf that does not support floating point at all and builds with `-mgeneral-regs-only`.

The printf/scanf code in tinystdio gets compiled many different ways so you can pick your preferred version at link time. You can hack the meson files to skip the variants which support float and double types, and then make sure you select one of the others as the default (which gets bound to the vfscanf and vfprintf names). That should eliminate all actual use of float types. I'm unsure how you'll manage the varargs issue though; it looks like the aarch64 ABI requires storing the float registers into the varargs structure?

> So I think we could just add a meson option that just removes libm/printf from the build and maybe also guarantees that optimized memset/memcpy implementations aren't used (automatically sets PREFER_SIZE_OVER_SPEED). If you want, we could also bundle tinyprintf as a replacement for the stdio formatting for this option, but I think it's also reasonable to expect users of this option to bring their own printf/formatting implementation.

Yeah, removing anything which explicitly uses float values in the API should be all you really need -- that's libm and the string/float conversion functions in stdio (ecvt*, fcvt*, gcvt*, strtod*, strtold*), and a few other misc functions (drand48, erand48, difftime, wcstod*, wcstold*)

zyedidia, 2023-12-07:
When compiling with `-mgeneral-regs-only` it looks like GCC/Clang omit the float registers from the varargs structure on aarch64. I'm currently looking at vfprintf.c which has a SKIP_FLOAT_ARG macro that uses a double type even with floating point support disabled. I'm not sure if it's worth it to try modifying the tinystdio's printf implementation to work with `-mgeneral-regs-only` rather than just use tinyprintf or something else that already works well enough.

keith-packard, 2023-12-09:
> When compiling with `-mgeneral-regs-only` it looks like GCC/Clang omit the float registers from the varargs structure on aarch64. I'm currently looking at vfprintf.c which has a SKIP_FLOAT_ARG macro [...]

I'd work on patching picolibc's printf and scanf bits instead; you aren't far now -- it just needs to stop even skipping float values in your case. That way you'll get a complete stdio implementation instead of having to bodge some other printf code into the library.

BlamKiwi, 2024-04-04:
> When compiling with `-mgeneral-regs-only` it looks like GCC/Clang omit the float registers from the varargs structure on aarch64. I'm currently looking at vfprintf.c which has a SKIP_FLOAT_ARG macro [...]

We use Clang as it has better ARM64 soft float support. GCC was a non-starter, as they disabled float/double entirely in lieu of ARM defining it properly. 

Good news! It is defined properly and is available in LLVM Trunk: 
https://github.com/ARM-software/abi-aa/pull/232
https://github.com/llvm/llvm-project/pull/84146

I don't know when the Linaro/GCC guys plan to implement it.

keith-packard, 2024-04-04:
this is great news. It doesn't really matter what the GCC schedule is; we can use clang to get support integrated into picolibc and if gcc supports it, the picolibc bits should 'just work'.

keith-packard, 2024-06-28:
Check out #753 which should do the trick if you can figure out how to create a compiler-rt library for soft float emulation.

keith-packard, 2024-07-02:
Fixed in #753.

closed 2024-07-02 by keith-packard