picolibc/picolibc #652: Error when enabling options for legacy stdio

zyedidia, 2023-12-25:
I am running into build errors when using the legacy stdio with some additional options enabled. It seems like enabling any of `newlib-fseek-optimization`, `newlib-fvwrite-in-streamio`, or `newlib-unbuf-stream-opt` causes build errors.

Here is my full build script:

```sh
#!/bin/sh

mkdir -p build
cd build

meson \
    -Dtls-model=local-exec \
    -Dmultilib=false \
    -Dpicolib=false \
    -Dpicocrt=false \
    -Dposix-console=true \
    -Dnewlib-global-atexit=true \
    -Dprefix=$PWD/install \
    -Dincludedir=include \
    -Dlibdir=lib \
    -Dformat-default=long-long \
    -Dthread-local-storage=false \
    -Dtinystdio=false \
    -Dnewlib-nano-malloc=false \
    -Dnewlib-fseek-optimization=true \
    -Dspecsdir=none \
    ../picolibc
ninja
ninja install
```

Thanks!

keith-packard, 2023-12-25:
I've hacked up some fixes which at least make the tests pass. Can you explain why you're trying to use the legacy stdio code? It's not well tested at this point, and I'm considering removing it to reduce the effort to track newlib changes in areas  which matter to most picolibc users.

zyedidia, 2023-12-26:
Thanks! I'm using legacy stdio over tinystdio because it has much better performance. I haven't really looked into why, but one culprit is tinystdio's implementation of `fread`, which seems to copy one byte at a time by calling `getc`.

keith-packard, 2023-12-26:
Yes, the tinystdio code is designed to have a very narrow interface to the underlying system so it funnels all operations through getc/putc. We could improve the speed of simple operations like fread/fwrite when using the buffered I/O backend if that would help your use case. Speeding up more complicated things like printf would be quite a bit more complicated though; a lot of the code size benefits in the library are from keeping special cases out of those functions.

zyedidia, 2023-12-26:
Performance improvements for the buffered I/O (fread/fwrite) would be much appreciated. The performance of printf and formatting is not important for my use-case.

asiekierka, 2023-12-26:
I have experimented with optimizations for buffered I/O, but due to lack of time to work on testing (beyond hobbyist uses), as well as the slightly gross quality of the code and making some slightly questionable optimizations, I haven't been able to find time to work towards cleaning it up for a pull request.

If anyone wants to look into it, the WIP branch is here: https://github.com/WonderfulToolchain/wf-picolibc/commits/wonderful/

In general, my goal was to minimize the amount of function calls and memory copies happening within one `fread()` call. To this effect, I:

* Added `read` and `write` functions to `struct __file_ext`, allowing reading/writing data from the file at an interface wider than a char. This allows `fread` and `fwrite` to call such wider methods on buffered I/O.
* Added fast paths in `__bufio_seek` which prevent trashing the entire buffer if the area being seeked to is within the buffer. Some programs use patterns like `fseek(1, SEEK_CUR)` to skip bytes.
* Allowed the new `__bufio_read` and `__bufio_write` to skip the buffer for large reads (say, if you're reading 8KB of data, but the buffer is only 512 bytes, it will prefer to read 8KB directly to the user-provided destination, as opposed to 16 reads to the buffer first). Modern SD cards benefit heavily from doing reads and writes in large batches; in addition, one of the platforms I target has an unusually high additional cost to starting a read and write.

The more questionable optimizations include:

* Word-aligning reads, where possible. This makes sense given the characteristics of platforms I target (mostly ARMv4/ARMv5-based) and the accessed devices (at least one of which operates on a word-granularity bus), but may not be the best idea to enforce always.
* Setting the default buffer size to 0. This makes sense given all file accesses in platforms I target are generally routed through Elm-ChaN's FatFS library, which already keeps a 512-byte buffer per file; keeping two buffers would thus be an unnecessary memory copy. Upstream picolibc should probably not do this. (It also breaks a test which tries to initialize an array of size `BUFSIZ`.)
* Making `fread()` and `fwrite()` not call the internal `read` and `write` methods for each `nitems`. This is probably not fully POSIX-compliant as a result, but the performance benefit was too good to pass up. Upstream picolibc should probably not do this.

keith-packard, 2023-12-26:
> I have experimented with optimizations for buffered I/O, but due to lack of time to work on adequate testing (beyond hobbyist uses), as well as the slightly gross quality of the code and making some slightly questionable optimizations, I haven't been able to find time to work towards cleaning it up for a pull request.

I was thinking of just stacking these performance improvements on top of the existing bufio layer instead -- that already uses read/write instead of getc/putc. Hacking fread/fwrite to access the buffer directly would help a bunch. Additional hacks to bypass the buffer for operations larger than the buffer size could improve performance further.

Digging into the bufio layer to improve performance of __bufio_seek when the target offset is within the existing buffer would also be helpful.