picolibc/picolibc #659: Suggestion: Lightweight ctype implementation

badcf00d, 2024-01-03:
Using the standard ctype header for `isprint()` and friends end up including a fairly chunky 257 byte lookup table.

https://github.com/picolibc/picolibc/blob/368044915e4929ec4cba9f6ec630b2d5e28080ea/newlib/libc/ctype/ctype_.c#L117-L121

These functions could instead be implemented as macros / if statements like
```c
#define isprint(x) (((x) >= ' ') && ((x) <= '~'))
```
 Instead of the lookup table, which ends up being quite a bit smaller, especially if you're only calling a couple of these functions.

Perhaps this could be a config option to choose between lookup-style or conditional-style ctype implementation?

I could work on a PR for this if it sounds useful, but I wanted to check first if there's some particular reason why this should remain as a lookup table to conform to standards etc.?

badcf00d, 2024-01-03:
Ah yeh good point, something like this should work I think?

```c
#define isprint(x)                  \
    ({                              \
        __auto_type _x = (x);       \
        (_x >= ' ') && (_x <= '~'); \
    })
```

Perhaps that `__auto_type` should be a fixed type, but whether it should be int, char, or unsigned char I don't know

badcf00d, 2024-01-03:
Yes of course I didn't mean only change `isprint`, I was just using it as an example.

The use case I'm thinking of is just for a fairly simple bare metal implementation that's just doing a bit of string parsing and wants to use say, `isprint` and `isdigit` or something.

Those can be each done in 6-7 instructions with a load, subtract, a couple of compares, and maybe a branch or two. So something like 14 bytes of thumb instructions. 

Whereas as the moment, the LUT adds 250 bytes ontop of that since you also need a few instructions to index into that LUT and do the bitwise checking.

My application uses `isgraph`, `isxdigit`, and `isdigit`, and moving to macros instead of ctype saved 264 bytes from the final binary

rdiez, 2024-01-03:
If you think it is worth the effort, about the __auto_type above, I would use _int_, since the prototype for isprint(), as defined in ctype.h, is:

int isprint(int c);

keith-packard, 2024-01-03:
This does seem like a good idea -- it could be implemented entirely within the header file so that applications could choose this when using the library, we wouldn't have to choose it while building the library (although, there are few uses of ctype within the library itself that could be conditionalized).

One of the core reasons ctype used the array was to avoid evaluating the parameter more than once in a macro.  This was long before statement expressions or static inline functions.

In any case, I'd prefer static inline functions to fancy statement macros as you get a lot fewer surprises this way, and they're easier to read.

I just read through the C17 spec and it doesn't appear that these are required to be macros (which happens sometimes).

keith-packard, 2024-01-10:
> Should these macros still be there?

Yup, looks like they're part of POSIX. https://pubs.opengroup.org/onlinepubs/009695399/functions/_toupper.html

(yes, this is new to me too)

badcf00d, 2024-01-10:
So it is, how strange

keith-packard, 2024-01-15:
I hacked up some bits; turns out there was a bunch of tricky stuff to make sure the library could be built with the _ctype_b array while still letting internal code use the array-less implementation. Would love to know what you think.

badcf00d, 2024-01-15:
That looks really good! Much more comprehensive than the thing I'd made, I was holding off pushing anything till the llvm inline stuff was sorted.

Only thing I would suggest is l noticed an errant re-implementation of `tolower` here:

https://github.com/picolibc/picolibc/blob/9a1ed2b9807096391217f949359bfc368466df52/newlib/libc/tinystdio/scanf_private.h#L116-L117

It is quite a cute implementation, since the difference between the lower case and upper case letters is 32, you can implement tolower and toupper as bitwise functions `c |= ('a' - 'A');` and `c &= ~('a' - 'A');`

But I tried them out in godbolt targeting armv6. Actually the more verbose `c = c - 'A' + 'a';` is 1 instruction shorter because armv6 can has an immediate `add` instruction, but no immediate `or` instruction. So you have an extra `mov r3, #32` in the bitwise version, rather than just `adds    r4, r4, #32` in the additional version. Armv7 has an immediate `or` instruction, so in that case they're the same length.

So yeh, perhaps remove that re-implementation for consistency, and as an added bonus the armv6 build should get a tiny bit smaller

badcf00d, 2024-01-15:
> There's one additional benefit to using the bitwise operator here -- you can pass a lower case letter to TOLOW and it doesn't get changed. So the code doesn't need to gate use of this macro with an 'isupper' check.

That's true yes, but it breaks for some non-alphabetic characters.
https://godbolt.org/z/7M8fn7svM

> I think this means we should use arithmetic in ctype.h for _tolower and _toupper.

So actually it makes sense to the bitwise one for `_toupper` / lower since those ones are supposed to be unsafe, and without the `isupper` if statement, the bitwise is much shorter. But I think the standard `toupper` / lower needs to work for all inputs, so should probably stick with the arithmetic there

badcf00d, 2024-01-16:
Yes I think that makes sense, I just slightly misread what you said about `_toupper` / lower.

Perhaps the bitwise tolower could be `__tolower` haha. 

So I think everything is fine as it is then in #672? Arithmetic-style `tolower` and `_tolower`, and leave the `TOLOWER` macro in scanf_private.h as-is, given that behaviour where you can safely pass it lower-case characters is useful there.

keith-packard, 2024-01-16:
> Yes I think that makes sense, I just slightly misread what you said about `_toupper` / lower.
>
> Perhaps the bitwise tolower could be `__tolower` haha.
>
> So I think everything is fine as it is then in [...]

Thanks much for your review; I'll go ahead and merge this in.

closed 2024-01-16 by badcf00d