picolibc/picolibc #680: sparse: picolibc/include/complex.h: error: too many errors

marc-hb, 2024-01-23:
[This was found in zephyrproject-rtos/zephyr/issues/63417 (which is fixed now) but it's not related to it. It was also filed earlier in zephyrproject-rtos/zephyr/issues/67035 but I don't think it's related to Zephyr. Also: I'm less confused now]

When trying to use the "sparse" on Zephyr with picolibc, `complex.h` is **alone** responsible for 7,300 lines of warnings out of 10,000 lines total.

In fact complex.h triggers so many warnings that `sparse` aborts when compiling some (most?) files:
```
zephyr/build/modules/picolibc/picolibc/include/complex.h:103:15: error: too many errors
```

Of course one of the reason is very likely that `complex.h` is included in many C files (which is not going to change).

If not all warnings, it should be possible to reduce the number of these warnings to a volume that is easier to filter and that does not cause sparse to abort entirely.

And hey who knows: maybe some of these warnings are finding actual problems?

Reproduction with sparse version 365b95203727. Compile with `make`, add to PATH.
```
Initialize Zephyr
cd zephyr
git checkout 6df6935b674c
west update
 west build  samples/hello_world  -b rpi_4b --  -DZEPHYR_SCA_VARIANT=sparse   -DCONFIG_PICOLIBC_USE_MODULE=y -DCONFIG_PICOLIBC_USE_MODULE=y 
```

Warning: `-DCONFIG_PICOLIBC_USE_MODULE=y` is required to avoid issue #63003

cc: @keith-packard

keith-packard, 2024-01-23:
As far as I can tell, this is just a bug in sparse. It doesn't appear to support `_Complex`? If so, why is something including `<complex.h>`?

I also cannot reproduce this with picolibc 1.8.6 via PR https://github.com/zephyrproject-rtos/zephyr/pull/67891 -- I wonder if something has changed to make it magically work?

marc-hb, 2024-01-23:
I just switched to picolibc version 764ef4e401a8f4c6a86ab723533841f072885a5b (the one in https://github.com/zephyrproject-rtos/zephyr/pull/67891) and it made no difference.

> If so, why is something including <complex.h>

All the files in  `newlib/libm/complex/*.c` unsurprisingly include `complex.h`.

keith-packard, 2024-01-23:
> All the files in `newlib/libm/complex/*.c` unsurprisingly include `complex.h`.

Oh, of course, you're building all of picolibc :-) Sorry for being dense.

I'm using sparse 0.6.4 and not having any issues though. It sure looks like this is just a bug in sparse.

marc-hb, 2024-01-23:
I downgraded to sparse 0.6.4 + a cherry-pick of 3ddf02ddcb9 to avoid the infinite loop in tls.c (https://github.com/zephyrproject-rtos/zephyr/issues/63417) and the reproduction is exactly the same. I'm very surprised you don't see these thousands of warnings.

> It doesn't appear to support _Complex? 

So far I vaguely assumed sparse didn't like the underscore. Now I just found that full symbol `_Complex` is actually part of a "blocklist", see https://github.com/lucvoo/sparse-dev/commit/16ae3ac5c5d07 ("keyword: explicitly add C99 & C11 keywords"). I'm still not sure what that means: is `sparse` unsuited to scanning libc implementations? @lucvoo could you explain?

Anyway it's not just `_Complex`, far from it. I ran again with the quick hack below. The `_Complex` warning is gone but `complex.h` still hits the "too many errors in complex.h" threshold (each time on each .c file) and it still produces about 7000 lines alone out of a 10,000 total.

keith-packard, 2024-01-23:
> So far I vaguely assumed sparse didn't like the underscore. Now I just found that full symbol `_Complex` is actually part of a "blocklist", see [lucvoo/sparse-dev@16ae3ac5c5d07](https://github.com/lucvoo/sparse-dev/commit/16ae3ac5c5d07) ("keyword: explicitly add C99 & C11 keywords"). I'm still not sure what that means: is `sparse` unsuited to scanning libc implementations? @lucvoo could you explain?

That commit makes it pretty clear that sparse doesn't support `_Complex` at all, so there's no way you're going to get it to deal with any file using those datatypes (`double _Complex`, `float _Complex` and `long double _Complex`).

The cmake configuration *should* be autodetecting whether the compiler supports this or not; it seems like there's an assumption that sparse supports the same language as the target compiler? Because when I run the command described above, it sets `_HAVE_COMPLEX` to `TRUE`, which isn't correct for sparse.

Maybe we just need an explicit parameter to disable `_Complex` support in picolibc. Alternatively, sparse could be fixed to actually support C11?

lucvoo, 2024-01-23:
On Tue, Jan 23, 2024 at 10:34:35AM -0800, Keith Packard wrote:
> > So far I vaguely assumed sparse didn't like the underscore. Now I just found that full symbol `_Complex` is actually part of a "blocklist", see ***@***.***(https://github.com/lucvoo/sparse-dev/commit/16ae3ac5c5d07) ("keyword: explicitly add C99 & C11 keywords"). I'm still not sure what that means: is `sparse` unsuited to scanning libc implementations? @lucvoo could you explain?
> 
> That commit makes it pretty clear that sparse doesn't support `_Complex` at all, so there's no way you're going to get it to deal with any file using those datatypes (`double _Complex`, `float _Complex` and `long double _Complex`).

Not exactly, it just inhibits you to use the '_Complex' as the name of
a variable/function/typedef/etc. But it's just a detail,
it's indeed the case that Sparse doesn't support complex types (its
origin as a tool for the Linux kernel explains this).
 
> The cmake configuration *should* be autodetecting whether the compiler supports this or not; it seems like there's an assumption that sparse supports the same language as the target compiler?

In truth, this is the goal.

> Because when I run the command described above, it sets `_HAVE_COMPLEX` to `TRUE`, which isn't correct for sparse.
> 
> Maybe we just need an explicit parameter to disable `_Complex` support in picolibc. Alternatively, sparse could be fixed to actually support C11?

As far as I understand support for complex types is optional in C11 and
can/should be tested via the macro __STDC_IEC_559_COMPLEX__ (which Sparse
doesn't define, of course).

But sure, ideally, Sparse should support complex types (and the different
vectors type and a few other things). It shouldn't be very hard, it's just
a question of time and priorities.

Best regards,
-- Luc

keith-packard, 2024-01-23:
> As far as I understand support for complex types is optional in C11 and can/should be tested via the macro __STDC_IEC_559_COMPLEX__ (which Sparse doesn't define, of course). But sure, ideally, [...]

Yes, and the picolibc build system checks to see if the 'toolchain' supports `_Complex` during the configuration process. It seems that the issue here is that those checks aren't using sparse. Thanks for your explanation.

marc-hb, 2024-01-23:
> > > it seems like there's an assumption that sparse supports the same language as the target compiler?

> > In truth, this is the goal.

> Yes, and the picolibc build system checks to see if the 'toolchain' supports _Complex during the configuration process. It seems that the issue here is that those checks aren't using sparse

Pretty much all static C/C++ analyzers I've used work the same: they inject themselves in the middle of the build system to gather compiler invocations: source file names, options, -D macro definitions etc. For zephyr and sparse see this small and insightful commit https://github.com/zephyrproject-rtos/zephyr/commit/91902c5fd4db756. Note the REAL_CC magic there, also explained here: https://github.com/lucvoo/sparse-dev/blob/1cf3d98cf171/cgcc.1

So from a build perspective I'm afraid there is no such thing as "the sparse toolchain" or "the sparse build" that could be configured differently. I don't think anyone would like to analyze sources _different_ from what they actually use.

On the other hand, is there a simple way to disable complex.h _from the Zephyr build system/Kconfig_? As opposed to auto-detecting it from the toolchain. Not a solution for projects actually using complex numbers but a great solution for everyone else. It would also save some build time; never hurts.

>  It was also filed earlier in https://github.com/zephyrproject-rtos/zephyr/issues/67035 but I don't think it's related to Zephyr.

Should we close this picolibc issue and move back to a re-opened 67035?

marc-hb, 2024-01-23:
> So from a build perspective I'm afraid there is no such thing as "the sparse toolchain" or "the sparse build" that could be configured differently. I don't think anyone would like to analyze sources different from what they actually use.

For completeness: some developers may be selfishly interested in warnings only in the subset/component  _they_ wrote. From past experience again, filtering this at build time is tricky to achieve: you have to fight the build system. This also hides  information from analyzers which have a global view and don't just look at each .c file one by one. See an example and good explanation there: https://www.synopsys.com/blogs/software-security/coverity-and-issues-in-third-party-code.html

So for these "selfish" people it's simpler, safer and generally better to exclude warnings from picolibc or other _after_ analysing the complete project as is. Display filters, no build-time filter.

With sparse we have to filter after analysis anyway due to `-Werror` complexities: https://marc.info/?l=linux-sparse&m=170571398021000&w=2

So filtering warnings in a per-component basis can be done at the same, post-analysis time; two birds with one stone.

keith-packard, 2024-01-23:
There's currently no way to disable `_Complex` usage in picolibc via the CMake configuration; autodetection of what the toolchain supports has been sufficient so far. It would be easy enough to add, if someone wanted to do that.

closed 2026-11-10 by keith-packard