picolibc/picolibc #683: Linking fails when C++ new operator is used

EmergReanimator, 2024-02-11:
Greetings!

I am evaluating the picolib as newlib-nano replacement for mixed C/C++ project.
While doing the evaluation I faced some odd issue with C++ (**new** operator) support.
ARM GCC fails to link the sample code below as soon as **new** operator is used. It succeeds with **malloc** however.

### How should I properly link this code with `picolib` library?

---

## Here is the C++ source code that causes undefined errors during the linking

```c++
#include <new>
#include <cstdlib>

int main()
{
#if (true)
	int *p = new (std::nothrow) int();
#else
	int *p = (int *)malloc(sizeof(int));
#endif

	return 0;
}
```
----

## Compilation command line
```
arm-none-eabi-g++ -Wall -Wextra  -mcpu=cortex-m3 -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -ffreestanding -g3 -isystem /usr/lib/picolibc/arm-none-eabi/include -std=c++17 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -o picolibcpp-test main.cpp --specs=picolibcpp.specs
```

----

## Compiler messages
```
main.cpp: In function 'int main()':
main.cpp:7:14: warning: unused variable 'p' [-Wunused-variable]
    7 |         int *p = new (std::nothrow) int();
      |              ^
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/picolibc/arm-none-eabi/lib/thumb/v7-m/nofp/libc.a(libc_stdlib_abort.c.o): in function `abort':
./arm-none-eabi/../../../newlib/libc/stdlib/abort.c:63: undefined reference to `_exit'
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/picolibc/arm-none-eabi/lib/thumb/v7-m/nofp/libc.a(libc_signal_signal.c.o): in function `raise':
./arm-none-eabi/../../../newlib/libc/signal/signal.c:132: undefined reference to `getpid'
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: ./arm-none-eabi/../../../newlib/libc/signal/signal.c:132: undefined reference to `kill'
collect2: error: ld returned 1 exit status
```

---

## Version information
`picolib-arm-none-eabi` package version: 1.8-1
`gcc-arm-none-eabi` version: 15:12.2.rel1-1

keith-packard, 2024-02-11:
When the implementation of `new` in libstdc++ gets a `NULL` value returned from `malloc`, it looks for an exception handler. If there isn't one, it calls `__builtin_abort`, which is just some gcc magic around a call to `abort`.

The posix spec for `abort` (https://pubs.opengroup.org/onlinepubs/9699919799/functions/abort.html) says that `abort` effectively calls `raise(SIGABRT)`, and that it must not return. The picolibc implementation follows the call to `raise` with a call to `_exit`.

Picolibc version 1.8 implements `raise(signal)` as `kill(getpid(), signal)`. Newer versions call the signal handlers directly, implementing `SIG_DFL` with a call to `_exit(128+signal)`.

To make this work in your environment, you need to provide implementations for `_exit`, `kill` and `getpid`. If you move to a newer version of picolibc, you'd only need to provide `_exit`.

Picolibc can't implement these functions internally as they depend upon the operating environment of the application.

There is some documentation about what kind of support you need to run picolibc (https://github.com/picolibc/picolibc/blob/main/doc/os.md), but that doesn't mention abort.

EmergReanimator, 2024-02-11:
> When the implementation of `new` in libstdc++ gets a `NULL` value returned from `malloc`, it looks for an exception handler. If there isn't one, it calls `__builtin_abort`, which is just some gcc magic around a call to `abort`.

Thanks for quick reply!
Now it makes more sense.

However I wonder whether `abort` is really needed in this case. The [exception-free](https://en.cppreference.com/w/cpp/memory/new/nothrow) version of `new` operator is used this code sample. It returns the `nullptr` instead of raising the C++ exception, so `abort` is not supposed to be called. Did I miss anything here?

keith-packard, 2024-02-11:
Ah, I think I understand. *your* call to `new` won't call abort, but take a look at how `new` is implemented:
```
void*
operator new[] (std::size_t sz, const std::nothrow_t&) noexcept
{
  __try
    {
      return ::operator new[](sz);
    }
  __catch (...)
    {
      return nullptr;
    }
}
```
It's a simple wrapper around the regular new operator which catches the exception and dumps it on the floor. And throughout that exception handling adventure, lots of calls to the regular version of `new` happen as well.

I'd suggest implementing your own version of `operator new[] (std::size_t sz, const std::nothrow_t&) noexcept` that calls malloc directly. This works for me:

```
void*
operator new[] (std::size_t sz, const std::nothrow_t&) noexcept
{
	return malloc(sz);
}
```

EmergReanimator, 2024-02-14:
I can confirm, that the overloading of `new` operator solves build issue.
@keith-packard Thanks for your help!

closed 2024-02-14 by EmergReanimator