picolibc/picolibc #687: return in main function, program can't exit, must call exit(0) to exit ?

rockcarry, 2024-02-18:
I found return in main function, the program can't exit.

```
int main(void)
{
    printf("hello world !\n");
    return 0;
}
```

I have implemented the _exit function:

```
void _ATTRIBUTE ((__noreturn__)) _exit(int code)
{
    (void) code;
    __asm__("li x17, 93");
    __asm__("ecall");
    while (1);
}
```

but after return 0; the _exit had not been called.

How to setup build configure or modify code, to fix this issue ?

keith-packard, 2024-02-18:
The default crt0 startup code doesn't call `exit` (or `_exit`) because most embedded systems can't do anything useful at that point anyways. You can select the "hosted" startup code which does call exit by adding the `--crt0=hosted` to the gcc command line when linking.

Check out the documentation about the crt0 options here: https://github.com/picolibc/picolibc/blob/main/doc/using.md

rockcarry, 2024-02-19:
Hi Keith

Thank you for your help. I add --crt0=hosted to Makefile. It perfectly fixes this problem.
I test not on a real embedded systems SoC, but on my ffvm virual machine, so I need to exit from main.

Belows are my ffvm project:
riscv32imc virtual machine written in 500 lines C.
https://github.com/rockcarry/ffvm

test programs that can run on ffvm:
https://github.com/rockcarry/riscv32-test

toolchain for ffvm and riscv32-test (build via crosstool-ng)
https://github.com/rockcarry/riscv32-toolchain

rockcarry, 2024-02-19:
This issue can be closed.

closed 2024-02-19 by rockcarry