picolibc/picolibc #689: Use of single tls PHDR causes giant program header load command with GNU ld

relativityspace-jsmith, 2024-02-27:
Looking at the default picolibc.ld linker script, it uses a single `:tls` PHDR section for both .tdata and .tbss.  However, I think that this confuses the linker, because the two different TLS sections are marked with two different LMAs -- .tdata is AT flash and .tbss is AT ram.

This issue can be observed by compiling the tls.c test and looking at it with readelf.  I'm using a toolchain built by crosstool 1.26, which contains gcc 13 and picolibc 1.8.1.

```
> arm-picolibc-eabi-gcc tls.c -o tls.elf --specs=picolibc.specs -l c -l semihost -Wl,-print-memory-usage
Memory region         Used Size  Region Size  %age Used
           flash:       19948 B         4 MB      0.48%
             ram:        5160 B         2 MB      0.25%
> arm-picolibc-eabi-readelf -Wl tls.elf

Elf file type is EXEC (Executable file)
Entry point 0x78
There are 4 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x001000 0x00000000 0x00000000 0x04c68 0x04c68 R E 0x1000
  LOAD           0x000200 0x20000200 0x20000200 0x00000 0x01228 RW  0x1000
  LOAD           0x006000 0x20000000 0x00004c68 0x00184 0x00184 RW  0x1000
  TLS            0x006100 0x20000100 0x00004d68 0x00084 0x1fffb5a0 RW  0x100

 Section to Segment mapping:
  Segment Sections...
   00     .init .text
   01     .tbss_space .bss .stack
   02     .data .tdata
   03     .tdata .tbss
```
Note the massive MemSiz under the TLS segment -- it seems to think that the TLS region is almost 512MiB in size!

Ordinarily, this issue seems benign, as the program headers are not actually used on bare metal platforms.  However, on one of our internal linker scripts, this was also causing the bin file generated by `objcopy -O binary` to equal the size of the TLS region!  Unfortunately, I can't seem to reproduce this with simpler linker scripts or with the picolibc.ld linker script, but I do think it's related to this load command issue.

If I change the PHDRS segment to
```
PHDRS
{
	text PT_LOAD;
	ram PT_LOAD;
	ram_init PT_LOAD;
	tdata PT_TLS;
	tbss PT_TLS;
}
```
and change the .tdata and .tbss sections to use the new headers, the issue seems to go away:
```
> arm-picolibc-eabi-readelf -Wl tls.elf

Elf file type is EXEC (Executable file)
Entry point 0x78
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x001000 0x00000000 0x00000000 0x04c68 0x04c68 R E 0x1000
  LOAD           0x000200 0x20000200 0x20000200 0x00000 0x01228 RW  0x1000
  LOAD           0x006000 0x20000000 0x00004c68 0x00184 0x00184 RW  0x1000
  TLS            0x006100 0x20000100 0x00004d68 0x00084 0x00084 RW  0x100
  TLS            0x006200 0x20000200 0x20000200 0x00000 0x00108 RW  0x100

 Section to Segment mapping:
  Segment Sections...
   00     .init .text
   01     .tbss_space .bss .stack
   02     .data .tdata
   03     .tdata
   04     .tbss
```
Doing this also fixed the giant bin file being generated by objcopy for our internal linker script.  However, I don't feel like I really understand *why* this works, so I'm a little uneasy using it without looking into it further.  If there's someone who better understands the behavior of PHDRs with PT_TLS, I'd really appreciate your insight into what could be going on and whether this is the correct way to fix this issue.

keith-packard, 2024-02-27:
I wonder if it's somehow assuming that the entire TLS segment needs initialization data and spans the whole physical address range.

In any case, it's probably reasonable to split these up into two phdr blocks; it shouldn't cause any issue and it sounds like it makes some tools less confused.

relativityspace-jsmith, 2024-02-27:
Glad you agree!  It does seem pretty mysterious, I feel like the behavior of TLS is really badly documented outside of picolibc itself...

One other thing I wanted to ask about... I was upconverting some old linker scripts from picolibc 1.7 to 1.8, and I noticed that not having the `AT>ram` marker on the .tbss section causes the linker to go a bit crazy and (this time) make the :ram section huge.  Do you happen to know why this marker is needed / why it was added?  Logically it feels like it shouldn't be needed, because the .tbss section isn't actually loaded from anywhere in the program image.  And yet, the linker seems to really want it in order to have correct behavior.

keith-packard, 2024-02-27:
> One other thing I wanted to ask about... I was upconverting some old linker scripts from picolibc 1.7 to 1.8, and I noticed that not having the `AT>ram` marker on the .tbss section causes the linker [...]

I have no idea why it's needed, but maybe the linker defaults to using whatever AT was in the last chunk? The linker remains a mystery to me; I generally poke at the file until the output works ...

> Anyway, thanks for the quick response on this, just wanted to make sure I wasn't missing anything!

Thanks. Let me know if #690 looks correct.

relativityspace-jsmith, 2024-03-04:
Looks like this is fixed now!

closed 2024-03-04 by relativityspace-jsmith