picolibc/picolibc #695: Add sample support for RP2040 multicore TLS

sgstreet, 2024-03-06:


sgstreet, 2024-03-11:
@keith-packard here is a reasonable but untested implementation of TLS support for the RP2040, the key is the [SIO CPUID register on page 24 of the datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf).  Unfortunately this wastes the default TLS block allocated int the default piclibc linker script. This could of course be fixed.

I'm not using this approach as I also need the concept of `core_local` data to support my SMP scheduler so I have implemented a custom system startup and matching linker scripts tuned to my application.  Bypassing the libc CRT and linker scripts is pretty typical of bare metal application in my experience.

The nonsense you see in `_init_tls` is driven by the the pico-sdk overrides of `__aeabi_memset` and `__aeabi_memcpy` with ROM based versions, which are not guaranteed to have been initialized before the TLS. 

```
#include <tls.h>

#include <unistd.h>

#define PREINIT_SECTION(SECTION) ".preinit_array." #SECTION
#define PREINIT(NAME) void *__attribute__((used, section(PREINIT_SECTION(NAME)))) preinit_##NAME = NAME

extern char __arm32_tls_tcb_offset;
extern char __tdata_source[];
extern char __tdata_size[];
extern char __tbss_size[];
extern char __tbss_offset[];
extern char __tls_size;

static void *__tls[2] = { 0, 0 };

__attribute__((naked)) void *__aeabi_read_tp(void)
{
	asm volatile (
		".syntax unified\n"
		"push {r1, lr}\n"
 		"ldr r1, =0xd0000000 /* Address of SIO->CPUID */ \n"
 		"ldr r1, [r0, #0]    /* Now we have the active core */ \n"
		"lsls r1, r1, #2     /* Array offset is 4 bytes */ \n"
		"ldr r0,=__tls       /* Load the address of the tls pointers */ \n"
		"ldr r0, [r0, r1]    /* Load the current TLS pointer for the active core */ \n"
		"pop {r1, pc}\n"
	);
}

void _set_tls(void *tls)
{
	/* Pointer to SIO->CPUID, the active core, this is similar to FCSEIDR */
	volatile uint32_t *core = (uint32_t *)(0xd0000000);

	/* Save the new tls pointer */
	__tls[*core] = tls - ((size_t)&__arm32_tls_tcb_offset);
}

void _init_tls(void *tls)
{
	char *src;
	char *dst;
	char *end;

	/* Copy tls initialized data, done like this due to Pico SDK setup */
	src = __tdata_source;
	dst = tls;
	end = tls + (size_t)__tdata_size;
	while (dst < end)
		*src++ = *dst++;

	/* Clear tls zero data */
	dst = tls + (size_t)__tbss_offset;
	end = tls + (size_t)__tbss_offset + (size_t)__tbss_size;
	while (dst < end)
		*dst++ = 0;
}

static void tls_setup(void)
{
	/* Get some memory for the initial tls blocks */
	__tls[0] = sbrk((intptr_t)__tls_size);
	__tls[1] = sbrk((intptr_t)__tls_size);

	/* Initialize them */
	_init_tls(__tls[0]);
	_init_tls(__tls[1]);
}
PREINIT(tls_setup);
```

keith-packard, 2024-03-11:
Yup, I expect lots of projects (most?) will end up with a custom crt0 and linker configuration. The goal of the samples provided with picolibc is to a) allow for stand-alone CI and b) provide a starting point for people building software.

Thanks for the description of how the TLS stuff works on this platform; I wondered if it might be as simple as using the core ID register to select an appropriate value stored in memory.

You could probably hack the linker script to allocate TLS blocks statically, or (possibly easier) simply remove the static block entirely -- that's the "normal" way to use TLS where the runtime system (`tls_setup` in this case) does the allocation. You can also allocate them at the top of the stack; another totally normal place to stick these things. Lots of options; maybe the picolibc docs should demonstrate a couple of them...

sgstreet, 2024-03-11:
> Thanks for the description of how the TLS stuff works on this platform; I wondered if it might be as simple as using the core ID register to select an appropriate value stored in memory.

See below.
 
> You could probably hack the linker script to allocate TLS blocks statically, or (possibly easier) simply remove the static block entirely -- that's the "normal" way to use TLS where the runtime [...]

Sorry for any confusion.  I have multiple working TLS implementations for the RP2040 and I thought from our discussion on PR #693 that you were [curious](https://github.com/picolibc/picolibc/pull/693#issuecomment-1980169216) to see an example.

Feel free to close this issue. :)

keith-packard, 2024-03-11:
Yes, I was curious to see the working code. Now to ponder how best to provide this as an example in the picolibc repository. We could make it a picolibc build option; that seems like the easiest way to at least make the code visible.

keith-packard, 2024-03-11:
I've pulled the `__aeabi_read_tp` and `_set_tls` bits from this and stuck them into #698. I think the rest of the changes here are OS specific, so they shouldn't live inside the picolibc source code. Let me know if this looks reasonable; it would be even more awesome if you could make sure it works as I don't have any of these chips to test with and QEMU doesn't appear to support it (yet?).

sgstreet, 2024-03-11:
> I've pulled the `__aeabi_read_tp` and `_set_tls` bits from this and stuck them into #698. I think the rest of the changes here are OS specific, so they shouldn't live inside the picolibc source [...]

I will test PR#698 in-situ with the **pico-sdk** to get community provided chip initialization in to a running application. I will need a couple of days. I do not think there is QEMU support at this time. Does **picolibc** have QEMU tests for the **cortex-m0+** MCU?  If so can you point me at the QEMU usage?

sgstreet, 2024-03-11:
> I'm not sure what to do with the `_init_tls` bits; if `memset` and `memcpy` aren't available, presumably you're also open-coding the regular `.data` and `.bss` initialization sequences?

While I build up the test for PR#698, I try to get TLS initialization after the **__aebi_memXXX** initialization sequence and enable usage of memset/memcpy in optimized builds.

> You might consider copying the sample linker script hacks to get one of the TLS blocks initialized along with those, and then just allocating another chunk for whatever code runs on core 1; it looks like you have to manually start core 1 from core 0? Dunno if that could work or not; feel free to ignore all of this :-)

Yea, I have code which uses ARM CMSIS multi-block .data and .bss initialization sequence to also initialize the TLS block of both cores at system reset.  Core 1 repeats the TLS init sequence any time it restarts.  None of this has been sent upstream to the RP2040 team, so far not a lot of interest, but still working on it.

keith-packard, 2024-03-11:
> > I've pulled the `__aeabi_read_tp` and `_set_tls` bits from this and stuck them into #698. I think the rest of the changes > I will test PR#698 in-situ with the **pico-sdk** to get community provided chip initialization in to a running application. I will need a couple of days. I do not think there is QEMU support at this time. Does **picolibc** have QEMU tests for the **cortex-m0+** MCU? If so can you point me at the QEMU usage?

No hurry; just add comments to #698 when you get a chance. Picolibc runs tests on all of the arm variants supported by the ARM GCC toolchain using qemu. You can see how that works by looking at https://github.com/picolibc/picolibc/blob/main/scripts/run-arm

PR #698 (2024-03-18): picolib: Add (optional) rp2040 support to ARM TLS code
The RP2040 has two cortex-m0 cores. Allocate an array of TLS base pointers in __tls and then use the CPUID register to index into that.

Closes: #695
closed 2024-03-18 by keith-packard