picolibc/picolibc #696: Sample of using cmake to build picolibc

sgstreet, 2024-03-11:
Can you point me at the documentation or example of using cmake to build picolibc? I understand that the picolibc cmake support is limited to a single lib configuration, but I not clear on how to set that up.  For example what is the format of the multilib-list build option?  Is the a sample cmake toolchain file available?

keith-packard, 2024-03-11:
The cmake stuff really only exists to allow Zephyr to build it as a module. I don't really want to create a complete parallel build system using cmake as meson really does what picolibc needs a lot better. It's pretty easy to invoke meson to build picolibc from cmake. We had that prototyped for Zephyr but they didn't want to take on a dependency on ninja, and meson doesn't plan on providing make support ever.

sgstreet, 2024-03-11:
So, assuming -GNinja on the **cmake** command line, is the **picolibc** **cmake** infrastructure functional?  I'm asking because the RP2040 [RP2040 pico-sdk](https://github.com/raspberrypi/pico-sdk) uses **cmake** for its build generator and the code base builds successfully with a Ninja output configuration.  

My goal is to integrate **picolibc** into the RP2040 ecosystem.

keith-packard, 2024-03-11:
Zephyr didn't want to use meson because meson requires ninja. Picolibc's Zephyr-specific cmake support works fine with either ninja or make. But, using that code outside of the very specific Zephyr module use case is pretty much use-at-your-own-risk; it does get tested in picolibc CI, but not as extensively as the meson builds.

If the RP2040 ecosystem is fine taking a dependency on meson+ninja, I'd encourage you to integrate picolibc using meson via cmake's `ExternalProject` bits as that will ensure you're getting picolibc built in the usual way along with supporting all of the picolibc build configuration options.

But, if you want to play with the cmake stuff and support it yourself, I'd be happy to take whatever patches you generate and work on getting them integrated, assuming they don't impact how Zephyr works.

sgstreet, 2024-03-22:
Ok, I will setup up an `ExternalProject` to build picolibc.  Unfortunately meson skills of not very good.  Could you post a setup incantation to build just picolibc for the cortex-m0+ MCU?

sgstreet, 2024-03-22:
> Here's the cross compilation control script; I started with the cortex-a9 sample and switched the cpu name to m0 and adjusted the memory addresses to match the qemu m0 model so that I could run the tests:
>
> cross-cortex-m0-none-eabi.txt<https://github.com/picolibc/picolibc/files/14726061/cross-cortex-m0-none-eabi.txt>
>
> Invoke this with
>
> $ mkdir build-m0
> $ cd build-m0
> $ meson setup  --cross-file cross-cortex-m0-none-eabi.txt -Dtests=true -Dmultilib=false ..
> $ ninja test
>
> The scripts/GeneratePicolibcCrossFile.sh shell script helps generate the cross files, but I've found it easier in this case to start with an existing one which was close and edit it.

Thanks!!!

keith-packard, 2024-03-27:
Looks like this has been resolved by using the cmake `ExternalProject` command to build picolibc with meson.

closed 2024-03-27 by keith-packard