picolibc/picolibc #697: picolibc read() assumes stdin will never encounter EOF

pm215, 2024-03-11:
The picolibc read() currently looks like this:
```
ssize_t
read(int fd, void *buf, size_t count)
{
        if (fd == 0) {
                int ch = sys_semihost_getc((FILE *) 0);
                *(char *) buf = ch;
                return 1;
        }
	fd = _map_stdio(fd);
	uintptr_t ret = sys_semihost_read(fd, buf, count);

	ssize_t got = count - (ssize_t) ret;
	return got;
}
```
where sys_semihost_getc() is a wrapper for the SYS_READC semihosting call.

This read() implementation assumes that stdin will never encounter EOF, because it always returns 1 in the 'fd == 0' codepath. As it happens, on QEMU's implementation of semihosting at the moment, the effect is that the guest (and QEMU) hang on the read: see bug https://gitlab.com/qemu-project/qemu/-/issues/1963 .

The semihosting spec for SYS_READC is regrettably non-specific about what should happen if the console is at EOF, so different implementations are likely to handle it (or mishandle it) in different ways. For robustness, I think it would be preferable for picolibc not to try to special-case fd 0 like this, but instead use SYS_READ for it as well as for other filedescriptors. I believe this is what newlib does.

keith-packard, 2024-03-11:
We could do. It seems like we'd need to switch semihosting to using buffered I/O though so that stdin would actually go through `read`? I'm not sure what other stdio paths could reach this function. Are you using the POSIX `read` function directly?

pm215, 2024-03-11:
I'm just forwarding on what I noticed when investigating the QEMU bug report. The reporter there quotes the test program they used as calling scanf() to read from stdin. I haven't cross-checked that that really does wind up in SYS_READC the way the reporter says it does, or tried to get a backtrace on the guest side.

keith-packard, 2024-03-11:
Thanks for the clarification. Changing `read` won't help this case as `scanf` will end up in `sys_semihost_getc` instead. Maybe a build option that uses the POSIX interfaces provided by the semihosting code rather than the current code which uses `SYS_READC` and `SYS_WRITEC`. That would make the semihosting code use a lot more memory for I/O buffers, but would provide EOF semantics for stdin as well as making stdout a lot more efficient.

keith-packard, 2024-03-11:
Ok, I went and played with this for a while today and finally remembered why picolibc works this way. The trouble is that qemu's implementation of `SYS_READ` will *never* block waiting for input, so you cannot use it with anything other than a regular file, not with ttys nor pipes. If qemu ever changes this, we can reconsider how picolibc uses the API.

keith-packard, 2024-03-27:
I added a comment to the qemu issue about this noting that the lack of blocking I/O for `SYS_READ` makes it unsuitable for stdin. https://gitlab.com/qemu-project/qemu/-/issues/1963