picolibc/picolibc #705: Invalid offsets for TLS variables when linking with lld

domin144, 2024-03-26:
Hello @keith-packard,

Since this change https://github.com/picolibc/picolibc/commit/fe5191e347427a00e0b5aba34e002f914e6e5ad8, I am observing failures in `tls` and `rand` tests, when compiled with clang. 

Comparing the headers in ELF files before and after the change (dumped with `arm-none-eabi-objdump --all-headers tls`):
before:
```
     TLS off    0x00030100 vaddr 0x61000100 paddr 0x6000343c align 2**8
         filesz 0x00000100 memsz 0x00000208 flags rw-
```
after:
```
     TLS off    0x00030100 vaddr 0x61000100 paddr 0x6000343c align 2**8
         filesz 0x00000084 memsz 0x00000084 flags rw-
     TLS off    0x00030200 vaddr 0x61000200 paddr 0x61000200 align 2**8
         filesz 0x00000000 memsz 0x00000108 flags rw-
```
Effect is as expected: the tls header goes just after the tls_init header. Some bytes in the ELF file are saved.

However, looking at symbols:
before:
```
00000200 g       .tbss    00000004 overaligned_bss_var
00000100 g       .tbss    00000004 bss_var
00000000 g       .tdata    00000004 data_var
00000080 g       .tdata    00000004 overaligned_data_var
```
after:
```
00000100 g       .tbss    00000004 overaligned_bss_var
00000000 g       .tbss    00000004 bss_var
ffffff00 g       .tdata    00000004 data_var
ffffff80 g       .tdata    00000004 overaligned_data_var
```
It looks as if the offsets, were now calculated relative to the tls header (the one, which should contain .tbss section), not the tls_init, which comes first. Hence, e.g. the `data_var` symbol received a negative address.

I experimented a little with `lld`. Changing this fragment in `lld/ELF/Writer.cpp`:
```
    for (PhdrEntry *p : mainPart->phdrs)
      if (p->p_type == PT_TLS)
        Out::tlsPhdr = p;
```
to:
```
    for (PhdrEntry *p : mainPart->phdrs)
      if (p->p_type == PT_TLS) {
        Out::tlsPhdr = p;
        break;
      }
```
fixes the issue for this example, but does not seem to be generic solution. Looking at the comments in `lld` source code, I have an impression no one ever expected there to be more than one `PT_TLS` marked `PHDR` entry. The ELF docs [1] seem to assume, that all the TLS symbols go to single program entry.

Could you advice on how should this be handled? I could file a bug report against `lld`, but I would have to justify the "expected behavior" somehow.

[1] https://www.sco.com/developers/gabi/

keith-packard, 2024-03-26:
Sounds like we need to make the split tls/tls_init change conditional on not using lld then. We did that to solve a problem with some tooling that created an enormous initialized data block because the initialized TLS bits were in the same section as the uninitialized bits. I'm not sure we can make everyone happy with one solution though...

keith-packard, 2024-03-26:
looks like current lld is able to link the samples now, so I can switch the CI bits from GNU ld to lld to catch these issues.

keith-packard, 2024-03-26:
Let me know if #706 works for you -- that switches the linker script back to a single TLS segment for lld compatibility. That PR also switches the clang samples to lld so that we'll get testing with that linker.

domin144, 2024-03-26:
Yes, the changes fix the issue for me. Thanks!

keith-packard, 2024-03-26:
Fixed by #706

closed 2024-03-26 by keith-packard