picolibc/picolibc #738: tinystdio/fseek.c implementation is problematic

kaishaoshao, 2024-05-30:
We can test this with the following code:
Running results ： Error seeking file: Success
```
#include <errno.h>
#include <stdio_private.h>
#define FSEEK_TYPE __off_t
#define fseek fseeko1

int fseek(FILE *stream, FSEEK_TYPE offset, int whence)
{
        struct __file_ext *xf = (struct __file_ext *) stream;
        if ((stream->flags & __SEXT) && xf->seek) {
                if ((xf->seek) (stream, (__off_t) offset, whence) >= 0) {
                        stream->flags &= ~__SEOF;
                        (void) __atomic_exchange_ungetc(&stream->unget, 0);
                        return 0;
                }
                return -1;
        }
	errno = ESPIPE;
	return -1;
}

#define EXIT_SUCCESS 0
int main()
{
    FILE *fp;
    char *filename = "testfile.txt";
    off_t offset = 100; 
    char buffer[1024];
    size_t read_size;

    fp = fopen(filename, "w+");
    if (fp == NULL) {
        perror("Error opening file");
        return EXIT_FAILURE;
    }

    fputs("Hello, world! This is a test file.\n", fp);
    fclose(fp);

 
    fp = fopen(filename, "r+");
    if (fp == NULL) {
        perror("Error opening file");
        return EXIT_FAILURE;
    }

    if (fseeko1(fp, offset, SEEK_SET) != 0) {
        perror("Error seeking file");
        fclose(fp);
        return EXIT_FAILURE;
    }

    read_size = fread(buffer, sizeof(char), sizeof(buffer), fp);
    if (ferror(fp)) {
        perror("Error reading file");
        fclose(fp);
        return EXIT_FAILURE;
    }

    printf("Read data: %.*s\n", (int)read_size, buffer);

    fclose(fp);

    return EXIT_SUCCESS;
}
```

keith-packard, 2024-06-27:
I'm not sure what problem you're trying to report here. Please indicate which operation failed and describe the incorrect result as well as the correct result. If you can refer to a specific part of the C language specification to confirm your interpretation of the data, that would be helpful but is not required.

kaishaoshao, 2024-07-11:
Sorry, I encountered this problem when I linked the c library of picolibc to compile the riscv program and used the emulator to simulate it. I will confirm that it is the emulator's problem later. Please forgive my previous non-standard expression

closed 2024-07-11 by kaishaoshao

keith-packard, 2024-07-11:
No worries; let us know if you find a bug!