picolibc/picolibc #773: ld.lld version check (for -mno-relax) seems to be broken

ArcaneNibble, 2024-07-15:
When building the latest main of picolibc using a relatively-recent (but patched) main of LLVM (i.e. LLVM 19), ld.lld seems to be getting detected as older than version 15:

```text
Cross compiler sanity tests disabled via the cross file.
C compiler for the host machine: /Users/<username>/code/llvm-project/build-native/bin/clang --sysroot /Users/<username>/code/toolchain-garbage/sysroot --target=riscv32-unknown-none-elf -march=rv32imac_xwchc -ffunction-sections -fdata-sections -fno-ident -nostdlib (clang 19.0.0 "clang version 19.0.0git (https://github.com/llvm/llvm-project.git 69bd8547e9cc1268ca41339c98b2b17d13d69be3)")
C linker for the host machine: /Users/<username>/code/llvm-project/build-native/bin/clang --sysroot /Users/<username>/code/toolchain-garbage/sysroot --target=riscv32-unknown-none-elf -march=rv32imac_xwchc -ffunction-sections -fdata-sections -fno-ident -nostdlib ld.lld 19.0.0
C compiler for the build machine: ccache cc (clang 14.0.0 "Apple clang version 14.0.0 (clang-1400.0.29.102)")
C linker for the build machine: cc ld64 819.6
Build machine cpu family: aarch64
Build machine cpu: aarch64
Host machine cpu family: riscv32
Host machine cpu: riscv32
Target machine cpu family: riscv32
Target machine cpu: riscv32
<snip>
Message: Linking for RISCV with ld.lld < 15, forcing -mno-relax
```

I'm not sure why this is happening or how to debug it

keith-packard, 2024-07-15:
Old versions of lld (pre-15) didn't support linker relaxations (changing the code depending on the values of symbols). Picolibc needs to add the -mno-relax option for those versions. It attempts to detect the version of lld using a heuristic about command line options which is just completely broken; I'm not sure how it ever worked.

In any case, check out #774 and see if that helps.

arichardson, 2024-07-23:
@ArcaneNibble would you be able to share your meson.log for the old version that detected the wrong thing? I'd like to fix this properly and would need to know why it failed for you.

ArcaneNibble, 2024-07-23:
Okay, I decided to actually do some digging and investigate this properly. Relevant part of meson.log is:
```text
-----------
Command line: `/Users/<username>/code/llvm-project/build-native/bin/clang --sysroot /Users/<username>/code/toolchain-garbage/sysroot --target=riscv32-unknown-none-elf -march=rv32imac -ffunction-sections -fdata-sections -fno-ident -nostdlib /Users/<username>/code/toolchain-garbage/testesttest-build/meson-private/tmpokd5w6d2/testfile.c -o /Users/<username>/code/toolchain-garbage/testesttest-build/meson-private/tmpokd5w6d2/output.exe -D_FILE_OFFSET_BITS=64 -O0 -Werror=implicit-function-declaration -Wl,--fatal-warnings --package-metadata=1` -> 1
stderr:
clang: error: unknown argument: '--package-metadata=1'
-----------
Compiler for C supports link arguments --package-metadata=1: NO 
Message: Linking for RISCV with ld.lld < 15, forcing -mno-relax
```

The problem seems to be that the argument isn't preceded by `-Wl,`
```
$ ../llvm-project/build-native/bin/clang --target=riscv32-unknown-none-elf -march=rv32imac -nostdlib --package-metadata=1 -o /dev/null aaa.c
clang: error: unknown argument: '--package-metadata=1'
$ echo $?
1
$ ../llvm-project/build-native/bin/clang --target=riscv32-unknown-none-elf -march=rv32imac -nostdlib -Wl,--package-metadata=1 -o /dev/null aaa.c
ld.lld: warning: cannot find entry symbol _start; not setting start address
$ echo $?
0
```

arichardson, 2024-07-23:
Oh in that case I have no idea how it ever worked! It should be `-Wl,--package-metadata=1`... I wonder if there was some meson behaviour change related to `cc.has_link_argument`.

arichardson, 2024-07-23:
> Oh in that case I have no idea how it ever worked! It should be `-Wl,--package-metadata=1`... I wonder if there was some meson behaviour change related to `cc.has_link_argument`.

I could also be a clang/gcc behaviour change that previously forwarded unknown arguments on the linker command line to the linker.

arichardson, 2024-07-23:
Submitted #786 to fix this.

PR #774 (2024-07-19): clang: Adjust lld version detection method
Instead of attempting to detect the lld version by checking for a specific compiler flag, just assume the clang version is the same as the lld version.

Closes: #773
closed 2024-07-24 by keith-packard