picolibc/picolibc #778: Tab character is missing the blank flag in _CTYPE_DATA

DavidSpickett, 2024-07-19:
One of libcxx's tests has an XFAIL for Picolib - https://github.com/llvm/llvm-project/blob/main/libcxx/test/std/localization/locale.categories/category.ctype/facet.ctype.special/facet.ctype.char.statics/classic_table.pass.cpp

It fails because its expectation of isblank is tab or space, and it then expects to see the blank flag set for both of those. In picolib it's only set for space (https://github.com/picolibc/picolibc/blob/main/newlib/libc/ctype/ctype_.c).

I found the newlib change (https://sourceware.org/git/?p=newlib-cygwin.git;a=commit;h=a29a76934280b14aa7f13bbb3e86b1a8a2bcdfb7) and a discussion (https://newlib.sourceware.narkive.com/5ImQq910/patch-rfa-fix-ctype-table-and-isblank) about it. The sticking point was that isprint checked for this _B flag and adding it to tab had caused tab to return true from isprint.

isprint is still defined the same way in picolib 
https://github.com/picolibc/picolibc/blob/1c4d89e5c66c22b1768aeaa66348515214a29a72/newlib/libc/ctype/isprint.c#L92

My initial idea is to have isprint check that the char is not tab then check the flags, and add _B back to the table.

Is that viable or are we stuck with this decision now? I have no idea of the consequences of such a change.

DavidSpickett, 2024-07-19:
Also this is only affecting this one test, I have no other reports of problems in the wild. So if it's unfixable or a big job, I'd be fine with won't fix and just having this issue to record the details.

keith-packard, 2024-07-19:
Reading the C17 spec, it's very clear that isblank could be implemented as an inline function which returned `isspace(c) || c == ' ' || c == '\t';` I'm not sure what the fuss is about here; I'll just go do that.

keith-packard, 2024-07-19:
Oh, I'll note that picolibc already had a fix for this, but it was wrong for non-C locales as it strictly checked for space or tab. So your test should pass with current picolibc bits. I'm putting together a fix for the bug I found, along with adding a test for the C locale values.

DavidSpickett, 2024-07-24:
Thanks for the swift response, but I think I was not clear about the issue initially and should have given you a smaller example not just the whole test case. 

Maybe if I restate it more clearly you can tell me if you expect this issue to have been fixed already.

The test gets a pointer to `classic_table` which in this case is a table of masks. It then loops over the first 128 characters and "manually" (not using the clib's isprint/blank/etc functions) works out what flags *should* be set in that table.

So specifically for tab the problem happens with:
```
bool expect_blank = (i == '\t' || i == ' ');
```
By this logic tab should be blank, and picolib's isblank agrees.

```
assert(bool(p[i] & F::blank) == expect_blank);
```
But this fails, because `_B` is not set in the masks table. Which I'm assuming to be this data:
https://github.com/picolibc/picolibc/blob/48fbc2009c6473293d03d5ec6f190565c6223a5c/newlib/libc/ctype/ctype_.c#L36

But I could be missing some intermediate processing steps there.

A smaller program that reproduces the error (at least from C++) is:
```
#include <locale>
#include <cassert>

int main(int, char**)
{
  int tab_code = 9;
  typedef std::ctype<char> F;

  assert(!std::isprint(tab_code));
  assert((F::classic_table()[tab_code] & F::print) == 0);

  // isblank does not agree with the table flags.
  assert(std::isblank(tab_code));
  // This fails because the _B flag is missing from the table.
  assert((F::classic_table()[tab_code] & F::blank) != 0);

  return 0;
}
```

I have tested this with the latest picolib (48fbc2009c6473293d03d5ec6f190565c6223a5c)

This is the docs for classic_table, for what it's worth - https://en.cppreference.com/w/cpp/locale/ctype_char/classic_table

This is what the standard says about it:
```
22.4.1.3.3 ctype<char> static members [facet.ctype.char.statics]
static const mask* classic_table() noexcept;
1 Returns: A pointer to the initial element of an array of size table_size which represents the classifications of characters in the "C" locale.
```
So given the newlib change happened in 2009, prior to c++ 11, it was a valid change at the time. It's only later that this internal table was exposed to c++ (unless there are C uses of it too?).

Given that no one has complained, and there are very few references to classic_table anywhere online, it's unlikely to be breaking anyone.

But I would like to get your opinion on whether it would be a simple fix, or whether changing that table now would break a lot of assumptions made elsewhere.

I tried a few hacks but every one ended up making `isprint(tab)` true as a side effect.

keith-packard, 2024-07-24:
Oh! There's some C++ thing I was unaware of. Now I understand why the library needs to always use the table-driven logic for C++ -- that standard requires the use of a table, effectively exposing the internal implementation through the API. Sigh.

Doing post-processing hacks on the table data to distinguish between `isblank` and `isprint` won't work, so we need to expose a table sufficient for the C++ standard. I'll have to review that in more depth to understand the exact requirements; it's far more complicated than the C standard.

DavidSpickett, 2024-07-25:
> So given the newlib change happened in 2009, prior to c++ 11, it was a valid change at the time. It's only later that this internal table was exposed to c++ (unless there are C uses of it too?).

I was mistaken, classic_table has been present since c++98 (c++11 just changed a keyword). From a quick look I don't see a similar test case in libstdc++ so I guess it just never came up until now.

> Doing post-processing hacks on the table data to distinguish between isblank and isprint won't work, so we need to expose a table sufficient for the C++ standard. I'll have to review that in more depth to understand the exact requirements; it's far more complicated than the C standard.

If you get to an idea of how it might be done, we (myself or a colleague) can work on the implementation of it. After all, we're the ones relying on this picolibc/libcxx stack, we should contribute more than bugs.

keith-packard, 2024-07-25:
Ok, reading through the C++ 13 spec, that's pretty prescriptive about the underlying ctype implementation. In particular, it requires distinct mask values for the following names:
```
// numeric values are for exposition only.
static const mask space = 1 << 0;
static const mask print = 1 << 1;
static const mask cntrl = 1 << 2;
static const mask upper = 1 << 3;
static const mask lower = 1 << 4;
static const mask alpha = 1 << 5;
static const mask digit = 1 << 6;
static const mask punct = 1 << 7;
static const mask xdigit = 1 << 8;
static const mask blank = 1 << 9;
static const mask alnum = alpha | digit;
static const mask graph = alnum | punct;
```
Note that this contains unique values for space, print and blank. Also note that it has ten unique values, which requires that we use a type larger than 8 bits to hold 'mask'.

Given this, I believe we've got two choices:

 1. Switch picolibc to a 16-bit mask, replacing the existing mask values with the set above.
 2. Add a separate table exposed only for C++ with those semantics.

I think 1. is a pretty clear favorite here. We can eliminate the support for negative char values which consumes 128 bytes (and only helps broken applications). At that point, expanding the table to 16 bits will add only 128 more bytes to the table. C applications interested in minimizing the size of the ctype implementation already have the `_PICOLIBC_CTYPE_SMALL` option which has no table at all.

keith-packard, 2024-08-24:
I'm going to push this off past the upcoming picolibc release; it looks like fixing this will mostly involve fixing the relevant C++ libraries anyways, and I need a new picolibc release to keep it from dropping out of Debian testing.

keith-packard, 2026-11-10:
The picolibc side of this is complete; your C++ library can use the C++-specific bits in picolibc to resolve this there.

PR #779 (2024-07-19): ctype: Fix isblank for non-C locales
In non-C locales, isblank is supposed to return true for locale-specific characters for which isspace is true and are used to separate words within a line of text. That's what the _B bit does in the ctype tables. That bit is also used by 'isprint', which should not include tab, so we need an explicit test for tab in isblank.

Closes: #778
closed 2026-11-10 by keith-packard